forkineye/ESPixelStick #265: Configuration for FPP remote show effects as Secondary Input mode  Should it?

patdelaney, 2020-12-26:
**ESPixelStick Firmware Version**
Latest Check-in 12/26/2020
git rev-parse origin/unify
c0712ca83b26fa76dada34739f0bf3b9188f3865

**Hardware Version**
ESP8266

**Binary release or compiled yourself?**
self compiled

**Describe the bug**
When trying to configure the Pixel Stick as a FPP remote the Secondary input defaults to effects (see screenshot)
Should it?
[image]

MartinMueller2003, 2020-12-27:
Not on mine:

forkineye, 2020-12-27:
Web frontend is still under re-work, lots to be done but I've been away for the past week.

patdelaney, 2021-04-08:
This no longer happens. This could have been a browser cache issue or I didn't run gulp

closed 2021-04-08 by patdelaney