forkineye/ESPixelStick #266: Multiple FPP "Upload Output" appears to corrupt Pixelstick Config and break webserver

patdelaney, 2020-12-26:
**ESPixelStick Firmware Version**
ESPixelStick_TRAVIS-20201224134950.zip

**Hardware Version**
esp8266

**Binary release or compiled yourself?**
ESPixelStick_TRAVIS-20201224134950.zip

**Describe the bug**
I'm trying to test workflows with xLights and there appears to be a bug with uploading the output configuration from xLights. If I do it more than once it appears to corrupt the config and basically I have to re flash the pixelstick.  I can't reliably connect to the web interface.

in xlights configure a pixelstick as a device.
Create a test sequence
upload Inputs and upload outputs to pixel stick.

It appears that the color order is saved during the upload outputs, so I try to output again, but then that fails.

Since it's my understanding that xlights polls the pixelstick to build the config that we are going to send back, there must be some bad data in the config and when xlights trys to parse it and repackages it, something in the new packet is bad, and breaks the config.

patdelaney, 2020-12-26:
`@ about 14:12  I uploaded inputs
@ about 14:13  I uploaded Outputs
I checked Web Ui to verify content
@ about 14:14  I uploaded Outputs a 2nd time 
I check Web Ui and I got site can't re reached. It is pingable but the sebser appears to be down.
Here are the logs 

ESPixelStick v4.0-dev (NOT STABLE)(Dec 24 2020 - 13:51:01)
ESP Version: SDK:2.2.2-dev(38a443e)/Core:2.7.4=20704000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be
File system initialized.
Listing directory: /
'config.json': 	'285'
Listing directory: www
'alexa.html.gz': 	'134'
'ddp.html.gz': 	'130'
'disabled.html.gz': 	'130'
'dmx.html.gz': 	'374'
'e1_31.html.gz': 	'393'
'effects.html.gz': 	'579'
'esps.css.gz': 	'13698'
'esps.js.gz': 	'57745'
'fpp_remote.html.gz': 	'391'
'gece.html.gz': 	'425'
'index.html.gz': 	'2649'
'mqtt.html.gz': 	'714'
'relay.html.gz': 	'483'
'renard.html.gz': 	'439'
'serial.html.gz': 	'544'
'ws2811.html.gz': 	'1405'
Configuration file: '/config.json' loaded.
** OutputDisabled Initialization for Chan: 0 **
** OutputDisabled Initialization for Chan: 1 **
Configuration file: '/output_config.json' not found.
EEEE Error loading Output Manager Config File. EEEE
--- WARNING: Creating a new Output Manager configuration Data set - Start ---
** WS281x Initialization for Chan: 0 **
************** Cannot Start WS2811 for channel '1'. **************
** OutputDisabled Initialization for Chan: 1 **
** GECE Initialization for Chan: 0 **
************** Cannot Start GECE for channel '1'. **************
** OutputDisabled Initialization for Chan: 1 **
** DMX Initialization for Chan: 0 **
************** Cannot Start DMX for channel '1'. **************
** OutputDisabled Initialization for Chan: 1 **
** Renard Initialization for Chan: 0 **
************** Cannot Start Renard for channel '1'. **************
** OutputDisabled Initialization for Chan: 1 **
** Serial Initialization for Chan: 0 **
************** Cannot Start Generic Serial for channel '1'. **************
** OutputDisabled Initialization for Chan: 1 **
************** Cannot Start RELAY for channel '0'. **************
** OutputDisabled Initialization for Chan: 0 **
** Relay Initialization for Chan: 1 **
** OutputDisabled Initialization for Chan: 1 **
Configuration file: '/output_config.json' saved.
**** Saved Output Manager Config File. ****
--- WARNING: Creating a new Output Manager configuration Data set - Done ---

WiFi Entering State: Connecting Using Config Credentials

WiFi Connecting to paddycakes as artstick-sd2
** 'Disabled' Initialization for input: '0' **
** 'Disabled' Initialization for input: '1' **
.Configuration file: '/input_config.json' not found.
EEEE Error loading Input Manager Config File. EEEE
--- WARNING: Creating a new Input Manager configuration Data set - Start ---
** 'E1.31' Initialization for input: '0' **
** 'Disabled' Initialization for input: '1' **
** 'E1.31' Shut Down for input: '0' Requires a reboot. **
** 'Disabled' Initialization for input: '0' **
[... 444 lines cut ...]
'serial.html.gz': 	'544'
'ws2811.html.gz': 	'1405'
Configuration file: '/config.json' loaded.
** OutputDisabled Initialization for Chan: 0 **
** OutputDisabled Initialization for Chan: 1 **
** WS281x Initialization for Chan: 0 **
Configuration file: '/output_config.json' loaded.

WiFi Entering State: Connecting Using Config Credentials

WiFi Connecting to paddycakes as artstick-sd2
** 'Disabled' Initialization for input: '0' **
** 'Disabled' Initialization for input: '1' **
.** 'E1.31' Initialization for input: '0' **
Configuration file: '/input_config.json' loaded.
- Web Server started on port 80
FPPDiscovery subscribed to broadcast
FPPDiscovery FAILED to subscribed to multicast messages

WiFi Entering State: Connected To AP
WiFi Connected with DHCP
WiFi Connected with IP: 192.168.1.161
E1.31 Multicast Enabled.
E1.31 Unicast Enabled on port: 5568
Listening for 768 channels from Universe 1 to 2.`

patdelaney, 2020-12-26:
[xlights.log](https://github.com/forkineye/ESPixelStick/files/5743597/xlights.log)
attaching xlights log

forkineye, 2020-12-27:
Does this go with https://github.com/forkineye/ESPixelStick/issues/255?

patdelaney, 2020-12-27:
They may be related in the end, but with this issue the pixelstick webserver just stops working.  I can ping it but that is it.

MartinMueller2003, 2020-12-31:
This tells me there is something odd going on in the Ethernet layer: "FPPDiscovery FAILED to subscribed to multicast messages". That should never fail since the WiFi and Ethernet objects are established long before the setup call is made. That means the WiFi config got corrupted. I can tell you that based on the logs, the App configs are ok. It also looks like you have the Web UI connected to the ESP at the same time as the XLights sockets are open. That uses a lot of memory and causes reboots on the device.  The ESP does not work well with multiple concurrent web sockets open. File transfers will fail badly in such a mode.

patdelaney, 2021-04-05:
tested with Build Date
Apr 5 2021 - 11:58:32   and upload input adn upload output work as expected.
Tested ESP32 and ESP8266
tested against xLights 2021.11

closed 2021-04-05 by patdelaney