forkineye/ESPixelStick #267: Please add New Section to Home screen called FPP Remote and then display the sync packet count

patdelaney, 2020-12-26:
**Is your feature request related to a problem? Please describe.**
Not really related to a problem , but more for Troubleshooting

**Describe the solution you'd like**
As a user I would like too see the number of sync packets that Pixelstick is getting just like E1.31 data

**Additional context**
Add any other context or screenshots about the feature request here.
[image]

MartinMueller2003, 2020-12-27:
I am not sure what additional information you are looking for. DDP does not have the concept of universes, nor does the user have a choice about how the data is wrapped. DDP sends blocks of contiguous channels and adjusts the block size based on the ack rate (if the sender implements adaptive rate control.
Martin

forkineye, 2020-12-27:
Would be nice to have the address of the FPP Master that's sending the data.  Not sure if we want to start tracking frame counts or not, but something to actively let the user know that it's receiving data would be nice.   I was planning to change this section anyways to only show status for the active input mode.

MartinMueller2003, 2021-01-06:
I did make the changes to show only the active stats.

patdelaney, 2021-02-08:
This has been completed and tested

closed 2021-02-08 by patdelaney