forkineye/ESPixelStick #268: ESP32 ethernet support

watkinrt, 2020-12-27:
**Is your feature request related to a problem? Please describe.**
In areas with congested WiFi (such as urban areas with 70+ APs), pure wireless based data transmission can suffer from packet errors and ultimately stuttering effects. The current implementation of FPP's multisync in v4 addresses this issue (although I haven't fully tested it out yet), but then requires the overhead of storing fseq files locally on all controllers. Integrations with xLights definitely helps ease the burden of distributing these files, but file transfers can take a long time (10s of minutes for 3x 20fps sequences when I tested the current implementation out.). For small display layouts, or in specialized locations, it would be nice to hardwire a device to improve reliability without the overhead of FPP multisync.

**Describe the solution you'd like**
Add support for ethernet based network communication; in particular, on ESP32 boards which have seemingly robust ethernet capabilities (ethernet MAC and PHY).

**Additional context**
I am willing to attempt an implementation of this feature. Before I do so though, I wanted to check to see:

- Have other people already tried this and ran into issues that I'm not aware of? I know people have tried this with Arduino UNOs and MEGAs, but I believe their primary issues were slow clock speed and reliance on slow I2C based ethernet communication. I think boards like the ESP32 overcome these issues.
- Is this a feature anyone else is interested in? If not, I can fork of of this firmware and just maintain my own build.

If people are interested, I could use some help getting started. I've poked around the current v4 firmware updates and think I have a rough sense on how things are structured. Is there a preferred implementation structure that the primary maintainers would prefer? Right now, I'm planning on creating a EthMgr structure similar to the WiFiMgr structure on the Unify branch.

MartinMueller2003, 2020-12-27:
this would be up to Shelby. However, tyimg a mac to a buss on the esp32 should be doable if not already done.
Martin

forkineye, 2020-12-30:
This is actually something I was planning on implementing in the future once Unify / 4.0 is stable. You're more than welcome to work on it, but It wouldn't be merged in until after 4.0 is released.  If you do, you'll want to sync against the unify branch daily. 
 The older micros usually used the SPI based Wiznet modules.  Given the ESP32 already has an Ethernet MAC, it just needs the physical stuff added.  I haven't dug into the ESP32 Ethernet details yet, but it could be that we migrate WiFiMgr into a more generic NetworkManager in the future.  For now, I'd keep it separate though given the big changes that currently occurring.

watkinrt, 2020-12-30:
Sounds good. I'll start a branch off of Unify to do my implementation and just make sure to merge in updates as they come. Definitely also sounds reasonable to wait for the stable version of v4 to release before merging this implementation in.

I just ordered an ethernet module for my ESP32. Should be here in a few weeks. As a side note, it looks like most ESP32 modules that contain an ethernet functionality have either GPIO2 or GPIO13 tied up on a peripheral (SD card for most of them, but really depends on the module). As the current firmware is hardcoded to use both GPIO2 and GPIO13 for the ESP32, there is a conflict here with the hardware. I'm currently getting around this by getting a separate [Waveshare LAN8720 ethernet](https://www.waveshare.com/wiki/LAN8720_ETH_Board) module to connect to my Lolin D32 board, which doesn't have this hardware conflict. Something to consider in the future though. I know that the GPIO2 is hard to change for backward compatibility with all of your boards (especially those that utilize an ESP8266-01), but is there a way to migrate from GPIO13 to another pin for the second output? I haven't poked around much in the firmware, but my guess is that you just want the second output on the ESP32 to be on a different output port than the second one?

forkineye, 2020-12-30:
The current implementation is based off the way I had to do it on the ESP8266.  Expect more ESP32 support and more port options once 4.0 is stable.

MartinMueller2003, 2020-12-31:
the gpio pins used for the sd card are configurable on the ui. the compiled in defaults work with most common dev boards but they can be changed at any time.
martin

watkinrt, 2021-01-10:
Just wanted to post a quick update. Just got my ethernet module board in yesterday and I was successfully able to get ethernet up and running without too much of an issue. Here are a few details:

- I currently have it running on a Waveshare LAN8720 board connected to a D32 Board. With that being said, I don't foresee any issues with running this on any of the current ethernet boards for ESP32s.
- In the WiFiMgr, I essentially added in new fsm_WiFi_state subclasses for ethernet "connecting" and "connected" states. As I don't know how the ESP32 deals with network traffic when both the ethernet and Wifi are connected, I currently only allow one to connect. As you presumably want ethernet to take priority over WiFi, I patched the ethernet checks in prior to the wifi AP checks.
- I was successfully able to stream over E 1.31 but not DPP (haven't looked into how the DPP library is setup and whether or not is supports ethernet yet).

There is still quite a bit of cleanup work to do, but minimally encouraging that it didn't take too much work to get things working.

I currently have this on my forked version of the code and my own "ethernet" branch that comes off of the unify branch. Do you want to add in a branch off of unify to carry this or do you want to wait until v4 is released before pulling this in?

MartinMueller2003, 2021-01-10:
Shelby's call but I would pull it in sooner rather than later. No real reason to wait until the code is released. How are you handling the ESP8266 nodeMcu mode of operation?
Martin

MartinMueller2003, 2021-01-10:
And then there is this card as an all in one.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-ethernet-kit.html

Martin
    On Sunday, January 10, 2021, 09:07:58 AM EST, Martin Mueller <martinmueller2003@yahoo.com> wrote:  
 
  Shelby's call but I would pull it in sooner rather than later. No real reason to wait until the code is released. How are you handling the ESP8266 nodeMcu mode of operation?
Martin

watkinrt, 2021-01-10:
> How are you handling the ESP8266 nodeMcu mode of operation?

Right now, I'm not. For the ESP-32 boards, they have hardware support for ethernet, which made this conversion pretty seamless. For the ESP8266 boards, it is possible to add ethernet capability (with something like the W5500 ethernet module), but the communication is through SPI rather than the hardware layer that exists on the -32 boards. So, definitely not an option for the -01 ESP boards, but maybe an option for the -12 boards. I haven't fully looked into this, but I believe reliance on SPI could cause some performance issues. So, right now, I only have ethernet capability setup for the -32 boards. I will try to look into the W5500 modules and get one for my ESP8266-12 module and see if I can get something working.

> And then there is this card as an all in one.
> 
> https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-ethernet-kit.html

That is definitely an options. I went with my current setup as I already had a D32 board and wanted to stay consistent with the current -32 hardware that has been used with the firmware (You definitely need to be more careful with my setup as these ethernet signals are very sensitive to noise, so you need to keep your trace distances as short as possible). With that being said, for a fully integrated board, I would probably lean more towards to [Olimex boards](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware) as they are cheaper, have a smaller form factor, and include an SD card reader for those that want to maintain FPP compatibility). If you don't want/need the SD card capability, then the [WT32-ETH01](https://www.cnx-software.com/2020/11/09/6-wt32-eth01-is-a-tiny-esp32-board-with-ethernet/) is probably the best option as it is even smaller and cheaper (~$8 vs ~$20). There is also some interesting potential for PoE, but I'm not sure if that is really cost effective at the end of the day (because you either need a switch that is PoE capable -- which tend to be quite expensive -- or an addition PoE injector ), nor capable of suppling the necessary power for these applications.

MartinMueller2003, 2021-01-10:
I vote for boards WITH an SD card slot on them. Even with an Ethernet port, playing from a local SD card is MUCH more bandwidth efficient.
Martin

 
 

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-ethernet-kit.html

mkozik13, 2021-01-10:
Plus allows for standalone props for all holidays!

watkinrt, 2021-01-10:
Then the [Olimex ESP32-POE](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware) is probably the best option based on available hardware features, form factor, and price point (although I haven't tested it out). Another potential option is the [TTGO T-Internet-POE](http://www.lilygo.cn/prod_view.aspx?TypeId=50033&Id=1307&FId=t3:50033:3), although the form factor is larger than the Olimex board and I believe it needs a secondary board for programing.

### Olimex
[image]

### TTGO
[image]

watkinrt, 2021-01-14:
I poked around a little bit more into adding in ethernet functionality for the ESP8266. Unfortunately, Arduino's Ethernet library (which is the standard library for I2C ethernet boards) doesn't appear to have event handling (i.e., disconnector or got IP), which is how network connect and disconnect events are currently handled. You could probably hack something in so that you could poll the ethernet interface to detect these event, but that is less than ideal. My current thoughts are that this implementation isn't worth it and that ethernet functionality will only be available for the ESP32 boards. Thoughts?

MartinMueller2003, 2021-01-14:
ok by me.

processor83, 2021-02-13:
I would like to help testing this feature. Just to make sure I can test with a lan8207 and a wemos d32 mini?

watkinrt, 2021-02-13:
> Just to make sure I can test with a lan8207 and a wemos d32 mini?
@processor83 That is my current setup (noting that I think you have a typo regarding the ethernet board: LAN8720). Note that, due to the way board hardware configuration, there are some conflicts between the ethernet setup and the SD card reader. So, you can either use the ethernet capability OR the SD card capability, but not both at the same time. In the future, I think we will want to switch over to one of the Olimex or TTGO boards moving forward as they don't have this hardware conflict.

> I would like to help testing this feature.
@processor83 I have a prototype version up running, but I'm waiting for the unify branch to settle out and for v4 to release before I move anything more forward. You can find my forked branch here: https://github.com/watkinrt/ESPixelStick on the "ethernet" branch. If you actually want to try it out, I'd have to send you a few more details on how to build the code (which depends on how you have your ethernet board setup).

processor83, 2021-02-13:
>

About the hardware limitation is it because of the wemos d32 min that dont have all needed pin broken out ?? is it just for SD card CS pin or other pins? Also can i take the 3.3V from the esp board or need other regulator?

Thank you.

watkinrt, 2021-02-13:
> @processor83 About the hardware limitation is it because of the wemos d32 min that dont have all needed pin broken out ?? is it just for SD card CS pin or other pins? 
Some of the ethernet pins have to be connected to specific GPIO pins on the ESP32: EMAC_TXD0, EMAC_TXD1, EMAC_TX_EN, EMAC_RXD0, EMAC_RXD1, EMAC_RX_DV and EMAC_TX_CLK. Unfortunately, some of the SD pins on the Wemos D32 board are hardwired up to the some of mandatory ethernet pins (GPIO18, GPIO19, and GPIO23 if I recall correctly). See [this](https://sautter.com/blog/ethernet-on-esp32-using-lan8720/) article for how to setup the ethernet board with your D32.

processor83, 2021-02-13:
I will look at the article but i guess when you say wemos SD you mean the SD card shield ?

Thank you

processor83, 2021-02-13:
maybe my question is a little prematured but if i would like to move SD to other pins on my design and make it possible to use sd and ethernet should i follow olimex pinout?

processor83, 2021-02-13:
or are you talking about the wemos D32 pro that have the micro sd on it ?

MartinMueller2003, 2021-02-13:
yes, i am using the lolin d32 pro and that is what the defaults are set up for

MartinMueller2003, 2021-02-13:
FYI: While not exposed on the UI, It is possible to move almost any functionality to a different pin. Even the pixel outputs can be moved/configurable. 

I do not have an 'SD Shield" The LoLin D32 PRO has the SD card integrated into it. All of the other common buses have connectors ready to be used. It sounds like you need to look at the driver configuration to see if you can change the Ethernet IO pin assignments. However, if the Ethernet driver expects to use the UARTs then we have a problem.
Martin    On Saturday, February 13, 2021, 05:37:03 PM EST, Martin Mueller <martinmueller2003@yahoo.com> wrote:  
 
 yes, i am using the lolin d32 pro and that is what the defaults are set up for

processor83, 2021-02-14:
>  
> FYI: While not exposed on the UI, It is possible to move almost any functionality to a different pin. Even the pixel outputs can be moved/configurable. 
> 
> I do not have an 'SD Shield" The LoLin D32 PRO has the SD card integrated into it. All of the other common buses have connectors ready to be used. It sounds like you need to look at the driver configuration to see if you can change the Ethernet IO pin assignments. However, if the Ethernet driver expects to use the UARTs then we have a problem.
>  
>  yes, i am using the lolin d32 pro and that is what the defaults are set up for
> 
> Sent from Yahoo Mail on Android 
>  
> 
> 
> or are you talking about the wemos D32 pro that have the micro sd on it ?

I looked at Olimex pin out and it don't use gpio 10 or 13 for Ethernet or sd card so I will not touch led pin just SD and to be able to use the mac pins.

processor83, 2021-02-15:
> Some of the ethernet pins have to be connected to specific GPIO pins on the ESP32: EMAC_TXD0, EMAC_TXD1, EMAC_TX_EN, EMAC_RXD0, EMAC_RXD1, EMAC_RX_DV and EMAC_TX_CLK. Unfortunately, some of the SD [...]

I looked at olimex ESP32-POE board that was mentioned and they use une 1 data line mode for se card can it be used with espixelstick firmware? Or only spi mode for se card?

processor83, 2021-02-15:
Sorry I'm not super used to using GitHub is it the right place to continue this discussion or there is other place to get in contact?  I looked at it a little more and there is two spi pinout on the esp32 and the vspi I think don't use any pin used for Ethernet. I'm waiting on esp32/lan8720 and more SD module to test all this!

processor83, 2021-03-04:
I received some mh-et live esp32 mini boards and china LAN8720 and tryed to replicate the setup but esp32 will not boot. i used a resistor in series with the clk signal on gpio0 and it boot but is not able to start phy there is a time out.  can a diffrwent pull up resistor be used is it posible that the problem is comming from the esp32 board it have auto program and not prog button on it.  How hard it is to use esp32 internet clock in the software and remove this osciator on the lan board maybe.

processor83, 2021-03-04:
Ok i managed to make the hardware work with the example. i removed the crystal and using esp32 clokc signal. Where i must define setting for eth.h?

#ifdef ETH_CLK_MODE
#undef ETH_CLK_MODE
#endif
#define ETH_CLK_MODE    ETH_CLOCK_GPIO17_OUT

// Pin# of the enable signal for the external crystal oscillator (-1 to disable for internal APLL source)
#define ETH_POWER_PIN   -1

// Type of the Ethernet PHY (LAN8720 or TLK110)
#define ETH_TYPE        ETH_PHY_LAN8720

// I²C-address of Ethernet PHY (0 or 1 for LAN8720, 31 for TLK110)
#define ETH_ADDR        1

// Pin# of the I²C clock signal for the Ethernet PHY
#define ETH_MDC_PIN     23

// Pin# of the I²C IO signal for the Ethernet PHY
#define ETH_MDIO_PIN    18

processor83, 2021-03-05:
Ok for any one that is trying to have ethernet here is how i managed to make it work.

Hardware:
lan8720 module https://fr.aliexpress.com/item/32825720322.html?spm=a2g0w.12010615.8148356.1.50a267c7LIIgqO
i removed the osciator on it to make wire more simple.
esp32 module https://fr.aliexpress.com/item/32816073234.html?spm=a2g0s.9042311.0.0.11236c37xMOMbc

WIRING:
GPIO22 - EMAC_TXD1   : TX1
GPIO19 - EMAC_TXD0   : TX0
GPIO21 - EMAC_TX_EN  : TX_EN
GPIO26 - EMAC_RXD1   : RX1
GPIO25 - EMAC_RXD0   : RX0
GPIO27 - EMAC_RX_DV  : CRS
GPIO17 - EMAC_TX_CLK : nINT/REFCLK (50MHz)
GPIO23 - SMI_MDC     : MDC
GPIO18 - SMI_MDIO    : MDIO
GND                  : GND
3V3                  : VCC

This way of wiring dont need any resistor but make sure to remove the osciator on the lan8720 board.

Software side i used watkinrt fork https://github.com/watkinrt/ESPixelStick/tree/ethernet.

Since i use the esp32 module clock signal we have to edit WiFiMgr.cpp.

add the following lines afther #   include <ETH.h>

#define ETH_CLK_MODE    ETH_CLOCK_GPIO17_OUT

// Pin# of the enable signal for the external crystal oscillator (-1 to disable for internal APLL source)
#define ETH_POWER_PIN   -1

// Type of the Ethernet PHY (LAN8720 or TLK110)
#define ETH_TYPE        ETH_PHY_LAN8720

// I²C-address of Ethernet PHY (0 or 1 for LAN8720, 31 for TLK110)
#define ETH_ADDR        1

// Pin# of the I²C clock signal for the Ethernet PHY
#define ETH_MDC_PIN     23

// Pin# of the I²C IO signal for the Ethernet PHY
#define ETH_MDIO_PIN    18

change this line:
  ETH.begin;
to:
  ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE);

Its all tested and work perfectly web ui is so fast with ethernet :).

Hope it will help other get ethernet working and more to test this feature!

Feal free to ask if you have any question :)

MartinMueller2003, 2021-03-05:
Nice. Now we just need to find two gpio pins we can use for the pixel outputs.
Martin

processor83, 2021-03-05:
My plan is to setup SD card then find two gpio for the pixels hope its possible ! I will make a spead cheat to see what is left and see if all hspi pins are free for sd and see what is left! The othe solution is to change sd library to use single wire for sd card like olimex boards this will free some pins! Will let you know what is free now!

processor83, 2021-03-05:
Ok if i use HSPI pin out for the SD card its compatible with ethernet and leave those pin free but i also see that maybe sd could prevent boot if it pull gpio 12 high at boot have to check this!

GPIO0
GPIO4   --- if i look at the doc this pin
GPIO22 --- and this one are a good choice
GPIO1
GPIO34
GPIO35
GPIO36
GPIO39
GPIO16
GPIO32
GPIO33

processor83, 2021-03-05:
where should i report bug using ethernet?? can io report them in a new issue here ?

watkinrt, 2021-03-05:
Glad you were able to get things up and running. Because of a lot of these setups are hardware dependent, it is hard to know what you’re going to get until you just try it out. 

I recently tested out the WT32-ETH01 board (https://www.seeedstudio.com/Ethernet-module-based-on-ESP32-series-WT32-ETH01-p-4736.html <https://www.seeedstudio.com/Ethernet-module-based-on-ESP32-series-WT32-ETH01-p-4736.html>), which is pretty cheap (~$7 US) but doesn’t have an SD card built in. I’m currently looking to see if I can hook up an SD card reader to this board to get full functionality along with 2 outputs. Unfortunately, the default SPI configurations aren’t possible due to the limited IO pins that are exposed (missing GPIO13 to use default HSPI interface), but they can be modified as needed (noting that you just take a performance hit from a high end of 80 MHz down to 40 MHz). At the end of the day, for my application, I will either be using ethernet or SD, so this limitation isn’t a big issue for me if I can’t get it working on this board.

Regarding bugs, if they have to do with ethernet functionality, it is probably worth reporting them as issues on my forked version of the code.

MartinMueller2003, 2021-03-05:
Hi Ryan
Are you keeping your fork up-to-date with unify?
Martin

processor83, 2021-03-05:
i was looking at comment on this board WT32-ETH01 and when i read this comment i did not 100% know what problem it can give but did not want anything that is not standard!   the solution with those cheap lan8720 module is easy when you remove the osciator its way less complicated to do !

For issue i think you must enable the issue tracker is not there now.

his module does not accept large chunks of the ESP32 AT command set detailed here:
https://www.espressif.com/s... (PDF), many returning 'ERROR'
Especially, very few of the Wi-Fi AT commands work, most return "module ok" and no error, which I suspect means it triggered a reboot.
Issuing a AT+CWMODE? returns +CWMODE:240 which is not a valid mode in my reading so far.

If anyone has gotten the Wi-Fi on this thing actually listing AP's and maybe even *connecting* I would be delighted to hear...

watkinrt, 2021-03-06:
> @processor83 i was looking at comment on this board WT32-ETH01 and when i read this comment i did not 100% know what problem it can give but did not want anything that is not standard! "his module does not accept large chunks of the ESP32 AT command set detailed here:
https://www.espressif.com/s... (PDF), many returning 'ERROR'
Especially, very few of the Wi-Fi AT commands work, most return "module ok" and no error, which I suspect means it triggered a reboot.
Issuing a AT+CWMODE? returns +CWMODE:240 which is not a valid mode in my reading so far.
If anyone has gotten the Wi-Fi on this thing actually listing AP's and maybe even connecting I would be delighted to hear..."

This comment on the seeed.com website refers to the stock firmware on the devices rather than the capabilities of the ESP32. I just cleared the flash and reprogrammed the board without any issues. Works like a normal ESP32 board from there. It is kind of a pain not having a build in USB programming interface, but I build a custom cable to make this easier.

> @MartinMueller2003 Are you keeping your fork up-to-date with unify?

I have been, but I'm a little bit behind right now. I was planning on pulling in the current version this weekend.

> @processor83 For issue i think you must enable the issue tracker is not there now.

Just enabled issue tracking on my fork of the repository.

processor83, 2021-03-06:
Ok never thinked some one will complain about a pre flashed board ... to me i never looked at those board with pre flashed firmware always flashed them ! then it make more sense and will look itot it  and all sd card on my board would be less work !

MartinMueller2003, 2021-04-06:
> @forkineye would you consider changing default esp32 pin to not interfer with lan module pinout?

What pin would you suggest?
Martin

processor83, 2021-04-08:
Here is my pinout that allow for 4 led micro Sd and ethernet. I used this pinout to test on expixelstick ethernet branch and WLED ( for 4 led pins).

GPIO# | MY USE
-- | --
GPIO2 |  
GPIO0 |  
GPIO4 | LED4
GPIO15 | SD- CS/tdo/D8
GPIO22 | PHY-TX1
GPIO21 | PHY-TX_EN
GPIO18 | PHY-MDIO
GPIO19 | PHY-TX0
GPIO23 | PHY-MDC
GPIO5 |  
GPIO1 |  
GPIO3 |  
GPIO9 |  
GPIO10 |  
GPIO34 |  
GPIO35 |  
GPIO36 |  
GPIO39 |  
GPIO25 | PHY-RX0
GPIO13 | SD- MOSI/tck/D7
GPIO12 | SD- MISO/tdi/D6
GPIO14 | SD- CLK/tms/D5
GPIO16 | LED1
GPIO17 | PHY-nINT/REFCLK
GPIO26 | PHY-RX1
GPIO27 | PHY-CRS
GPIO32 | LED2
GPIO33 | LED3

Here is a picture of the pcb i designed to test Ethernet.

[image]

MartinMueller2003, 2021-04-08:
What are LED1 - LED4 used for? Are they the pixel outputs?
Martin

processor83, 2021-04-08:
Yes its 4 data pin i use for led... only two when using espixelstick.

patdelaney, 2021-04-08:
I want one  :-)

processor83, 2021-04-08:
Here it is with esp32 and lan8720 module the black rj45 is to connect to a differential receiver.
[image]

watkinrt, 2021-04-09:
Very cool. Pretty nice set you have there. You'll have to let me know how
your further testing goes.

Ryan

processor83, 2021-04-21:
@MartinMueller2003 just saw you added configurable data pin that so cool! Thank you!

processor83, 2021-06-28:
If any dev want one of my board to test esp32 ethernet i can ship you one. Let me know.  It have ethernet microSD and 4 fused pixels outputs.

processor83, 2021-06-28:
https://canadapixelscoro.ca/products/PRE-ORDER-ESP3DEUXQuattro-p363667332
[image]
  it use those cheap(waveshare clone) lan8720 modules.

watkinrt, 2021-08-16:
Just a quick updated. I just merged v4.0 (i.e., the main branch) into my fork, along with some general robustness improvements. Right now, I've kept everything in the WiFiMgr class, but I think it would ultimately make sense to convert this over to a new class named something like NetworkMgr. 

I haven't done any long term usage tests, but things seem to be reasonably well behaved.

MartinMueller2003, 2021-12-25:
I would take you up on the offer to use one of your boards. Would you also send me a link to your branch? I would like to take a look at the code and see what I can do to integrate it into the product. We changed how we define hardware platforms recently to make it simpler to control what gets compiled and what gets skipped.

MartinMueller2003, 2021-12-27:
@processor83 : Your fork is off the V3 branch which is no longer in development. Did you have a public branch made from V4 (aka latest main)?

processor83, 2021-12-27:
> I would take you up on the offer to use one of your boards. Would you also send me a link to your branch? I would like to take a look at the code and see what I can do to integrate it into the [...]

Can you send me a pm with your shipping inf. I will send one to you. Also i played with watkinrt branch and changed some pin option but did not play with this for some time.

watkinrt, 2021-12-27:
> @processor83 : Your fork is off the V3 branch which is no longer in development. Did you have a public branch made from V4 (aka latest main)?

Yes, take a look at the "ethernet" branch. Note, this is a functioning prototype. It works for the most part, but has one of two known bugs. The primary one that I recall offhand is that static IPs have issues. So, only use DHCP when testing it out.

MartinMueller2003, 2021-12-28:
I have been hand integrating it. I created an ethernet manager and will join them into a network mgr

watkinrt, 2021-12-28:
> I have been hand integrating it. I created an ethernet manager and will join them into a network mgr

That is definitely the way to go. At the time of my development, I wanted to stay within the context of the v4 code so as not to fight branch merging every time something changed.

MartinMueller2003, 2021-12-28:
Your branch is very old and would have been very difficult to merge. When unify became main I had so much trouble I deleted my forks and started over by hand merging my last changes. You are even further behind than that. Since then we have changed to using platform IO and GPIO Defs files to control hardware platforms. That changes how the auto builds bring in libraries and create binary images.

MartinMueller2003, 2022-01-04:
I have created a network management set of classes and have gotten most of the config stuff working. Just got my Ethernet board and I see why you decided to remove the oscillator. The pin for turning the osc is missing from the design. It is present on some boards. We may have to spec that people use a specific Ethernet interface board so that we can get the clock data off of GPIO0.

MartinMueller2003, 2022-01-05:
As I am integrating this I am coming to a different solution than removing the OSC (which many people cant do). The driver has a power control that is used for turning on the interface. I am looking at adding a mosfet on the 3.3v going to the interface and controlling it from a GPIO. That would make the interface not impact GPIO 0 until after boot is complete. The opto isolated version would require a 6 pin dip and a resistor.

MartinMueller2003, 2022-01-05:
> 

@processor83 I cant seem to PM you. Are you on the red or green forums? You can PM me there. MartinMueller2003 is my username on both sites.

MartinMueller2003, 2022-01-08:
Ethernet is working but not stable at all.

[18 more replies]

closed 2022-07-29 by MartinMueller2003