forkineye/ESPixelStick #269: Add file size and Time stamp to File Management  page output

patdelaney, 2020-12-28:
**Is your feature request related to a problem? Please describe.**
As a user I would like to know that date stamp and the file size of sequence files on my pixelstick

**Describe the solution you'd like**
Please add time stamp and file size to the output of the file management page

**Additional context**
[image]

MartinMueller2003, 2020-12-28:
There is a minor problem here. The ESP does not know what time it is so it cannot have meaningful time stamps on the files. Size is no problem. Have to look into the file transfer to see if we can add a time marker to set the ESP local time at the moment the file is transfered. That will not be trivial (aka time zone support would need to be added as well).
Martin

MartinMueller2003, 2020-12-29:
At the moment the timestamp uses whatever time is in the browser client. This is the only time source available to the ESP. Size was easy to get. Had to replace the entire display mechanism since it did not support displaying file attributes.

patdelaney, 2020-12-30:
Just tested time stamp an it appears wrong?
[image]

MartinMueller2003, 2020-12-30:
any file downloaded prior to my latest code changes has a random small time value. delate the file and download it again to get a new date.
martin

patdelaney, 2021-01-07:
Tested on ESP32 - OK
Tested Wemos D1 - OK

Closing

closed 2021-01-07 by patdelaney