forkineye/ESPixelStick #276: ESP32 unable to delete files on SD card after uploaded via FPP connect

patdelaney, 2021-01-03:
**ESPixelStick Firmware Version**
ESPixelStick_TRAVIS-20201230114122.zip

**Hardware Version**
ESP32 Lolin Dev board wit CD caard

**Binary release or compiled yourself?**
Built myself

**Describe the bug**
I uploaded about 12 files to ESP32.
I then uploaded them a second time to see the logs or if it has an issue with overwriting.

I went into the web UI and I can't delete the files that we uploaded. 

I power cycled the stick and  I still can't delete them. There are no error messages on the serial output
Here is the output after hitting reset:
17:03:26.611 -> ets Jun  8 2016 00:22:57
17:03:26.611 -> 
17:03:26.611 -> rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
17:03:26.644 -> flash read err, 1000
17:03:26.644 -> ets_main.c 371 
17:03:26.979 -> ets Jun  8 2016 00:22:57
17:03:26.979 -> 
17:03:26.979 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
17:03:26.979 -> configsip: 0, SPIWP:0xee
17:03:26.979 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:03:26.979 -> mode:DIO, clock div:1
17:03:26.979 -> load:0x3fff0018,len:4
17:03:27.013 -> load:0x3fff001c,len:1044
17:03:27.013 -> load:0x40078000,len:8896
17:03:27.013 -> load:0x40080400,len:5816
17:03:27.013 -> entry 0x400806ac
17:03:27.380 -> 
17:03:27.380 -> ESPixelStick v4.0-dev (NOT STABLE)(Jan  2 2021 - 20:05:12)
17:03:27.380 -> ESP Version: v3.2.3-14-gd3e562907
17:03:27.380 -> Flash File system initialized.
17:03:27.380 -> Listing directory: /
17:03:27.380 -> '/config.json': 	'278'
17:03:27.413 -> '/input_config.json': 	'1273'
17:03:27.413 -> '/output_config.json': 	'1790'
17:03:27.413 -> Listing directory: /www
17:03:27.413 -> '/www/alexa.html.gz': 	'134'
17:03:27.446 -> '/www/ddp.html.gz': 	'130'
17:03:27.446 -> '/www/disabled.html.gz': 	'130'
17:03:27.446 -> '/www/dmx.html.gz': 	'374'
17:03:27.446 -> '/www/e1_31.html.gz': 	'393'
17:03:27.479 -> '/www/effects.html.gz': 	'579'
17:03:27.479 -> '/www/esps.css.gz': 	'13502'
17:03:27.479 -> '/www/esps.js.gz': 	'56038'
17:03:27.479 -> '/www/fpp_remote.html.gz': 	'391'
17:03:27.522 -> '/www/gece.html.gz': 	'425'
17:03:27.522 -> '/www/index.html.gz': 	'2715'
17:03:27.522 -> '/www/mqtt.html.gz': 	'714'
17:03:27.522 -> '/www/relay.html.gz': 	'483'
17:03:27.522 -> '/www/renard.html.gz': 	'439'
17:03:27.522 -> '/www/serial.html.gz': 	'544'
17:03:27.546 -> '/www/ws2811.html.gz': 	'1405'
17:03:27.546 -> SD Card Size: 7594MB
17:03:27.546 -> /System Volume Information/
17:03:27.546 -> 	/System Volume Information/IndexerVolumeGuid		76
17:03:27.580 -> 	/System Volume Information/WPSettings.dat		12
17:03:27.580 -> /butterfly.fseq		4872
17:03:27.580 -> /butterfly2.fseq		4872
17:03:27.580 -> /FPP_Multisync_Test.fseq		1224144
17:03:27.580 -> /FPP_Multisync_Test1.fseq		1224144
17:03:27.580 -> /FPP_Multisync_Test10.fseq		1224144
17:03:27.580 -> /FPP_Multisync_Test11.fseq		1224144
17:03:27.580 -> /FPP_Multisync_Test12.fseq		1224144
17:03:27.614 -> /FPP_Multisync_Test2.fseq		1224144
17:03:27.614 -> /FPP_Multisync_Test3.fseq		1224144
17:03:27.614 -> /FPP_Multisync_Test4.fseq		1224144
17:03:27.614 -> /FPP_Multisync_Test5.fseq		1224144
17:03:27.647 -> /FPP_Multisync_Test6.fseq		1224144
17:03:27.647 -> /FPP_Multisync_Test7.fseq		1224144
17:03:27.647 -> /FPP_Multisync_Test8.fseq		1224144
17:03:27.680 -> /FPP_Multisync_Test9.fseq		1224144
17:03:27.680 -> /Good_Test_Sequence.fseq		4872
17:03:27.680 -> /WGA_Crawl.fseq		4872
17:03:27.713 -> Configuration file: '/config.json' loaded.
17:03:27.713 -> ** OutputDisabled Initialization for Chan: 0 **
17:03:27.713 -> ** OutputDisabled Initialization for Chan: 1 **
17:03:27.713 -> ** OutputDisabled Initialization for Chan: 2 **
17:03:27.747 -> Configuration file: '/output_config.json' loaded.
17:03:27.747 -> ** WS281x Initialization for Chan: 0 **
17:03:27.747 -> ** WS281x Initialization for Chan: 1 **
17:03:27.747 -> 
17:03:27.747 -> WiFi Entering State: Connecting Using Config Credentials
17:03:27.848 -> 
17:03:27.848 -> WiFi Connecting to Pixel-Lab as esp-D32pro-1
17:03:27.848 -> ** 'Disabled' Initialization for input: '0' **
17:03:27.848 -> ** 'Disabled' Initialization for input: '1' **
17:03:27.880 -> Configuration file: '/input_config.json' loaded.
17:03:27.880 -> ** 'E1.31' Initialization for input: '0' **
17:03:27.880 -> - Web Server started on port 80
17:03:27.880 -> FPPDiscovery subscribed to broadcast
17:03:27.880 -> FPPDiscovery subscribed to multicast: 239.70.80.80
17:03:27.880 -> WiFi Reset has been requested
17:03:28.751 -> 
17:03:28.751 -> WiFi Entering State: Connecting Using Config Credentials
17:03:29.085 -> 
17:03:29.085 -> WiFi Connecting to Pixel-Lab as esp-D32pro-1
17:03:30.217 -> .
17:03:31.483 -> WiFi Entering State: Connected To AP
17:03:31.483 -> WiFi Connected with DHCP
17:03:31.483 -> WiFi Connected with IP: 192.168.50.17
17:03:31.583 -> E1.31 Multicast Enabled.
17:03:31.682 -> E1.31 Unicast Enabled on port: 5568
17:03:31.682 -> Listening for 96 channels from Universe 1 to 1.

MartinMueller2003, 2021-01-04:
Missing code to add the "/" in front of the filename to be deleted. This makes the name incorrect and no file gets deleted.

MartinMueller2003, 2021-01-04:
Code has been uploaded and a pull request generated.

patdelaney, 2021-01-07:
Tested on ESP32 - OK
Tested Wemos D1 - OK

Closing

closed 2021-01-07 by patdelaney