forkineye/ESPixelStick #278: Unify Branch: Jan 6 2021 - 18:58:02  After flashing the ESP 32 Even though it connects to Wireless, SSID displays as null

patdelaney, 2021-01-07:
**ESPixelStick Firmware Version**
Jan 6 2021 - 18:58:02 Code pulled today 1/6/2021

**Hardware Version**
ESP32 Lolin Pro

**Binary release or compiled yourself?**
Compiled myself

**Describe the bug**
After compiling the code and uploading. The Wifi page show null for SSID even though it has connected.
19:10:15.589 -> rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
19:10:15.623 -> configsip: 0, SPIWP:0xee
19:10:15.623 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
19:10:15.623 -> mode:DIO, clock div:1
19:10:15.623 -> load:0x3fff0018,len:4
19:10:15.623 -> load:0x3fff001c,len:1044
19:10:15.623 -> load:0x40078000,len:8896
19:10:15.623 -> load:0x40080400,len:5816
19:10:15.623 -> entry 0x400806ac
19:10:15.999 -> 
19:10:15.999 -> ESPixelStick v4.0-dev (NOT STABLE)(Jan  6 2021 - 18:58:02)
19:10:15.999 -> ESP Version: v3.2.3-14-gd3e562907
19:10:16.033 -> Flash File system initialized.
19:10:16.033 -> Listing directory: /
19:10:16.033 -> '/config.json': 	'273'
19:10:16.033 -> '/input_config.json': 	'1296'
19:10:16.033 -> '/output_config.json': 	'1752'
19:10:16.033 -> Listing directory: /www
19:10:16.068 -> '/www/alexa.html.gz': 	'134'
19:10:16.068 -> '/www/ddp.html.gz': 	'130'
19:10:16.103 -> '/www/disabled.html.gz': 	'130'
19:10:16.103 -> '/www/dmx.html.gz': 	'374'
19:10:16.103 -> '/www/e1_31.html.gz': 	'393'
19:10:16.103 -> '/www/effects.html.gz': 	'579'
19:10:16.137 -> '/www/esps.css.gz': 	'13502'
19:10:16.137 -> '/www/esps.js.gz': 	'56173'
19:10:16.137 -> '/www/fpp_remote.html.gz': 	'391'
19:10:16.172 -> '/www/gece.html.gz': 	'425'
19:10:16.172 -> '/www/index.html.gz': 	'2790'
19:10:16.172 -> '/www/mqtt.html.gz': 	'714'
19:10:16.172 -> '/www/relay.html.gz': 	'483'
19:10:16.207 -> '/www/renard.html.gz': 	'439'
19:10:16.207 -> '/www/serial.html.gz': 	'544'
19:10:16.207 -> '/www/ws2811.html.gz': 	'1405'
19:10:16.207 -> SD Card Size: 7594MB
19:10:16.207 -> /System Volume Information/
19:10:16.242 -> 	/System Volume Information/WPSettings.dat		12
19:10:16.242 -> 	/System Volume Information/IndexerVolumeGuid		76
19:10:16.242 -> Configuration file: '/config.json' loaded.
19:10:16.242 -> ** OutputDisabled Initialization for Chan: 0 **
19:10:16.242 -> ** OutputDisabled Initialization for Chan: 1 **
19:10:16.276 -> ** OutputDisabled Initialization for Chan: 2 **
19:10:16.311 -> Configuration file: '/output_config.json' loaded.
19:10:16.311 -> 
19:10:16.311 -> WiFi Entering State: Connecting Using Default Credentials
19:10:16.378 -> 
19:10:16.378 -> WiFi Connecting to 'Pixel-Lab' as esps-40708C962BC8
19:10:16.378 -> ** 'Disabled' Initialization for input: '0' **
19:10:16.378 -> ** 'Disabled' Initialization for input: '1' **
19:10:16.412 -> Configuration file: '/input_config.json' loaded.
19:10:16.412 -> - Web Server started on port 80
19:10:16.412 -> FPPDiscovery subscribed to broadcast
19:10:16.412 -> FPPDiscovery subscribed to multicast: 239.70.80.80
19:10:16.412 -> WiFi Reset has been requested
19:10:17.303 -> 
19:10:17.303 -> WiFi Entering State: Connecting Using Default Credentials
19:10:17.646 -> 
19:10:17.646 -> WiFi Connecting to 'Pixel-Lab' as esps-40708C962BC8
19:10:18.770 -> .
19:10:20.002 -> WiFi Entering State: Connected To AP
19:10:20.002 -> WiFi Connected with DHCP
19:10:20.035 -> WiFi Connected with IP: 192.168.50.17
19:10:23.957 -> * WS Connect - 1

[image]

MartinMueller2003, 2021-01-07:
This is an artifact of a memset that destroyed the String objects in the config. This is also the same issue that kept the DHCP true default value from working properly. Fix is on my branch and will be sent to Shelby soon.

patdelaney, 2021-02-08:
This has been fixed and tested

closed 2021-02-08 by patdelaney