forkineye/ESPixelStick #280: Stand alone playlist mode

patdelaney, 2021-01-07:
**Feature Request**
As an xLights user, I would like the ability to create a series of sequences in xLights and then play them on a SD card version of pixelstick in a stand alone mode.  (note this may exist and I just don't know it)

We already have the ability in Unify to get the sequences over to the pixelstick, we can manually play them from the pixelstick UI, but I would like the ability to upload a playlist and have it just cycle trough the fseqs.

**Proposed idea** 
Upload a playlist.txt file that just lists the fseqs line by line.
The software sees the list and just runs each fseq in order.

**Keywords**
We have keywords as the last line of the list.
Stop - plays the list once and stops until next time it is triggered
Loop - plays the list over and over
Loop N - Play the list N times and then stops

**Triggers**
Provide a way to trigger the playlist via the pixelstick UI on demand
Provide a way to trigger the playlist via GPIO
Provide a way to trigger the playlist by default at boot up

MartinMueller2003, 2021-01-07:
Sounds good to me. Final decision is up to Shelby. There is an issue with multiple open files that would need to be resolved but it should not be too difficult.

Kamal-Sonani, 2021-01-08:
Sounds good, 

What about this >> 

Send rendered data to sd card as per assigned universe , channel & fps.

Stored in sd card , & then play that data with artnet timecode.

By doing that we can reduce data streaming over wifi & render process time for espixelstick.

And even can controll as standalone player.

MartinMueller2003, 2021-01-08:
What we have right now is this:

   
   - Render an fseq with just the channels the ESP needs to play. 
   
   
   - Both Vixen and xLights do this today.
   
   - Load the fseq file onto the SD card on the ESP. 
   
   
   - To get the files on the ESP you can:
   
   - use the ESP UI
   - install a preloaded SD into the ESP
   - let xLights upload the file.

We currently have two ways of playing the fseq file:   
   - Tell the ESP to play the file in an endless loop
   - Tell the ESP to be an FPP Remote and play a file that the FPP master is asking for.

The request I am seeing is to enhance method #1 to allow a playlist to be executed N times where N can be 1 to a large number or "forever". 
FYI: The ESP has no concept of global time so adding a scheduler would require adding an RTC and time management functionality. 
Martin

forkineye, 2021-01-08:
I have a bigger idea which encompasses this as well as the stand alone "effects" engine, whereas the effect can be what we have now, a fseq, or a playlist of fseqs.  These would be triggered via the web interface, configured to start at boot, or started via mqtt. The playlist should just be a simple list of sequences and "loop / repeat" set via the effects web interface or mqtt.

MartinMueller2003, 2021-01-16:
Going to lay out a json format to accommodate this functionality.  Also going to make the playlist files anyname.pl Then we can have multiple playlists and I will configure one of them to play just like we play the file today.

MartinMueller2003, 2021-01-28:
GPIO trigger has not been implemented. UI trigger is present and functioning. Supports Files and Effects.

MartinMueller2003, 2021-01-28:
Not sure a GPIO trigger would be useful. I suggest a separate enhancement request to start a playlist of file based on GPIO trigger. FYI: Not hard to do now that the player classes are set up.

MartinMueller2003, 2021-02-13:
Just saw that I am missing the mqtt trigger. I will add that next.

MartinMueller2003, 2021-03-04:
MQTT now has the ability to trigger the playing of an fseq file. Max replay count is 4 million iterations (32 bit integer)

patdelaney, 2021-04-05:
Stand-alone playlist mode is working. I can't Test MQTT at this time. Playlist functionality is there. If there are issues with the triggering mechanisms a separate bug can be filed

closed 2021-04-05 by patdelaney