forkineye/ESPixelStick #281: FPP discovery is reporting the wrong status when in Remote Mode

patdelaney, 2021-01-07:
**ESPixelStick Firmware Version**
ESPixelStick_TRAVIS-20210106235610.zip

**Hardware Version**
Wemos D1

**Binary release or compiled yourself?**
Travis Build

**Describe the bug**
Something changed with what Pixelstick reports back in in a FPP discovery call. It worked correctly in 
ESPixelStick_TRAVIS-20201224200715.zip

After talking with Dan the original code was write to return bridge mode or remote mode based on whether there was an SD card present. He said at the time of writing the code. 
Based on that description when the pixelstick is on E1.31 mode it reports back correctly based on the existence of the SD card. But if you change to FPP Remote mode, it reports back as a bridge with or without and SD card present.

It needs to report back as bridge in order for FPP to see the device as a remote.

[image]

MartinMueller2003, 2021-01-13:
Fixed issue in the status response that returned a string when an enum was expected. This caused odd mode reporting including "unknown" mode.

Fixed detection of FPP pings that was broken when the ping requests were seperated due to an endless loop of pings. FPP  sends a specific ping subtype. The remotes need to respond with an alternative sub type that will not trigger the other remote devices to generate pings. The correct fix should have been to send pings from the ESP using the remote device ping sub type.

Fixed remote vs bridge reporting to be aware of FPP settings in the config as well as file selection in the config. Local autoplay still trumps FPP master.

Added ping generation when the WiFi interface declares itself operational. This triggers the Master to talk to us. This fixes a race condition with the initial ping trying to go out before the WiFi is up.

patdelaney, 2021-01-18:
appears to be working now off martins branch. Waiting for it to get merged in to Unify and then I will retest

declan2511, 2021-01-23:
Testing as requested

ESPixelStick Firmware Version
ESPixelStick_TRAVIS-20210118114355

Hardware Version
Lolin Wemos D1 mini

Binary release or compiled yourself?
Both

Finally got this working after many hours - had to compile it to get it working (I'm a noob)

FPP reporting correctly,  although there where several reboots when playing around with config.  Most obvious issue is FPP Remote causing a crash at the end of a .fseq file.   It didn't seem to reboot when played in a loop or manually stopped.  I was only playing a 30 second sequence on 30 WS2811.

FPP remote version 4.5,  sequence uncompressed.

Below is the output

Exception (0):
epc1=0x4000e25d epc2=0x00000000 epc3=0x00000000 excvaddr=0x4002d4a1 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffdc0 end: 3fffffc0 offset: 0190
3fffff50:  3fff0c4c 00000258 3fff1728 40217c2f
3fffff60:  3fff0c4c 3fff0c48 3fff0c40 402119f6
3fffff70:  3fff0a80 3fff1828 3fff0a80 40213a18
3fffff80:  3fff0a80 3fff1828 3ffefb71 4020743e
3fffff90:  3fffdad0 00000000 3fff1938 3fff1978
3fffffa0:  3fffdad0 00000000 3fff1938 4022f6f0
3fffffb0:  feefeffe feefeffe 3ffe85a8 40101025
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 3584, room 16
tail 0
chksum 0xb0
csum 0xb0
v5d3af165
~ld

ESPixelStick v4.0-dev (NOT STABLE)(Jan 23 2021 - 15:15:51)
ESP Version: SDK:2.2.2-dev(38a443e)/Core:2.7.2-7-g5d3af165=20702007/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be
Flash File system initialized.
Listing directory: /
'config.json':  '284'
'input_config.json':    '1262'
'output_config.json':   '1283'
Listing directory: www
'alexa.html.gz':        '134'
'ddp.html.gz':  '130'
'disabled.html.gz':     '130'
'dmx.html.gz':  '374'
'e1_31.html.gz':        '393'
'effects.html.gz':      '579'
'esps.css.gz':  '13698'
'esps.js.gz':   '57745'
'fpp_remote.html.gz':   '391'
'gece.html.gz':         '425'
'index.html.gz':        '2655'
'mqtt.html.gz':         '714'
'relay.html.gz':        '483'
'renard.html.gz':       '439'
'serial.html.gz':       '544'
'ws2811.html.gz':       '1405'
SD Card Size: 14815MB (0MB used)
                                System Volume Information - 0
                                                             teststring.fseq - 3980
                                                                                   teststring3.fseq - 614468
                                                                                                            teststring2.fseq - 108068
                                                                                                                                     Configuration file: '/config.json' loaded.
** OutputDisabled Initialization for Chan: 0 **
** OutputDisabled Initialization for Chan: 1 **
Configuration file: '/output_config.json' loaded.
OutputMgr: Incorrect Version found. Using existing/default config.
** WS281x Initialization for Chan: 0 **

WiFi Entering State: Connecting Using Config Credentials

WiFi Connecting to 'Slooownet' as esps-15145d
** 'Disabled' Initialization for input: '0' **
** 'Disabled' Initialization for input: '1' **
Configuration file: '/input_config.json' loaded.
InputMgr: Incorrect Version found. Using existing/default config.
** 'FPP Remote' Initialization for Input: '0' **
- Web Server started on port 80
FPPDiscovery subscribed to broadcast
FPPDiscovery subscribed to multicast: 239.70.80.80
WiFi Reset has been requested

WiFi Entering State: Connecting Using Config Credentials

WiFi Connecting to 'Slooownet' as esps-15145d

WiFi Entering State: Connected To AP
WiFi Connected with DHCP
WiFi Connected with IP: 192.168.0.147

Thanks, good work though,  it becoming more stable.

MartinMueller2003, 2021-01-23:
Issue found. Race condition between interrupt level PDU notifications and background output polling. 
Martin

MartinMueller2003, 2021-01-24:
fixes are on my branch.

 
  On Sat, Jan 23, 2021 at 16:12, Martin Mueller<martinmueller2003@yahoo.com> wrote:    Issue found. Race condition between interrupt level PDU notifications and background output polling. 
Martin

declan2511, 2021-01-24:
Update on testing - generally stable using latest release

**ESPixelStick Firmware Version -**
ESPixelStick_TRAVIS-20210124013315

**Hardware Version**
Lolin Wemos D1 mini

**Binary release or compiled yourself?**
Using binary via flash tool -worked 1st time 

FPP remote function working well,  reporting correctly in FPP, no sign of the previous issues.  Ran various test sequences for two hours.

The device only crashed three times,  only when navigating the web interface  -  moving from the Device set up page to the  Wifi Setup page.  Only occurred after switch on or reboot (via admin function).  Sequences were running for crashes 1 & 2,  no changes were being made to the setup - just navigating.

Two reboots shortly after start-up:

**1st crash after startup** 
Exception (9):
epc1=0x4023590e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000092 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffe960 end: 3fffffb0 offset: 0190
3fffeaf0:  00000000 3fff641c 00000020 40100d73
3fffeb00:  3fff640c 402741c4 3fff557c 4022f4ee
3fffeb10:  3fffebb8 3fffebac 3fffebb8 4023062d
3fffeb20:  3fff551c 3fff557c 3fffeb40 40207b95
3fffeb30:  00000000 3fff641c 3fffebb8 402099cb
3fffeb40:  40236a98 00000000 000003e8 00035195
3fffeb50:  00000000 00000000 00000000 00000000
3fffeb60:  3fff1808 40236a98 00000000 000003e8
3fffeb70:  00000483 3fff39fc 3fff3cac 3fff549c
3fffeb80:  3fff54b4 3fff1808 3fffebdc 3fff5c2c
3fffeb90:  3fff5c32 3fff640c 3fff642c 00000100
3fffeba0:  3fff641c 3fff640c 3fffeba0 3fff5500
3fffebb0:  0019001f 80000001 3fff5500 0019001f
3fffebc0:  802741e6 3ffefd64 3fffec10 4020a395
3fffebd0:  3fffec00 3ffefd64 3fffec88 3fff83c8
3fffebe0:  3ffe8e58 3ffefd64 00000000 3fff83c8
3fffebf0:  3ffe8e58 3ffefd64 3fffec88 4020cbf9
3fffec00:  40274100 3fff83c8 80ffed18 3ffe8e58
3fffec10:  3ffe8e63 3fff83c8 3fffed18 3ffe8e58
3fffec20:  3ffefd2c 3ffefd64 3fffec88 4020d6b8
3fffec30:  00005080 00000001 3fffecb0 0000000a
3fffec40:  00000000 00000001 3fffecb0 4020c4cd
3fffec50:  3fff568c 000013fc 3fff6fdc 4020a395
3fffec60:  00000002 3fffed2c 3fffed14 3ffefd2c
3fffec70:  3fff568c 3fffed2c 3fffed14 4020d7dd
3fffec80:  3fff83c8 3fffed18 3fff83c8 3fffed18
3fffec90:  00000000 40251da7 3ffeefd0 3ffefd64
3fffeca0:  3fff568c 3ffefd2c 00000000 4020da0d
3fffecb0:  3fffed18 3fff6fe4 00000006 000013d4
3fffecc0:  3fff2001 3ffefd7b 0000007d 3fffed18
3fffecd0:  40258298 40242a15 3fff3e0c 3fff1a90
3fffece0:  40254855 00000030 0000000f ffffffff
3fffecf0:  3ffeed64 3fffee80 3ffee8a4 00000000
3fffed00:  3fff1e7c 3fff1e44 3fff5c2c 40242ca2
3fffed10:  00000000 3fff1e44 3fff6fdc 3fff6fea
3fffed20:  3fff83b8 3fff83d8 00000000 3fff83c8
3fffed30:  3fff83c8 00000020 3fffed18 4024a8d8
3fffed40:  000034c0 00000001 3fffed18 4024a8d8
3fffed50:  000034c0 00000001 0000015a 4000050c
3fffed60:  00000000 3fff5c2c 00000698 40100ad4
3fffed70:  00000000 000000d4 00000020 3fff5c2c
3fffed80:  000002c8 00000059 00000059 3fff568c
3fffed90:  3fff5468 3fff568c 00000004 4020da94
3fffeda0:  00000017 00000000 4bc6a7f0 00000000
3fffedb0:  00001df8 000003bf 40100582 40235005
[... 663 lines cut ...]
3ffffe60:  0000005a 3fff0c7c 3fff1994 00000000
3ffffe70:  3fff4694 00000020 3fffff20 402306b2
3ffffe80:  0000005a 3fff0dde 00000004 ffffffff
3ffffe90:  0000000b 3fff206c 40211bdc 00048944
3ffffea0:  3fff0d84 0000005a 3fff0d84 3fff1994
3ffffeb0:  3fff4694 00000010 3fffff20 402307c3
3ffffec0:  00000010 4bc6a7f0 0000da3b 3fff1994
3ffffed0:  40276360 00000010 3fffff20 402307ed
3ffffee0:  3fff4694 3fff1794 3fffff20 402309f0
3ffffef0:  3fff0c7c 0000005a 3fffff20 40230a20
3fffff00:  3fff1794 3fff17a8 3fff1758 40212025
3fffff10:  3fff0d84 0000005a 3fff1758 4021a041
3fffff20:  3fff4600 0010001f 8000372b 656c6469
3fffff30:  00000000 84c6a7f0 0000da3c 3fff1994
3fffff40:  00000000 00000000 3fff1758 4021b351
3fffff50:  3fff0d84 4021cb94 40100582 789374bc
3fffff60:  00000000 3fff0c78 3fff206c 40211c16
3fffff70:  3fff0ab0 3fff1844 3fff0c70 40216428
3fffff80:  3fff0ab0 3fff1844 3ffefba1 4020743e
3fffff90:  3fffdad0 00000000 3fff1954 3fff1994
3fffffa0:  3fffdad0 00000000 3fff1954 40231c28
3fffffb0:  feefeffe feefeffe 3ffe85a8 40101041
<<<stack<<<

Thanks again for the continued development  - now to see if I can getting it working on my ESP32

patdelaney, 2021-01-24:
**FPP Status message are not reporting correctly.**
"Version": "4.0-dev (NOT STABLE):Jan 18 2021 - 12:21:36"  (precompiled version)
Wemos D1 with SD card and a sequence on the card

Fresh install run discovery call and the unit reports back as:
{
    "MQTT": {
        "configured": false,
        "connected": false
    },
    "current_playlist": {
        "count": "0",
        "description": "",
        "index": "0",
        "playlist": "",
        "type": ""
    },
    "volume": 70,
    "media_filename": "",
    "fppd": "running",
    "current_song": "",
    "SyncCount": 0,
    "SyncAdjustmentCount": 0,
    "current_sequence": "",
    "playlist": "",
    "seconds_elapsed": "0",
    "seconds_played": "0",
    "seconds_remaining": "0",
    "sequence_filename": "",
    "time_elapsed": "00:00",
    "time_remaining": "00:00",
    **"status": 1,**
    **"status_name": "playing",**
    "mode": 1,
    "mode_name": "bridge",
    "advancedView": {
        "HostName": "shelby-sd1",
        "HostDescription": "ESPixelStick",
        "Platform": "ESPixelStick",
        "Variant": "ESPixelStick-ESP8266",
        "Mode": 1,
        "Version": "4.0-dev (NOT STABLE):Jan 24 2021 - 01:34:27",
        "majorVersion": 4,
        "minorVersion": 0,
        "typeId": 194,
        "Utilization": {
            "MemoryFree": 25576,
            "Uptime": 362325
        },
        "rssi": -70,
        "IPS": [
            "192.168.1.206"
        ]
    }
}

It should not be reporting as playing and status 1, it should be status 0 and idle

When in FPP remote mode it reports back as:
{
    "MQTT": {
        "configured": false,
        "connected": false
    },
    "current_playlist": {
        "count": "0",
        "description": "",
        "index": "0",
        "playlist": "",
        "type": ""
    },
    "volume": 70,
    "media_filename": "",
    "fppd": "running",
    "current_song": "",
    "SyncCount": 0,
    "SyncAdjustmentCount": 0,
    "current_sequence": "",
    "playlist": "",
    "seconds_elapsed": "0",
    "seconds_played": "0",
    "seconds_remaining": "0",
    "sequence_filename": "",
    "time_elapsed": "00:00",
    "time_remaining": "00:00",
    **"status": 1,
    "status_name": "playing",**
    **"mode": 1,
    "mode_name": "bridge",**
    "advancedView": {
        "HostName": "shelby-sd1",
        "HostDescription": "ESPixelStick",
        "Platform": "ESPixelStick",
        "Variant": "ESPixelStick-ESP8266",
        "Mode": 8,
        "Version": "4.0-dev (NOT STABLE):Jan 24 2021 - 01:34:27",
        "majorVersion": 4,
        "minorVersion": 0,
        "typeId": 194,
        "Utilization": {
            "MemoryFree": 23064,
            "Uptime": 49535
        },
        "rssi": -70,
        "IPS": [
            "192.168.1.206"
        ]
    }
}

The status returned is wrong and it says it's still in bridge mode.

patdelaney, 2021-01-24:
@declan2511 feel free to reach out to me if you have questions about my last comment. pat_delaney at colonydrive dot com

patdelaney, 2021-01-25:
Status message that is returned in not correct, it's returning as remote all the time
Jan 25 2021 - 15:51:26  compiled and travis build

patdelaney, 2021-02-08:
This has been fixed
   "mode_name": "remote",

closed 2021-02-08 by patdelaney