forkineye/ESPixelStick #284: no longer fits on ESP-01 modules

perigalacticon, 2021-01-15:
--------- Instructions - Delete this ---------
Please provide answers directly below each section.
--------- Instructions - Delete to here ---------
**ESPixelStick Firmware Version**
v3.0

**Hardware Version**
ESP-01 1MB

**Binary release or compiled yourself?**

**Operating System**

**Access Point**

**Describe the bug**

When compiled for "Generic ESP8266" option within the Arduino IDE, the firmware no longer fits on the ESP-01 modules with 1MB Flash.  It won't run after uploading the SPIFFS files.  Same firmware files does upload to WEMOS 4 MB modules.

I didn't change anything about the files, it just started not working on ESP-01s one day, I think something may have changed in the compiler so it's a little bit bigger now?

MartinMueller2003, 2021-02-08:
The V3 is shipped with an ESP12f style module. With all of the additional features, it is no surprise the code no longer fits in the 01.

forkineye, 2021-02-08:
The binary releases of V3.0 will still fit on the 1MB ESP-01 modules.  If you're building it yourself, reference the travis yml file for Arduino build options and verify you have the proper flash size selected. Unify / 4.0 however will not fit on the 1MB modules.

closed 2021-02-08 by forkineye