forkineye/ESPixelStick #293: ESP8266 cant entering in AP mode if no connection

1Bertomc, 2021-01-31:
Hi,

I want build some controllers with Wemos D1 Mini, and ALL cant enter in AP mode.
For example, if controller cant find the wifi network, controller cant create a Access Point.

i already have this problem with some esp01 models from different manufactures

How i can do?

Thank you so much :)

MartinMueller2003, 2021-01-31:
WHat version of Software are you building?

1Bertomc, 2021-02-01:
> WHat version of Software are you building? 

Hi :) Thanks for your repply.
i tested 3.1, 3.2 and some 2.X versions...
All versions works fine if sucessuful connection. But if i need connect to another wifi network, i only can do it if i flash firmware with new ssid and passphrase...

MartinMueller2003, 2021-02-01:
Did you enable AP Fallback? It is disabled by default. Set it up to connect to your network, change the setting to allow AP fallback mode and then move to your new environment.
Martin

1Bertomc, 2021-02-01:
> Did you enable AP Fallback? It is disabled by default. Set it up to connect to your network, change the setting to allow AP fallback mode and then move to your new environment.

Yes, its enable. But, when enable, if connection fails accidentaly, will clear wifi configs. its normal?

Because, i think when I used the ESP-01 (Not all modules, only not chinese versions), if connection fails, AP mode was initialized but keeping wifi config, and if I disconnected and reconnected the power, the module tried to connect to the same network again...

MartinMueller2003, 2021-02-01:
Unify branch keeps the config. The V2 and V3 branches did not (I considered that a bug).
Martin

MartinMueller2003, 2021-02-08:
I dont think there is anything more to do on this iss and would like to close the thread.

closed 2021-02-08 by forkineye