forkineye/ESPixelStick #296: Unify: Brightness selector shows a warning if I select .05 for brightness

patdelaney, 2021-02-05:
**ESPixelStick Firmware Version**
Jan 27 2021 - 10:43:05

**Hardware Version**
Wemos D1

**Binary release or compiled yourself?**
not sure

**Describe the bug**
The web ui in Unify branch show a warning if I change the brightness to .05. I can save the value and it does lower the brightness. For testing I don't need the pixels to be full on, I'm ok with a value of .05 just don't flag it as an error in the UI.

[image]

MartinMueller2003, 2021-02-05:
The minimum value is 0.1. That means your 0.05 value is too small.
                <input type="number" class="form-control is-valid" id="brightness" step="0.01" min="0.1" max="1.0" value="1.0" title="Maximum brightness is 1.0." onchange="refreshPixel()">

Do we want to change the definition?
Martin

patdelaney, 2021-02-05:
But .05 does lower the actual output whether the UI likes it or not :-)

forkineye, 2021-02-05:
We need to change this to a percentage (1-100%) instead of the decimal value that I originally coded it as.  Have had a few users confused about it.

MartinMueller2003, 2021-02-05:
ok I will change it.
Martin

MartinMueller2003, 2021-02-05:
Pull request created.

Martin
    On Friday, February 5, 2021, 11:05:04 AM EST, Martin Mueller <martinmueller2003@yahoo.com> wrote:  
 
  ok I will change it.
Martin

MartinMueller2003, 2021-02-08:
This has been changed to a value 1 - 100 and has been integrated into the Unify branch.

patdelaney, 2021-02-08:
Even though the UI was switch to be 1-100%   I can still put in .5
and click save it
[image]

MartinMueller2003, 2021-02-11:
Added code to disable the save button when a field has an invalid value.

patdelaney, 2021-03-07:
Fixed

closed 2021-03-07 by patdelaney