forkineye/ESPixelStick #302: Im lost about pinout what pin to use...

processor83, 2021-02-16:
I need some help about esp32 pinout.

i looked everywhere in the code but found nothing but now i suspect its because it is uart1 and uart2 tx pin and it connat be changed so its not defined anywhere ?

is it GPIO17 and GPIO09??

so io17 and sd2 on those wemos d32 mini ?

and on esp8266 it is GPIO02(D4)

Thank you for you help.

MartinMueller2003, 2021-02-16:
Since this information is shared by multiple output types, it is found in common.hpp. However in the esp32 these are "well known" names so they are not defined explicitly. 
typedef enum
{    GPIO_NUM_2  = 2,    GPIO_NUM_10 = 10,    GPIO_NUM_13 = 13,
} gpio_num_t;
typedef enum{    UART_NUM_0 = 0,    UART_NUM_1,    UART_NUM_2} uart_port_t;
Then in OutputCommon.cpp you get a table that tells the code how to map the IO for each of the three possible output paths:
static const OutputChannelIdToGpioAndPortEntry_t OutputChannelIdToGpioAndPort[] ={    {gpio_num_t::GPIO_NUM_2,  uart_port_t::UART_NUM_1},#ifdef ARDUINO_ARCH_ESP32    {gpio_num_t::GPIO_NUM_13, uart_port_t::UART_NUM_2},#endif // def ARDUINO_ARCH_ESP32    {gpio_num_t::GPIO_NUM_10, uart_port_t (-1)},};
The code uses these values. It is trivial to make them configurable and you can easily change the in this table.
Martin

processor83, 2021-02-16:
I don't know why I cannot understand this..   on esp32 it use gpio02 and gpio10 or 10 and 13?

MartinMueller2003, 2021-02-23:
This is the current mapping and it CAN be changed:

gpio_num_t::GPIO_NUM_2,   uart_port_t::UART_NUM_1
gpio_num_t::GPIO_NUM_13, uart_port_t::UART_NUM_2

With that said, I cannot predict how the module manufacturer connects the GPIOs to module pins. That is different for each module.

processor83, 2021-02-25:
I'm actually designing my module with sd card and optional network. I will wait officiel design or maintain a fork with my pinout. I'm not super good at programming that why I'm lost! So thi tell which user is used on a gpio?

Thank you for taking time to help me with this.  I will have my esp32 and lan8720 modules soon and will be able to test the firmware and the Ethernet fork.

MartinMueller2003, 2021-02-25:
Would you like a call on skype or slack?
Martin

processor83, 2021-02-25:
That would be nice but my English is not so good I will understand you but I'm not sure you will understand me haha! Let me know how to contact you and I will be more then happy to talk with you!

MartinMueller2003, 2021-02-25:
skype martin.mueller.brainware
i am pretty good at understanding broken english.
martin

closed 2021-02-25 by processor83