forkineye/ESPixelStick #307: SD card not detected with esp32

processor83, 2021-03-08:
**ESPixelStick Firmware Version**

FW Version
4.0-dev (NOT STABLE)

Build Date
Mar 7 2021 - 21:04:28

**Hardware Version**

ESP32 
MH-ET LIVE
LOLIN uSD card shield
SD card isd on 
sck:GPIO18
miso:GPIO19
mosi:GPIO23
cs:GPIO5

**Binary release or compiled yourself?**

Yes

**Operating System**
Windows 10 64
**Access Point**
not a wifi problem but
tp-link
**Describe the bug**

SD card is not detected.

This hardware setup work with the example code.
This code work on esp8266 with same sd card also.

So i really dont know where to look or what i can do to give more details.

processor83, 2021-03-08:
i was testing lots of thing and i editted the gpio in the source and it worked but was not working when changing pins via webinterface  i need to do more testing ont his one!   afther about 10 hours on this!

processor83, 2021-03-08:
My se module is a lolin clone and is not using same pin for CS so I was using wrong pin!

patdelaney, 2021-03-12:
The onboard SD card is seen on a Lolin32 Pro

closed 2021-03-12 by patdelaney