forkineye/ESPixelStick #309: Unify 4.0 DMX output appears to be broken

patdelaney, 2021-03-12:
**ESPixelStick Firmware Version**
Build Date Mar 12 2021 - 12:10:39

**Hardware Version**
ESP 8266 on Shelby's V3 module

**Binary release or compiled yourself?**
Release Build Date Mar 12 2021 - 12:10:39

**Describe the bug**
I am trying to test DMX Output with a DMX moving head. I first installed Version 3.2 to test against the moving head to make sure I had the cabling correct. (I am just sung 2 wires DMX+ and DMX-) After boot up I saw the the module was outputting a DMX signal because the moving head has an red led flashing indicator when DMS signal is present. So that proves that Cableing is OK and I can control the moving head with 3.2.

I reflashed the PixelStick with today's build of Unify. I am not seeing any DMX activity on the moving head. If I click the reset button on the esp8266 and then the DMX indicator flashes for a bit and the head moves and then resets, then the DMX stops.

Please note I have never tested DMX before so I don't know if this is a regression, or an existing issue.

The youtube links show the dmx light activity with the build from a few days ago. IT runs for a bit and shuts off. I don;t see any warnings in the IDE console window.
https://youtu.be/TyxzZWpi8Yc

MartinMueller2003, 2021-03-13:
It looks like UART interrupts stop happening.

MartinMueller2003, 2021-03-22:
The "Stopping" issue has been resolved. Still having an issue with "not all channels appear to be working properly". Waiting for test equipment so that I can see what is happening.

patdelaney, 2021-03-25:
ESP 32 has been working, we just have the issue with 8266. First channel is 2 and then it's every other channel... 2,4,6,8 ......

MartinMueller2003, 2021-03-28:
Fixed the ESP8266 issue. ready for final testing

patdelaney, 2021-04-05:
This has been resolved, although DMS outputs are working as expected

closed 2021-04-05 by patdelaney