forkineye/ESPixelStick #317: Multiple output streams to WS281X

brmeadows2841, 2021-04-18:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
ESP32 is capable of outputting multiple streams using multiple GPIOS pins.
Is this in the future for ESPixelstick?

**Additional context**
Add any other context or screenshots about the feature request here.

MartinMueller2003, 2021-04-19:
the esp32 unify branch supports two strings of ws281x pixels. was there something more that is needed?
Martin

brmeadows2841, 2021-04-19:
I am running the Unify branch on a ESP32, the second output is for effect only as far as I can tell. Is two outputs a limit of the software? I am asking for more than a few outputs from the firmware, and what is the future of the software? Will it add features that include a greater number of outputs in order to run a light show with a higher number of pixels?

> the esp32 unify branch supports two strings of ws281x pixels. was there something more that is needed? Martin Sent from Yahoo Mail on Android On Sun, Apr 18, 2021 at 19:54, ***@***.***> wrote: Is [...]

MartinMueller2003, 2021-04-19:
The 2nd output is fully featured. You are confusing the alternative input with an output. Inputs (primary and Secondary) deliver data to both ports.
How many additional outputs are you looking for? More outputs would require bit banging and we have been avoiding that.

Martin

brmeadows2841, 2021-04-19:
I see my error, I was using the effects as a test for the output.
I have both outputs in separate streams, thank for the correction.
I guess bit banging is a bad idea. I was hoping for more than a few output streams on a ESP32.

processor83, 2021-04-21:
I'm curious how fastled manage 10 pin??  Also since the driver now use uart what about using. Spi uart addon? Spi max3109?

MartinMueller2003, 2021-04-21:
we use the spi interface for the sd card.

processor83, 2021-05-06:
Did not think about this sd card!i just throw idea at it that come to mind I don't have skill to test or know if it's possible but I know that there is a way to change some pins on the f'y on the esp32 would it be possible to us it has internal multiplexer to split the two uart over 4 pins?? Or use external multiplexers??

MartinMueller2003, 2021-05-06:
The pins can be assigned to anything you want and that is already in the code. It is the SPI Hardware that is being used by the SD card interface. There is no good way to share the SPI hardware. Shelby is looking at another way to get some extra UARTs attached to the esp.
Martin

closed 2021-09-15 by forkineye