forkineye/ESPixelStick #321: Pixelstick 4.0 Web UI does does not work when behind FPP proxy

patdelaney, 2021-05-19:
This may be a dupe of #226 esp avec rasberry et fpp falcomplay 

**ESPixelStick Firmware Version**
Unify Branch
**Hardware Version**
esp8266 and ESP32
**Binary release or compiled yourself?**
compiled myself

**Describe the bug**
Proxy linked page not displaying full webpage of ESPPixelStick

Possible Fix will be similar to a bug in Sandevice 4.x code.
It's not returning a proper HTTP 1.x response (no HTTP headers at all) and thus the Apache reverse proxy has no idea what the response is or the size or anything and thus will only return part of the page. 

You need to configure FPP as a gateway, (similar to this, I can walk you through the setup)
[image]
[image]
[image]
[image]

MartinMueller2003, 2023-12-20:
I have completely restructured the Web UI to no longer use web sockets. I suspect it will work with an FPP proxy now since all traffic is HTTP.

patdelaney, 2025-07-08:
> Can you:
>
>    1. Open a browser
>    2. Put it in debug / dev mode
>    3. Start capturing the network and console data
>    4. Tell the browser to connect to the pixelstick
>    5. Save the logs and post them here.

Is thai what you need>

Pat

MartinMueller2003, 2025-07-08:
Just attach the logs to the ticket.

patdelaney, 2025-07-08:
[Recording 7_7_2025 at 7_39_54 PM.json](https://github.com/user-attachments/files/21112630/Recording.7_7_2025.at.7_39_54.PM.json)

patdelaney, 2025-07-08:
[192.168.1.250.zip](https://github.com/user-attachments/files/21112633/192.168.1.250.zip)

MartinMueller2003, 2025-07-08:
It looks like the content of this request was never delivered from fpp to the browser. Or the request got munged going to the esp.

          "method": "GET",
          "url": "http://192.168.1.250/proxy/192.168.50.19/esps.js",

patdelaney, 2025-07-08:
@MartinMueller2003 let me know when you have the fix posted in the web-flasher and I can test. I have it the scenario set up in the lab.

MartinMueller2003, 2025-07-08:
Getting a bit closer.

- configured FPP to be a hotspot with DHCP turned on and proxy enabled.
- Connected the ESPS to my test network.
- Used the UI to change the WiFi settings to connect to the FPP hotspot.
- checked proxy status on the FPP and the ESPS showed up there using the same IP as reported on the ESPS console.
- clicked the link on the FPP Proxy page (http://192.168.10.91/proxy/192.168.8.97)
-- Nothing loaded. Network trace on the browser indicated a 404 response.
- Tried to access the trivial report to get heap size: "http://192.168.10.91/proxy/192.168.8.97/heap". Get a 404 response.
- Added a console output that fires when the web page gets a request it cannot process. This output never gets triggered.
-- This means the ESPS did not receive an HTTP request.
Is there a tool on the FPP to do a network trace?

patdelaney, 2025-07-09:
Martin - FYI this works on a pixelstick that is flashed with WLED firmware. If you're stuck take a peek at WLED

[image]

MartinMueller2003, 2025-07-09:
Just got it to work. Missed a forwarding/NAT setting on the FPP.

MartinMueller2003, 2025-07-09:
I have a test version called "ESPixelStick_Firmware-4.0-FppProxyNoEsp8266". This only works on ESP32 builds. There is a bug in the ESP8266 that fails to bring up the tcp stack.

MartinMueller2003, 2025-07-10:
Thanks Guys. I will mark this complete once the code gets merged to main.

patdelaney, 2025-07-10:
I'm not sure it's all working. There was a comment a few lines above:

xLights, I'm unable to Upload Outputs. Appears to timeout and "ESPixelStick Input Upload Failed. Unable to Connect".

patdelaney, 2025-07-10:
That likely means there is still more work to be done on the PixelStick side as this functionality  works on other controllers.

MartinMueller2003, 2025-07-11:
I just tried to use xLights to manage an ESPS through an FPP Proxy. The ESPS does not receive ANY PDUs from xLights. I see that it is trying to send the IP address of the ESPS on the FPP subnet. The FPP is not doing proper routing so there is no response to the request. The ESPS does not know if it is behind the proxy so I am not sure what it should be doing.

patdelaney, 2025-07-11:
@MartinMueller2003 sorry  what does the PDU acronym stand for?

MartinMueller2003, 2025-07-11:
PDU = Protocol Data Unit = Packet = TCP start message to establish the connection between xLights and the ESPS.

MartinMueller2003, 2025-07-12:
Was working with the xLights team. Even when properly configured, xLights failed to add the proxy header on the requests going to the ESPS. We did get it to partially work but then my xLights started to crash.

MartinMueller2003, 2025-07-16:
I am still getting nothing but xLights crashes when I try to do an FPP Connect or upload outputs on a proxied ESPS. The ESPS itself is not seeing any data or requests coming over the WiFi interface.

MartinMueller2003, 2025-07-16:
I have the June version running on a PC.

joe8012, 2025-07-16:
Xlights will error out or crash at the 5 minute mark if the upload hasn't finished.

MartinMueller2003, 2025-07-27:
xLights still has a bug that will cause it to crash when a WebSocket test returns the wrong result and the system tries to establish a V3 connection to a V4 ESPS. This was being triggered by an extra set of quotes in the version string sent from the ESPS to xLights.  Proxy functionality does seem to be working from the ESPS UI.

patdelaney, 2025-07-29:
I'm currently at the office, I'll test this when I get home.  Thank you

sirbwolf, 2025-07-29:
> patdelaney

I’ll be able to test it this weekend - when I get back home. 

Thank you!