forkineye/ESPixelStick #322: Power Issue

aarondoonan, 2021-05-26:
I just received my first 2 ESPixelSticks V3, but I'm having difficulty in powering them up. I'm using a Macbook Pro laptop running Catalina and I installed the CH340 drivers manually as well for good measure. However, when I plug in the Wemos module to the computer, the power light only flashes and doesn't stay on. This also means I cannot find it in the flash tool. 

I've tried every microUSB cable I have thinking it was related to a problem with the cables, but each one has the same issue. I'm using the USB ports on the computer for other peripherals so I know those work. Am I missing something here? Any ideas on how to keep it powered up?

onewithhammer, 2021-05-26:
Verify you have the CH340 drivers installed correctly.

I followed these instructions and it worked perfectly on my Macbook Pro laptop.

https://learn.sparkfun.com/tutorials/how-to-install-ch340-drivers/all

Let me know if you need addition help.  We can use zoom to figure out your issue.

MartinMueller2003, 2021-05-26:
The ESP does not have a power on indicator. It has a WiFi activity indicator that also flashes when it sees USB traffic. 
In your USB device list, do you see a com port come and go as you plug in and remove the ESP? Do you have the arduino IDE or a com port terminal application? Turn it on and monitor the com port as you press the ESP reset button. You should see a lot of status messages coming out if the device has been flashed.
Martin

 
 

https://learn.sparkfun.com/tutorials/how-to-install-ch340-drivers/all

aarondoonan, 2021-05-26:
Ok, I didn't pay attention to the indicator much. I'll reinstall my drivers and watch the com port as you suggest. Thanks for the suggestions!

forkineye, 2021-05-26:
@aarondoonan The blue LED one the Wemos module is hooked to the output line. You'll see it turn solid once its flashed and outputting pixel data.  As for drivers on the mac, you shouldn't need to install anything on Catalina.  If you've put extra CH340 drivers on, I'd remove them and use the default drivers.  When launching ESPSFlashTool.jar, you'll have to go through the standard MacOS security exception stuff for unsigned software.  Reach out if you have any issues.

aarondoonan, 2021-05-27:
After following the directions for the drivers, I was able to get everything installed and running! Thanks for everyone's help!

closed 2021-08-11 by forkineye