forkineye/ESPixelStick #327: Missing CSS file

nyckidryan, 2021-07-02:
**ESPixelStick Firmware Version**
4.0 dev

**Hardware Version**
3.0

**Binary release or compiled yourself?**
Compiled from repository

**Operating System**
Windows 10

**Access Point**
Netgear Orbi

**Describe the bug**
Unable to configure using web UI, possibly due to mising linked stylesheet esps.css in index.html.

[screenshot](https://imgur.com/a/lIGFlvU)

nyckidryan, 2021-07-02:
[image]

MartinMueller2003, 2021-07-02:
esps.css is a compiled file created by gulp. This indicates that gulp failed and the files in the ../data/www directory are not complete. Most likely the esps.js file is also missing. If you press F12 and look at the errors on the console tab, you should see a single error indicating favicon could not be loaded (we dont have a favicon but the browser tries to load one by default).

MartinMueller2003, 2021-08-09:
This is an install issue and is not present any any of my testing

closed 2021-08-09 by MartinMueller2003