forkineye/ESPixelStick #358: OLED Support

computergeek1507, 2021-09-25:
Add support for the common 0.96" I2C 128X64 OLED LCD Displays to display IP, Playback status, etc.

jasonsomers, 2023-05-08:
Are there any plans to incorporate this feature? It would be a great addition!

noman1228, 2025-04-22:
If this is still a wanted item, I can clean up my code for both drop-in (VERY basic info shown) or a full-fledged FPP, status, Uploads, Unzips, Cur File, Cur Playlist, Cur time in/out.

jasonsomers, 2025-04-22:
My original thoughts were for just the basic info - such as IP and playback status. 
In the short term - having those would be great!

In the long term maybe a setting in the web interface that allows you to choose that basic display - or a more advanced one with customized layout - allowing them to choose which variables are listed.

MartinMueller2003, 2025-04-22:
Go for it. Keep in mind that there are not a lot of gpios available on things like the Octa with SD and Ethernet.

MartinMueller2003, 2025-04-22:
I think you will need to add a flag to the platform files to control when it is available.

noman1228, 2025-08-01:
It does work. Great searching BTW, It's EXTREMELY basic right now. The simple task of re-drawing the screen takes more memory and clock cycles than I would like. But if you want 4 channels of ws281x, SD Card, with a 128x32 OLED screen showing IP, host and signal strength.. it's working