forkineye/ESPixelStick #360: [4.0.0-Beta1] "Save Changes" always disabled - unable to save config

henrygab, 2021-09-26:
--------- Instructions - Delete this ---------
Please provide answers directly below each section.
--------- Instructions - Delete to here ---------
**ESPixelStick Firmware Version**
4.0.0-Beta1 (7083637f81f53b5b15c9fefd2b28f192c30334b6)

**Hardware Version**
ESPixelStick v3

**Binary release or compiled yourself?**
Binary release

**Operating System**
N/A

**Access Point**
Ubiquiti Nanostation M2 Loco

**Describe the bug**

### Steps to repro:
1. Load 4.0.0-Beta1 firmware onto new D1 mini
2. goto `Device Setup` tab
3. Change `Primary Input` to `DDP` (leave secondary input disabled)
4. Change `Output 1` to `WS2811`
5. Click on `Save Changes`

### Expected results:

The `Save Changes` button should be enabled, and thus clicking on it should save the changes.

### Actual results:

The button is disabled (HTML attribute).  Nothing happens when clicking on it (obviously).  The configuration is not saved.  Changing to other tab and back loses all changes.

### Other bits and pieces

Reviewing the page in Edge's DevTools (F12) shows the following properties that may be of interest for the save button:

* HTMLButtonElement
    * validity: ValidityState
        * badInput: false
        * customError: false
        * patternMismatch: false
        * rangeOverflow: false
        * rangeUnderflow: false
        * stepMismatch: false
        * tooLong: false
        * tooShort: false
        * typeMismatch: false
        * valid: true
        * valueMissing: false
    * willValidate: false

MartinMueller2003, 2021-09-26:
This indicates one or more fields had invalid data in them. Do any of the fields on the device setup screen have a red x in them? A screenshot of the device screen would help a lot.

henrygab, 2021-09-26:
None of the fields is invalid.  Using the built-in debugger of Edge, it appears that the function that checks validity is unhappy with the following elements:

element | reason
-----|-----
miso_pin | rangeUnderflow = true (1)
mosi_pin | rangeUnderflow = true (1)
cs_pin | rangeUnderflow = true (1)

(1) NOTE: These elements should **NOT** be considered in the validity check, as the output type is WS2811, which is single-pin (not SPI).  Moreover, the APA102 is not even provided as an output option.  Thus, I think this is likely a real bug....

NOTE:  This is a 100% clean install.  The d1 Mini came straight from the ESPixelstick v3 package ... I'm not updating the ones on my house (yet), although I am very interested in doing so (due to SD support / xLights / FPP).

<details><summary>boot log</summary><P/>

```
[ESP] ESP Version: SDK:2.2.2-dev(38a443e)/Core:3.0.2=30002000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:6105635
[FileMgr] Flash file system initialized.
[FileMgr] No SD card installed
[FileMgr] Configuration File: '/config.json' loaded.
[ESP] Incorrect Config Version ID
[OutputMgr] *** 'Disabled' Initialization for Output: 0 ***
[FileMgr] ***Configuration File: '/output_config.json' Could not open file for reading.***
[OutputMgr] EEEE Error loading Output Manager Config File. EEEE
[OutputMgr] --- WARNING: Creating a new Output Manager configuration Data set - Start ---
[OutputMgr] *** Shutting Down 'Disabled' on Output: 0 ***
[OutputMgr] *** 'WS2811' Initialization for Output: 0 ***
[OutputMgr] *** Shutting Down 'WS2811' on Output: 0 ***
[OutputMgr] *** 'GECE' Initialization for Output: 0 ***
[OutputMgr] *** Shutting Down 'GECE' on Output: 0 ***
[OutputMgr] *** 'DMX' Initialization for Output: 0 ***
[OutputMgr] *** Shutting Down 'DMX' on Output: 0 ***
[OutputMgr] *** 'Renard' Initialization for Output: 0 ***
[OutputMgr] *** Shutting Down 'Renard' on Output: 0 ***
[OutputMgr] *** 'Serial' Initialization for Output: 0 ***
[OutputMgr] *** Shutting Down 'Serial' on Output: 0 ***
[OutputMgr] *** Unknown output type: '5 for channel '0'. Using disabled. ***
[OutputMgr] *** 'Disabled' Initialization for Output: 0 ***
[OutputMgr] *** Shutting Down 'Disabled' on Output: 0 ***
[OutputMgr] *** Unknown output type: '6 for channel '0'. Using disabled. ***
[OutputMgr] *** 'Disabled' Initialization for Output: 0 ***
[OutputMgr] *** Saved Output Manager Config File. ***
[OutputMgr] --- WARNING: Creating a new Output Manager configuration Data set - Done ---
[FileMgr] ***Configuration File: '/input_config.json' Could not open file for reading.***
[InputMgr] EEEE Error loading Input Manager Config File. EEEE
[InputMgr] --- WARNING: Creating a new Input Manager configuration Data set - Start ---
[InputMgr] *** Shutting Down 'Disabled' on Input: 0 ***
[InputMgr] *** Shutting Down 'Disabled' on Input: 1 ***
[InputMgr] *** Shutting Down 'E1.31' on Input: 0 ***
[InputMgr] *** Shutting Down 'Disabled' on Input: 1 ***
[InputMgr] *** Shutting Down 'Disabled' on Input: 0 ***
[InputMgr] *** Shutting Down 'Effects' on Input: 1 ***
[InputMgr] *** Shutting Down 'Disabled' on Input: 0 ***
[InputMgr] *** Shutting Down 'MQTT' on Input: 1 ***
[InputMgr] *** Shutting Down 'Disabled' on Input: 0 ***
[InputMgr] *** Shutting Down 'Alexa' on Input: 1 ***
[InputMgr] *** Shutting Down 'DDP' on Input: 0 ***
[InputMgr] *** Shutting Down 'Disabled' on Input: 1 ***
[InputMgr] *** Shutting Down 'FPP Remote' on Input: 0 ***
[InputMgr] *** Shutting Down 'Disabled' on Input: 1 ***
[InputMgr] *** Shutting Down 'Artnet' on Input: 0 ***
[InputMgr] *** Saved Input Manager Config File. ***
[InputMgr] --- WARNING: Creating a new Input Manager configuration Data set - Done ---

[WiFiMgr] WiFi Entering State: Connecting Using Config Credentials
[FileMgr] Configuration File: '/input_config.json' loaded.
[FileMgr] Configuration File: '/output_config.json' loaded.

[WiFiMgr] WiFi Entering State: Connecting Using Config Credentials
[WiFiMgr] WiFi Connecting to 'ci' as TESTER

[... 34 lines cut ...]
[ESP] Incorrect Config Version ID
[InputMgr] *** Saved Input Manager Config File. ***
[FileMgr] Configuration File: '/input_config.json' loaded.
[OutputMgr] *** Saved Output Manager Config File. ***
[FileMgr] Configuration File: '/output_config.json' loaded.
[WebMgr] WS client disconnect - 2
[WebMgr] WS client connect - 3
[ESP] Incorrect Config Version ID
[InputMgr] *** Saved Input Manager Config File. ***
[[OutputMgr] *** Saved Output Manager Config File. ***
FileMgr] Configuration File: '/input_config.json' loaded.
[FileMgr] Configuration File: '/output_config.json' loaded.
[WebMgr] WS client connect - 4
[WebMgr] WS client disconnect - 3
[WebMgr] WS client connect - 5
[WebMgr] WS client disconnect - 4
```

</details><hr/>

<details><summary>Really large screenshot...</summary><P/>

[image]

</details><hr/>

henrygab, 2021-09-26:
[`ValidateConfigFields()`](https://github.com/forkineye/ESPixelStick/blob/f45f334217e265478edc72f331ca6cee4ea8a7ce/html/script.js#L935-L959) is getting the following list of elements passed in as the only parameter:

<details><summary>ElementList parameter in debugger</summary><P/>

[image]

</details><HR/>

henrygab, 2021-09-26:
Looking at the underlying elements, it appears that the `hidden` class is applied to the `div` element that encloses the `input` element, but that the `input` elements themselves do not have the `hidden` class.  See the elements marked in green....

<details><summary>large screenshot</summary><P/>

[image]

</details><hr/>

Perhaps, because ESP8266 is not allowed to select APA102 as an output mode, the code is failing to apply the `hidden` class to the input field?
Or, perhaps it's applying the `hidden` class to the `div`, but not the corresponding `input` field?

forkineye, 2021-09-26:
Those elements are for the SD card pins and only applicable to ESP32 builds, that's why they're hidden. Not sure why edge is tripping up on them though. I just tried in Edge 94.0.992.31 and Chrome 90.0.4606.61 and am not expriecing it. What OS and version of Edge are you on?

marksat666, 2021-09-26:
I am experiencing the same restult (cannot save any Device Setup changes).  Does not work for me in Edge or Chrome.

ESPixelStick v3

Widows 10.0.19043 Build 19043
MS Edge Version 94.0.992.31 (Official build) (64-bit)
Chrome Version 94.0.4606.61 (Official Build) (64-bit)

I have to remove the Wemos from the Pixelstick to program it via USB, then disconnect it from the PC and put it back into the Pixelstick socket (not sure if that has anything to do with it).

henrygab, 2021-09-26:
> Those elements are for the SD card pins and only applicable to ESP32 builds, that's why they're hidden. Not sure why edge is tripping up on them though. I just tried in Edge 94.0.992.31 and Chrome 90.0.4606.61 and am not experiencing it. What OS and version of Edge are you on?

Edge is not tripping up on them.
The input elements do NOT have the `hidden` class:

[image]

As you can see in the picture, only the `div` parent element  has the `hidden` class.  Therefore, those three input fields (`miso_pin`, `mosi_pin`, `cs_pin`) are included in the validation check.  The function [`ValidateConfigFields()`](https://github.com/forkineye/ESPixelStick/blob/f45f334217e265478edc72f331ca6cee4ea8a7ce/html/script.js#L935-L959) is CORRECTLY reporting that the input fields do not have the `hidden` class applied.

henrygab, 2021-09-26:
I'm going to update the bug title to reflect that the SPI input fields fail to get the `hidden` class applied, resulting in inability to save configuration.

forkineye, 2021-09-26:
The thing is though, those have default values even though they're hidden which is what's confusing me.  What happens if you run ```$('#cs_pin').val()``` and ```$('#miso_pin').val()``` in the console?  Do they have values? CS is valid for ESP8266 and MISO will be bogus.

henrygab, 2021-09-26:
They all have the value `0`, as expected, as they are not exposed in the UI (at least one parent element has the `hidden` class).

**_Perhaps the problem is that many of those elements do not have the `esp32` class applied?_**

If they did, then I am guessing that they would get the `hidden` class applied as well?

henrygab, 2021-09-26:
@forkineye -- discord?

forkineye, 2021-09-26:
> They all have the value `0`, as expected, as they are not exposed in the UI (at least one parent element has the `hidden` class).
>
> **_Perhaps the problem is that many of those elements do not have [...]

> @forkineye -- discord?

I can't at the moment. Fixing it is easy, but I'm more curious as to why it's happening for some and not others, even on the same browser versions. It seems to be something related to how the DOM is being built, possibly extension related.  I'll work on the fix.
[image]

forkineye, 2021-09-26:
@henrygab You care to try this build and see if it fixes the issue for you? I still can't replicate here.
https://github.com/forkineye/ESPixelStick/actions/runs/1276196142

forkineye, 2021-09-26:
@henrygab I just noticed the following in your boot log above:
```[ESP] Incorrect Config Version ID```

That shouldn't be happen either if using the release archive w/ the flasher. Did you edit config.json by hand before flashing?

henrygab, 2021-09-27:
@forkineye - No manual edits.  I will fully erase the flash and repro again.

However, I'm having similar issues on an ESP32 module, where the SD pins have the value zero (shown by enabling advanced options under Admin), and reproduces even when all add-ins are fully disabled (they're disabled anyways on my pixel controllers).  Hopefully, this is just a browser compatibility issue.  Here's browser info -- it's just standard (Chromium-based) Edge on Windows 10:

EDGE VERSION |   | 93.0.961.52
-- | -- | --
OS VERSION |   | Windows NT: 10.0.19043

henrygab, 2021-09-27:
Confirmed ... this occurs on a freshly-erased module at boot:

```
[ESP] ESP Version: SDK:2.2.2-dev(38a443e)/Core:3.0.2=30002000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:6105635
[FileMgr] Flash file system initialized.
[FileMgr] No SD card installed
[FileMgr] Configuration File: '/config.json' loaded.
[ESP] Incorrect Config Version ID
[OutputMgr] *** 'Disabled' Initialization for Output: 0 ***
[FileMgr] ***Configuration File: '/output_config.json' Could not open file for reading.***
[OutputMgr] EEEE Error loading Output Manager Config File. EEEE
[OutputMgr] --- WARNING: Creating a new Output Manager configuration Data set - Start ---
[OutputMgr] *** Shutting Down 'Disabled' on Output: 0 ***
[OutputMgr] *** 'WS2811' Initialization for Output: 0 ***
[OutputMgr] *** Shutting Down 'WS2811' on Output: 0 ***
[OutputMgr] *** 'GECE' Initialization for Output: 0 ***
[OutputMgr] *** Shutting Down 'GECE' on Output: 0 ***
[OutputMgr] *** 'DMX' Initialization for Output: 0 ***
[OutputMgr] *** Shutting Down 'DMX' on Output: 0 ***
[OutputMgr] *** 'Renard' Initialization for Output: 0 ***
[OutputMgr] *** Shutting Down 'Renard' on Output: 0 ***
[OutputMgr] *** 'Serial' Initialization for Output: 0 ***
[OutputMgr] *** Shutting Down 'Serial' on Output: 0 ***
[OutputMgr] *** Unknown output type: '5 for channel '0'. Using disabled. ***
[OutputMgr] *** 'Disabled' Initialization for Output: 0 ***
[OutputMgr] *** Shutting Down 'Disabled' on Output: 0 ***
[OutputMgr] *** Unknown output type: '6 for channel '0'. Using disabled. ***
[OutputMgr] *** 'Disabled' Initialization for Output: 0 ***
[OutputMgr] *** Saved Output Manager Config File. ***
[OutputMgr] --- WARNING: Creating a new Output Manager configuration Data set - Done ---
[FileMgr] ***Configuration File: '/input_config.json' Could not open file for reading.***
[InputMgr] EEEE Error loading Input Manager Config File. EEEE
[InputMgr] --- WARNING: Creating a new Input Manager configuration Data set - Start ---
[InputMgr] *** Shutting Down 'Disabled' on Input: 0 ***
[InputMgr] *** Shutting Down 'Disabled' on Input: 1 ***
[InputMgr] *** Shutting Down 'E1.31' on Input: 0 ***
[InputMgr] *** Shutting Down 'Disabled' on Input: 1 ***
[InputMgr] *** Shutting Down 'Disabled' on Input: 0 ***
[InputMgr] *** Shutting Down 'Effects' on Input: 1 ***
[InputMgr] *** Shutting Down 'Disabled' on Input: 0 ***
[InputMgr] *** Shutting Down 'MQTT' on Input: 1 ***
[InputMgr] *** Shutting Down 'Disabled' on Input: 0 ***
[InputMgr] *** Shutting Down 'Alexa' on Input: 1 ***
[InputMgr] *** Shutting Down 'DDP' on Input: 0 ***
[InputMgr] *** Shutting Down 'Disabled' on Input: 1 ***
[InputMgr] *** Shutting Down 'FPP Remote' on Input: 0 ***
[InputMgr] *** Shutting Down 'Disabled' on Input: 1 ***
[InputMgr] *** Shutting Down 'Artnet' on Input: 0 ***
[InputMgr] *** Saved Input Manager Config File. ***
[InputMgr] --- WARNING: Creating a new Input Manager configuration Data set - Done ---

[WiFiMgr] WiFi Entering State: Connecting Using Config Credentials
[FileMgr] Configuration File: '/input_config.json' loaded.
[FileMgr] Configuration File: '/output_config.json' loaded.

[WiFiMgr] WiFi Entering State: Connecting Using Config Credentials
[WiFiMgr] WiFi Connecting to 'ci' as TESTER

[WiFiMgr] WiFi Entering State: Connected To AP
[WiFiMgr] WiFi Connected with DHCP
[WiFiMgr] WiFi Connected with IP: 10.3.0.112
[WebMgr] Web server listening on port 80
[FPPD] Listening on port 32320
```

forkineye, 2021-09-27:
@henrygab Can you post the contents of ```fs/config.json``` with credentials removed of course.  The cfgver key seems to be invalid for some reason. Also, did the build I linked above resolve the original issue?

henrygab, 2021-09-27:
Sorry, didn't mean to ignore it ... I missed that build link earlier.  Will try very shortly.

The cfgver key message is only on the first boot after flashing.  It does not occur when rebooting later.  Have you tried fully erasing the board (using esptool.py), then uploading via the included ESPSFlashTool.jar?  The output I quoted was from that serial monitor, from immediately following a flash.

Is it possible that LittleFS's ... unique ... behaviors could be involved?  IIRC, file handles have to be closed before changes made are seen by other calls to the file system ... and maybe even require that the file be re-opened to see those changes?

henrygab, 2021-09-27:
Using the binary from https://github.com/forkineye/ESPixelStick/actions/runs/1276196142:

The ability to save changes is still disabled.  However, each of `miso_pin`, `mosi_pin`, and `clock_pin` each now properly have the `hidden` class applied to them.  Thus, even though they are each invalid, the function ignores them for this purpose.

However, the `cs_pin` input is still marked as invalid.  When I enable advanced mode, the HTML shows it is set to zero.

This is from a fully-erased state.  No prior configuration in flash.  _(I presume that the EEPROM is not used.)_

henrygab, 2021-09-27:
<details><summary>What did I mean about LittleFS "unique" behaviors</summary><P/>

See https://github.com/littlefs-project/littlefs/issues/352#issuecomment-567178172.

This is my recollection, but should give hints of what to be concerned about.

* littleFS uses [copy-on-(bounded)-write](https://github.com/littlefs-project/littlefs/blob/1863dc7883d82bd6ca79faa164b65341064d1c16/DESIGN.md#wear-leveling)
* each file handle is a snapshot of the file as of the time it was opened, plus those changes that were made using that same file handle
* when two file handles are opened to the same file, writing to the first file handle should **_NOT_** alter the data as seen from the other file handle
* when the first file handle is closed (or manually sync'd), then the data will appear in all future opens of that file ... but still won't update what is seen via the second  (still open) file handle

</details><hr/>

I'm just throwing this out there, as I was [surprised](https://github.com/adafruit/Adafruit_nRF52_Arduino/pull/397) by littleFS behavior in the past, which resulted in some really hard-to-find causes of corruption.

My learnings:
1. serialize all access to littleFS APIs
2. put all files into a subdirectory that only you will read / write / modify
3. ensure you never have more than one file handle per file
4. ensure you cause file sync whenever you need the state to persist

MartinMueller2003, 2021-09-27:
A few comments about how little FS is used.
ESP32 fields are NOT present in the ESP8266 build. They have unique fields.Files are opened and closed immediately. If the files are missing at boot time, they are created and closed one at a time.At boot the littleFS files are read and immediately closed.When the UI needs the config, it is asked for one file at a time. The file is opened, data read into a buffer and closed. Then the buffer is sent to the UI.When the UI is done the files are set to the ESP one at a time. Each file is written to the FS from a buffer and closed before the next file is sent. After a file is received a flag is set indicating that a new config is present. One subsystem at a time opens its file reads into a buffer and closes the file. Then process the buffer.More than one open file in the littleFS does not happen.  

 
 

See littlefs-project/littlefs#352 (comment).

forkineye, 2021-09-27:
@henrygab 
In ```firmware/firmware.json``` there are baud rates defined for flashing each platform. For example - https://github.com/forkineye/ESPixelStick/blob/26d6f6acb70b8a67884d0ea8fc88bea2cdb555ab/dist/firmware/firmware.json#L11.  They are all set to 460800.  Can you try lowering them to 115200 and see if that makes a difference?

forkineye, 2021-09-27:
@henrygab 
In regards to the original issue, can you monitor the websocket connection and see what is being sent when you go to the device setup tab? When navigating to the tab, you should see a message go out:
```json
{
  "cmd": {
    "get": "device"
  }
}
```
and a response similar to the following:
```json
{
  "get": {
    "device": {
      "cfgver": "1",
      "id": "ESPixelStickv3",
      "blanktime": "5",
      "miso_pin": "19",
      "mosi_pin": "23",
      "clock_pin": "18",
      "cs_pin": "15"
    },
    "network": {
      "ssid": "myssid",
      "passphrase": "mypassphrase",
      "hostname": "esps-f4b650",
      "ip": "192.168.1.10",
      "netmask": "255.255.255.0",
      "gateway": "192.168.1.1",
      "dhcp": true,
      "sta_timeout": 15,
      "ap_fallback": false,
      "ap_timeout": 120,
      "ap_reboot": true
    }
  }
}
```
[image]

marksat666, 2021-09-27:
> @henrygab @marksat666
> You care to try this build and see if it fixes the issue for you? I still can't replicate here.
> https://github.com/forkineye/ESPixelStick/actions/runs/1276196142

No difference for me for that version, also just tried CI-1278932930 - still cannot save the Device Setup

forkineye, 2021-09-27:
I just found an issue where the flash tool and firmware were out of sync for creating the configuration file and that erasure of the previous filesystem may not always occur.   Not sure if it's causing this issue yet, but possible.  I'll have a fix pushed soon.

forkineye, 2021-09-27:
@marksat666 Care to try this build?
https://github.com/forkineye/ESPixelStick/actions/runs/1279404057
thanks,
-shelby

marksat666, 2021-09-27:
> @marksat666 Care to try this build?
> https://github.com/forkineye/ESPixelStick/actions/runs/1279404057
> thanks,
> -shelby

CI-1279404057 also did not work in Edge, Chrome, Safari 13.1.2, Silk 92.2.11.4514.159.19 (figured I would give those  others a try - maybe it's not a browser issue?)  I also grabbed a spare Wemos D1 mini that I had from last year (I have been testing with the Wemos/ESPixelStick that I got in April) - same results.

forkineye, 2021-09-27:
@marksat666 Can you post the contents of both the flash upload and serial log windows from the flash tool?

marksat666, 2021-09-27:
Here are logs - let me know if you want me to try something else.
[Serial_Log.txt](https://github.com/forkineye/ESPixelStick/files/7239489/Serial_Log.txt)
[Flash_Upload.txt](https://github.com/forkineye/ESPixelStick/files/7239490/Flash_Upload.txt)
..

forkineye, 2021-09-27:
@marksat666 @henrygab
I thought this could be related to the websocket  keep-alive stuff I had implemented, but not sure. I made some changes and there are two CI builds now, one with the keep-alive removed, and the other with an updated version.  If you do don't mind to check these out, I would appreciate it. Also, please make sure that your browser is doing a force refresh so the javascript stuff is cached.
no ping - https://github.com/forkineye/ESPixelStick/actions/runs/1280125433
ping - https://github.com/forkineye/ESPixelStick/actions/runs/1280264355

forkineye, 2021-09-28:
Was able to replicate the issue finally. Can you verify if you were or were not setting a hostname via the flash tool? It seems to be related to that of all things.  If you were setting one, try without.

guedeaux, 2021-09-28:
I was having this same behavior. I can confirm that not setting a hostname resolved the issue for me. I am now able to save changes on the device setup page, and I now see input options, which I did not before.

forkineye, 2021-09-28:
> I was having this same behavior. I can confirm that not setting a hostname resolved the issue for me. I am now able to save changes on the device setup page, and I now see input options, which I did not before.

Thanks! I just posted a fix for it here - https://github.com/forkineye/ESPixelStick/actions/runs/1280577528

guedeaux, 2021-09-28:
Sweet. That version works for me with a hostname set. Now, to get some lights hooked up...

marksat666, 2021-09-28:
> Thanks! I just posted a fix for it here - https://github.com/forkineye/ESPixelStick/actions/runs/1280577528

Yes, I was setting a host name in the flash tool itself.  Omitting that during the flash works!  I can now save the settings.

Now the bad (using CI-1280577528) - FPP Connect in XLights no longer sees the ESPixelStick when I set Primary Input to be FPP Remote (it works with E1.31 input from XLights directly and from FPP)

forkineye, 2021-09-28:
@marksat666 That's probably related to this - https://github.com/forkineye/ESPixelStick/issues/364.  I'll fix that today, can track it there.

I'll leave this issue open for others until the next beta release

closed 2021-09-29 by forkineye

henrygab, 2021-09-30:
> Was able to replicate the issue finally. Can you verify if you were or were not setting a hostname via the flash tool? It seems to be related to that of all things. If you were setting one, try without.

Sorry for delay in testing and response.  Yes, I was setting a hostname via the flash tool.  I am glad to again be amazed at how complex systems interact.  😏 Well done discovering the underlying cause(s).

I wanted to confirm that v4.0-beta2 resolves the issue originally reported here.

davron12, 2024-12-12:
This appears to still be an issue in 4.0.0-Beta5, at least for me, running on as ESP32-CAM. I was thinking it was a hidden field causing validation issues, when I came across the hostname solution. Tried it out and it solved my issue.