forkineye/ESPixelStick #362: [4.0.0-beta1] Infinite boot loop on new ESP32?

henrygab, 2021-09-27:
--------- Instructions - Delete this ---------
Please provide answers directly below each section.
--------- Instructions - Delete to here ---------
**ESPixelStick Firmware Version**
4.0.0-Beta1 (7083637f81f53b5b15c9fefd2b28f192c30334b6)

**Hardware Version**
[TwilightLord ESP32 16MB](https://www.tindie.com/products/twilightlord/twilightlord-esp32-16mb/)

**Binary release or compiled yourself?**
Binary release

**Operating System**
N/A

**Access Point**
Ubiquiti Nanostation M2 Loco

**Describe the bug**

Bare, fresh module -- **nothing attached to any pins**.  Fully erased flash just before flashing this beta, choosing Generic ESP32 4MB (should just ignore the extra 12MB).

Gets into an infinite cycle, and appears to have errors saving its own config to the internal flash?

The following output lines may be of particular interest:
```
./components/esp_littlefs/src/littlefs/lfs.c:495:error: No more free space 50
...
[FileMgr] ***Configuration File: '/input_config.json' Deserialzation Error. Error code = EmptyInput***
[InputMgr] EEEE Error loading Input Manager Config File. EEEE
```

Attached an example log showing the infinitely looping output.

[espixelstick_fs_error.txt](https://github.com/forkineye/ESPixelStick/files/7232509/espixelstick_fs_error.txt)

(The module itself is fine ... WLED works on this same module.)

MartinMueller2003, 2021-09-27:
The File System on this device is in a bad state. It looks like it was not properly set up. This line is the key:
`./components/esp_littlefs/src/littlefs/lfs.c:495:error: No more free space 50`
We do not use anywhere near the total space available in a properly formatted FS.

forkineye, 2021-09-27:
@henrygab 
For this one, can you try this as well? https://github.com/forkineye/ESPixelStick/issues/360#issuecomment-927861402

forkineye, 2021-09-27:
Normally, we just erase the filesystem area on flash. I just changed the flash tool to do a full erase, can you give this one a try - https://github.com/forkineye/ESPixelStick/actions/runs/1279404057

forkineye, 2021-09-27:
@henrygab 
Normally, we just erase the filesystem area on flash. I just changed the flash tool to do a full erase, can you give this one a try - https://github.com/forkineye/ESPixelStick/actions/runs/1279404057

forkineye, 2021-09-28:
I've tried on all my modules can still can't re-produce this.  If you're still experiencing this in the newer builds, let us know.

closed 2021-09-28 by forkineye

henrygab, 2021-09-30:
I apologize for the delayed response.  (_Real life is sometimes busy._)

I had been fully erasing the flash using `espytool.py` before loading any of the 4.0 builds.  So, I don't think having the flash tool full erase would change the result.

However, it appears that [ESPixelStick_Firmware-4.0-ci1282596145](https://github.com/forkineye/ESPixelStick/actions/runs/1282596145) no longer reproduces the issue, so OK to close.