forkineye/ESPixelStick #363: "Time Elapsed" or "Time Remaining" Status not updating

computergeek1507, 2021-09-28:
**ESPixelStick Firmware Version**
FW Version
CI-1279404057

Build Date
Sep 27 2021 - 17:56:28

**Hardware Version**
Lolin D32 Pro

**Binary release or compiled yourself?**
Release Action https://github.com/forkineye/ESPixelStick/actions/runs/1279404057

**Operating System (and version)**
Windows 10 Pro 21H1

**Web Browser (and version)**
Chrome 93.0.4577.82 

**Access Point**
Ubiquiti UAP-AC-LR

**Describe the bug**
When using Local File Player the Status page doesn't update the "Time Elapsed" or "Time Remaining"

[image]

MartinMueller2003, 2021-09-28:
Someone else reported something similar. The issue turned out to be in the fseq file itself. It looks like on occasion xLights FPP Connect generates a file in a format we did not support. I am in the process of adding support and it should be ready shortly.

computergeek1507, 2021-09-28:
In my testing the FSEQ was playing and the lights were working, the status page just wasn't updating

MartinMueller2003, 2021-09-28:
Was anything updating? Was the uptime changing?

computergeek1507, 2021-09-28:
The uptime and heap were updating

MartinMueller2003, 2021-09-28:
So the screen was updating, The file was playing and the play status was not reflecting progress. Can you send me the file you are playing. I will try it on my system.

computergeek1507, 2021-09-28:
[ttttttt.zip](https://github.com/forkineye/ESPixelStick/files/7243610/ttttttt.zip)

MartinMueller2003, 2021-09-28:
That is new. Have not seen it in months. Also running fine on my build. It is not arduinojson that causes the issue. It is anything to do with string processing. Turning on the debugging itself causes issues in this area because it uses strings and all web actions are in an interrupt mode of operation. Using the String class does bad things to the heap and heap manager gets hosed.

forkineye, 2021-09-28:
@computergeek1507 How many channels of data does that sequence have? According to out header definition, it's 40,800.
As for playback status, it's working fine with my sequences. I'm thinking we may have an issue elsewhere.
[image]

forkineye, 2021-09-28:
According to the [FPP Spec](https://github.com/FalconChristmas/fpp/blob/master/docs/FSEQ_Sequence_File_Format.txt), we should be correct. Was xLights dumping the total channel count into the file? According to the spec, it should be for just the file.
```
(*) The channel count is per frame within this file which may not
be the full number of channels needed to output.  For example, if there
is a single "sparse range" of start channel 5000 with lengh 50, the
channel count in bytes 10-13 will be "50" as there will only be 50
channels per frame within the file.   If there are multiple sparse
ranges, each range is appended one after another into the frame
with the channel count being the total lengths of the ranges.
```

computergeek1507, 2021-09-28:
I will look into the xLights code maybe we have an issue there

MartinMueller2003, 2021-09-28:
I just looked at the raw file contents and it does not look right. The data start does not extend past the headers and the data area is completly blank. I will continue to analyze it tomorrow.
Martin

MartinMueller2003, 2021-09-29:
It took a refactoring and a bit of frustration to find this non bug. The issue is the FSEQ file is broken. The channels per frame count is 40800 with 600 frames. That needs a file that is about 20MB long. The file I received is ~700KB. The ESP ran out of data within the first second. Outputted an error message and then restarted the file. Since the playback never completed the first second the times did not change. The status was correct. On the plus side I did find and fix a parsing issue.

07:43:36.386 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:188:                     BytesRead: 32 ------
07:43:36.431 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:189:         sizeof (fsqRawHeader): 32 ------
07:43:36.431 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:197: Convert Raw Header into a processed header ------
07:43:36.431 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:213:                    dataOffset: 68 ------
07:43:36.431 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:214:                  minorVersion: 0 ------
07:43:36.431 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:215:                  majorVersion: 2 ------
07:43:36.476 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:216:             VariableHdrOffset: 32 ------
07:43:36.476 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:217:                  channelCount: 40800 ------
07:43:36.476 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:218: TotalNumberOfFramesInSequence: 600 ------
07:43:36.476 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:219:                      stepTime: 50 ------
07:43:36.476 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:220:                         flags: 0 ------
07:43:36.521 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:221:               compressionType: 0x0 ------
07:43:36.521 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:222:           numCompressedBlocks: 0 ------
07:43:36.521 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:223:               numSparseRanges: 0 ------
07:43:36.521 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:224:                        flags2: 0 ------
07:43:36.521 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:225:                            id: 0xf14e7700 ------

closed 2021-09-29 by MartinMueller2003

forkineye, 2021-09-30:
@computergeek1507 Were able to get the bottom of what happened with the file? Where you able to compare it to the original version? Or had it been uploaded and then extracted from the SD card?

computergeek1507, 2021-09-30:
I was working on testing the ESPixelStick 4 beta 2 firmware with xlights, but now xLights will lockup when trying to upload the channel outputs.

MartinMueller2003, 2021-09-30:
xLights also locks up if you try to send a test pattern. Not sure why

forkineye, 2021-09-30:
> I was working on testing the ESPixelStick 4 beta 2 firmware with xlights, but now xLights will lockup when trying to upload the channel outputs.

Which version of xLights? It worked for me yesterday with 2021.29 x64 windows

computergeek1507, 2021-09-30:
I was using 2021.30 and running from debug and it was doing it. The websocket call to get the outputs was not returning any response and xLights was locking up waiting for it. The input upload wasn't working either but I was working on a xLights fix for that.

MartinMueller2003, 2021-09-30:
Both 29 and 30 lock up on me.

forkineye, 2021-09-30:
I'll update to 30 and dig into it if I can reproduce it

forkineye, 2021-09-30:
@MartinMueller2003 @computergeek1507 
I haven't been able to reproduce the upload input/output issues, but I have found something broken with files uploaded via FPP connect. Let's move over to issue https://github.com/forkineye/ESPixelStick/issues/367 since this one is closed.

computergeek1507, 2021-09-30:
based on the FSEQ doc and talking to Dan, the channel count in the sparse file header is the channel size of main FSEQ before the other data is strips out to generate the sparse file specific the remote controller. The sparse range start channel of the sparse file is then relative to this. The frame channel size of the sparse file should be the global channel size minus the range start channel +1. The sparse ranges size value should also match this value.

forkineye, 2021-09-30:
> based on the FSEQ doc and talking to Dan, the channel count in the sparse file header is the channel size of main FSEQ before the other data is strips out to generate the sparse file specific the [...]

So xLights is not updating the channel count to reflect the sparse range count? 

```
(*) The channel count is per frame within this file which may not
be the full number of channels needed to output.  For example, if there
is a single "sparse range" of start channel 5000 with lengh 50, the
channel count in bytes 10-13 will be "50" as there will only be 50
channels per frame within the file.   If there are multiple sparse
ranges, each range is appended one after another into the frame
with the channel count being the total lengths of the ranges.
```

philasmith, 2021-09-30:
Thanks

Phil Smith
Phone 604-793-5972

MartinMueller2003, 2021-09-30:
This is very different than the documentation.
Martin

forkineye, 2021-09-30:
In regards to the xLights crash when uploading channel outputs, it only occurs with ESP32 for me. ESP8266 has been fine.

computergeek1507, 2021-09-30:
I think xLights should be changed to match the doc, I will talk to Dan about fixing it. I don't want to break FPP and the Falcons