forkineye/ESPixelStick #364: fppjson.php?command=getSysInfo&simple not returning correct major version

computergeek1507, 2021-09-28:
**ESPixelStick Firmware Version**
FW Version
CI-1279404057

Build Date
Sep 27 2021 - 17:56:28

**Hardware Version**
Lolin D32 Pro

**Binary release or compiled yourself?**
Release Action https://github.com/forkineye/ESPixelStick/actions/runs/1279404057

**Operating System (and version)**
Windows 10 Pro 21H1

**Web Browser (and version)**
Chrome 93.0.4577.82 

**Access Point**
Ubiquiti UAP-AC-LR

**Describe the bug**
the endpoint "/fppjson.php?command=getSysInfo&simple" returns a major version of '0'. xLights FPP Connect is looking for version "4" to upload a sequence.

[image]

MartinMueller2003, 2021-09-28:
Not sure when this changed. Other have been playing in this area. I will take a look.

MartinMueller2003, 2021-09-28:
Not sure why you are not getting the correct info. This is what I see:

[image]

computergeek1507, 2021-09-28:
I thing it's because I flashed the release action, Shelby changed it to be the build number and not have the major version number I think

MartinMueller2003, 2021-09-28:
That would be a problem. 
Martin

forkineye, 2021-09-28:
Yeah, there is no major version related to the CI builds. I'll change that today,

forkineye, 2021-09-28:
Fixed in the following build - https://github.com/forkineye/ESPixelStick/actions/runs/1282596145
Also, I'm going back to x.y versioning instead of x.y.z in the next beta. 

Leaving this open until the next beta others that may experience the same issue.

closed 2021-09-29 by forkineye