forkineye/ESPixelStick #367: FPP sequence playback is broken in 4.0 Beta 2

patdelaney, 2021-09-30:
**ESPixelStick Firmware Version**
4.0-beta2

**Hardware Version**
Artstick esp8266

**Binary release or compiled yourself?**
Actions Beta 2 build

**Web Browser (and version)**
Chrome

**Access Point**
Ubiquiti

**Describe the bug**
Playing know good sequence directly or via fpp remote display random data on the output see attached YouTube link.
Note: The same sequence played in Beta 1
https://youtu.be/6UE-LjuXRb8

forkineye, 2021-09-30:
Is that the same file you sent me? Can you attach it the FSEQ to this as well along with the pixel count / dimensions of the matrix?

forkineye, 2021-09-30:
There seems to be an issue when the files are uploaded with FPP Connect.  When uploaded manually (rendered for just the device), they play fine.  The file sizes are different as well when uploaded via FPP Connect.  I'll figure out what's going on today. Also, I've attached two fseq files based off the FPP_Multisync_Test sequence that you sent me. One is a standalone rendered for the 8x32 matrix you have. The other is rendered for an 8x32 matrix as well, but with the layout changed so that it renders the same in the diagnostics tab once the columns set to 32. If you could do some testing with these while I work on what's going on with FPP connect, I'd appreciate it. Thanks!
[FPP_Multisync_Test-esps.zip](https://github.com/forkineye/ESPixelStick/files/7260699/FPP_Multisync_Test-esps.zip)

patdelaney, 2021-09-30:
It's the xlights project I sent you guys the other day.

MartinMueller2003, 2021-09-30:
The header is messed up in this file. Now that we pay attention to sparse ranges, the data in the sparse ranges structure needs to be correct. My latest changes will validate the ranges and ignore them if they are not valid.

11:57:35.344 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:201: Convert Raw Header into a processed header ------
11:57:35.344 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:217:                    dataOffset: 144 ------
11:57:35.389 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:218:                  minorVersion: 0 ------
11:57:35.389 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:219:                  majorVersion: 2 ------
11:57:35.389 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:220:             VariableHdrOffset: 38 ------
11:57:35.389 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:221:                  channelCount: 1020 ------
11:57:35.389 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:222: TotalNumberOfFramesInSequence: 1200 ------
11:57:35.436 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:223:                      stepTime: 50 ------
11:57:35.436 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:224:                         flags: 0 ------
11:57:35.436 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:225:               compressionType: 0x0 ------
11:57:35.436 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:226:           numCompressedBlocks: 0 ------
11:57:35.436 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:227:               numSparseRanges: 1 ------
11:57:35.436 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:228:                        flags2: 0 ------
11:57:35.483 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:229:                            id: 0xc96551d8 ------
11:57:35.483 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:237:  ------
11:57:35.483 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:276:            Sparse Range Index: 0 ------
11:57:35.483 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:290:             RangeChannelCount: 16515077 ------
11:57:35.483 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:291:               RangeDataOffset: 0xfe0005 ------
11:57:35.529 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:276:            Sparse Range Index: 1 ------
11:57:35.529 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:279: No Sparse Range Data for this entry  ------
11:57:35.529 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:276:            Sparse Range Index: 2 ------
11:57:35.529 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:279: No Sparse Range Data for this entry  ------
11:57:35.529 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:276:            Sparse Range Index: 3 ------
11:57:35.529 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:279: No Sparse Range Data for this entry  ------
11:57:35.576 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:276:            Sparse Range Index: 4 ------
11:57:35.576 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:279: No Sparse Range Data for this entry  ------
11:57:35.576 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:296:                 TotalChannels: 16515077 ------
11:57:35.576 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:297:                 LargestOffset: 16646149 ------
11:57:35.576 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:298:                  LargestBlock: 33161226 ------
11:57:35.623 -> [InputMgr] ParseFseqFile:: Ignoring Range Info. FPP_Multisync_Test (1).fseq Too many channels defined in Sparse Ranges.

computergeek1507, 2021-09-30:
I think the range code is incorrect in xLights, if you have a FPP controller and a ESPixelstick, I think it breaks.

MartinMueller2003, 2021-09-30:
Just pushed code that ignores ranges if they are incorrect. 
Martin

forkineye, 2021-09-30:
Build from PR is here - https://github.com/forkineye/ESPixelStick/actions/runs/1291838875

forkineye, 2021-09-30:
With the range fix (PR https://github.com/forkineye/ESPixelStick/pull/370), It's failing to render in a single controller setup where the file contains all channels yet is still sparse since that is what xLights uploads. 
Sequences: [test.zip](https://github.com/forkineye/ESPixelStick/files/7262611/test.zip)

For instance, 'matrixTest' is a single controller setup. The full fseq that xLights saves renders fine, the sparse uploaded via FPP Connect does not.
```
matrixTest-full.fseq
--------------------
File ID:        PSEQ
Data Offset:    68
Minor Version:  0
Major Version:  2
Header Length:  32
Channel Count:  2560
Frame Count:    1200
Step Time(ms):  25
Flags:          0
Compression:    0 (uncompressed)
Comp Blocks:    0
Sparse Ranges:  0
FLags 2:        0
Unique ID:      1633021536762000
Created By:     xLights Windows 2021.30 64bit

matrixTest-sparse.fseq
----------------------
File ID:        PSEQ
Data Offset:    72
Minor Version:  0
Major Version:  2
Header Length:  38
Channel Count:  2560
Frame Count:    1200
Step Time(ms):  25
Flags:          0
Compression:    0 (uncompressed)
Comp Blocks:    0
Sparse Ranges:  1
FLags 2:        0
Unique ID:      1633021536762000
Created By:     xLights Windows 2021.30 64bit
``` 

However, if the controller is part of a larger show where the channel value is invalid and ignored, it works as in this sequence:
```
bogusTest-sparse.fseq
---------------------
File ID:        PSEQ
Data Offset:    72
Minor Version:  0
Major Version:  2
Header Length:  38
Channel Count:  2550
Frame Count:    1200
Step Time(ms):  25
Flags:          0
Compression:    0 (uncompressed)
Comp Blocks:    0
Sparse Ranges:  1
FLags 2:        0
Unique ID:      1633036158224000
Created By:     xLights Windows 2021.30 64bit
```

forkineye, 2021-09-30:
@MartinMueller2003 Maybe best if we just ignore the channel count in the main header when a sparse file is detected?

MartinMueller2003, 2021-10-01:
Please send me the files and I will take a look

forkineye, 2021-10-01:
> Please send me the files and I will take a look 

They're attached here - https://github.com/forkineye/ESPixelStick/issues/367#issuecomment-931721347

Something else interesting, I noticed ```bogusTest-sparse.fseq``` had the header channel count set per the spec.  I'm curious how some people are getting sparse fseq files with headers that aren't per spec? Maybe something going on in FPP connect when uploading to multiple controllers with different fseq formats?

I added the sparse range data to my script as well, here are what the above files look like:
```
bogusTest-sparse.fseq
---------------------
Channel Count:   2550
Sparse Start:    510
Sparse Channels: 2550

matrixTest-sparse.fseq
----------------------
Channel Count:   2560
Sparse Start:    0
Sparse Channels: 2560

MartinMueller2003, 2021-10-01:
in the current version bogus will get flagged and will not run. it is not valid. matrix should run ok.

They're attached here - #367 (comment)

MartinMueller2003, 2021-10-01:
Using current code, The bogus file will play using the non range mode. It will log an error message to the status report being sent to the UI.
09:08:17.227 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:219:                    dataOffset: 72 ------
09:08:17.227 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:220:                  minorVersion: 0 ------
09:08:17.227 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:221:                  majorVersion: 2 ------
09:08:17.273 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:222:             VariableHdrOffset: 38 ------
09:08:17.273 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:223:                  channelCount: 2550 ------
09:08:17.273 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:224: TotalNumberOfFramesInSequence: 1200 ------
09:08:17.273 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:225:                      stepTime: 25 ------
09:08:17.273 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:226:                         flags: 0 ------
09:08:17.320 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:227:               compressionType: 0x0 ------
09:08:17.320 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:228:           numCompressedBlocks: 0 ------
09:08:17.320 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:229:               numSparseRanges: 1 ------
09:08:17.320 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:230:                        flags2: 0 ------
09:08:17.320 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:231:                            id: 0xdc1dc680 ------
09:08:17.366 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:294:             RangeChannelCount: 16121857 ------
09:08:17.366 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:295:               RangeDataOffset: 0xfe0005 ------
09:08:17.366 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:302:                 TotalChannels: 16121857 ------
09:08:17.366 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:303:                 LargestOffset: 16646149 ------
09:08:17.366 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:304:                  LargestBlock: 32768006 ------
09:08:17.411 -> [InputMgr] ParseFseqFile:: Ignoring Range Info. bogusTest-sparse.fseq Too many channels defined in Sparse Ranges.
09:08:17.411 -> [InputMgr] Start Playing:: FileName: 'bogusTest-sparse.fseq'

Matrix test full play fine since it has no ranges and the header is valid
09:09:46.014 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:219:                    dataOffset: 68 ------
09:09:46.014 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:220:                  minorVersion: 0 ------
09:09:46.014 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:221:                  majorVersion: 2 ------
09:09:46.014 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:222:             VariableHdrOffset: 32 ------
09:09:46.014 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:223:                  channelCount: 2560 ------
09:09:46.060 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:224: TotalNumberOfFramesInSequence: 1200 ------
09:09:46.060 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:225:                      stepTime: 25 ------
09:09:46.060 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:226:                         flags: 0 ------
09:09:46.060 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:227:               compressionType: 0x0 ------
09:09:46.060 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:228:           numCompressedBlocks: 0 ------
09:09:46.060 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:229:               numSparseRanges: 0 ------
09:09:46.105 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:230:                        flags2: 0 ------
09:09:46.105 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:231:                            id: 0x749bf890 ------
09:09:46.105 -> [InputMgr] Start Playing:: FileName: 'matrixTest-full.fseq'

Playing the matrix test sparse currently results in the file "Playing" but nothing comes out. This is because the range does not have a channel count. I will modify the code to revert to playing without using the ranges.

09:14:07.986 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:219:                    dataOffset: 72 ------
09:14:07.986 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:220:                  minorVersion: 0 ------
09:14:07.986 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:221:                  majorVersion: 2 ------
09:14:07.986 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:222:             VariableHdrOffset: 38 ------
09:14:07.986 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:223:                  channelCount: 2560 ------
09:14:08.031 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:224: TotalNumberOfFramesInSequence: 1200 ------
09:14:08.031 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:225:                      stepTime: 25 ------
09:14:08.031 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:226:                         flags: 0 ------
09:14:08.031 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:227:               compressionType: 0x0 ------
09:14:08.031 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:228:           numCompressedBlocks: 0 ------
09:14:08.076 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:229:               numSparseRanges: 1 ------
09:14:08.076 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:230:                        flags2: 0 ------
[... cut ...]
09:14:08.076 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:295:               RangeDataOffset: 0x5 ------
09:14:08.121 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:302:                 TotalChannels: 0 ------
09:14:08.121 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:303:                 LargestOffset: 5 ------
09:14:08.121 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:304:                  LargestBlock: 5 ------
09:14:08.121 -> [InputMgr] Start Playing:: FileName: 'matrixTest-sparse.fseq'

forkineye, 2021-10-01:
@MartinMueller2003 
Hold off on any changes. I've already fixed it, will push soon.

MartinMueller2003, 2021-10-01:
Sorry I already submitted my change

MartinMueller2003, 2021-10-01:
Five line change using info already present

forkineye, 2021-10-01:
@MartinMueller2003 
There were two issues in the way we were handling the files. One was the using the sparse start channel. It references the channel offset of the master sequence and doesn't pertain to us. The other issue is it was looking for the start of sparse ranges in the wrong location.

MartinMueller2003, 2021-10-01:
That makes very little sense when compared to the FSEQ reference document which clearly states the channelOffset in the range is just that, an offset into the frame at which to send the data. FYI: My implementation is based on the code I was examining in the FPP 5.2 branch. In that code the range presents offset into the frame and number of bytes to send for each range. For a given range, offset plus length must not exceed frame size.

forkineye, 2021-10-01:
@patdelaney 
Could you test this and let me know if it fixes the issue?
https://github.com/forkineye/ESPixelStick/actions/runs/1295291307

forkineye, 2021-10-01:
Anyone able to verify if either the latest CI build or the one linked above this post takes care of this? I've tested it a few different way but would like some verification before we drop a beta3.

forkineye, 2021-10-02:
@patdelaney verified

closed 2021-10-02 by forkineye