forkineye/ESPixelStick #368: FSEQ will not load to ESPstick from the file manager and files are not seen that are on the disk

rbearwy, 2021-09-30:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
Version 4.02Beta

**Hardware Version**

**Binary release or compiled yourself?**

**Operating System (and version)**

Windows 10
**Web Browser (and version)**
Chrome

**Access Point**
wifi

**Describe the bug**
I tried loading FSEQ files after flashing the update that was just released.  It says it loaded, yet nothing comes up in the file management in the control setup.  I had it set to FPP remote when doing this.  I powered down the stick and removed the SD card and checked.  The FSEQ files are all there they are just not recognized by the device.  The original two that I downloaded back in June are there and show fine.  I tried to run the stick using the fppmaster and it would not play.  I then ran the fppmaster with one of the two FSEQ files that I could see and they ran fine.

MartinMueller2003, 2021-09-30:
Please attach one or two of the files you tried to download. I just downloaded a set of files and did not have an issue.

rbearwy, 2021-09-30:
The window below told me that it does not support that file type when I tried to drop my fseq file below

rbearwy, 2021-09-30:
[bobby_pickett_monster_mash.zip](https://github.com/forkineye/ESPixelStick/files/7261049/bobby_pickett_monster_mash.zip)

forkineye, 2021-09-30:
> [bobby_pickett_monster_mash.zip](https://github.com/forkineye/ESPixelStick/files/7261049/bobby_pickett_monster_mash.zip)

Those are compressed files, only sparse V2 uncompressed files are supported.

```
C:\Users\sporadic\Desktop\test>python fseqinfo.py *.fseq
Beetlejuice - Day-o.fseq:
  V2.0 compressed (zstd) 136780 channels with 2353 frames @ 50ms from xLights Windows 2021.29 64bit

bobby_pickett_monster_mash.fseq:
  V2.0 compressed (zstd) 136780 channels with 3522 frames @ 50ms from xLights Windows 2021.29 64bit
```

rbearwy, 2021-09-30:
I sent back a file through my email

rbearwy, 2021-09-30:
Github is rejecting the others as two big one is 6.5mb

forkineye, 2021-09-30:
> Github is rejecting the others as two big one is 6.5mb

You can send them to shelby@forkineye.com. I'll get them to Martin as well if needed.

rbearwy, 2021-09-30:
Ok sent them to shelby

forkineye, 2021-09-30:
> Ok sent them to shelby

Got them. Those are compressed as well and look like they're for your entire show. Those files must be sparse uncompressed v2 files which get created and uploaded by FPP connect, which we just found an issue with in beta2.  That is being tracked in issue https://github.com/forkineye/ESPixelStick/issues/367.  

```
C:\Users\sporadic\Desktop\test>python fseqinfo.py -v *.fseq

Beetlejuice - Day-o.fseq
------------------------
File ID:        PSEQ
Data Offset:    2088
Minor Version:  0
Major Version:  2
Header Length:  1936
Channel Count:  136780
Frame Count:    2353
Step Time(ms):  50
Flags:          0
Compression:    1 (zstd)
Comp Blocks:    238
Sparse Ranges:  0
FLags 2:        0
Unique ID:      1632694326446000
Media File:     G:\xlights\Downloads\Extreme Lights\xS_Beetlejuice - Day-O\xS_Beetlejuice - Day-O\_lost\Beetlejuice - Day-o.mp3
Created By:     xLights Windows 2021.29 64bit

john_carpenter_halloween_theme_-4329603140284364213.fseq
--------------------------------------------------------
File ID:        PSEQ
Data Offset:    2096
Minor Version:  0
Major Version:  2
Header Length:  1984
Channel Count:  143492
Frame Count:    2901
Step Time(ms):  50
Flags:          0
Compression:    1 (zstd)
Comp Blocks:    244
Sparse Ranges:  0
FLags 2:        0
Unique ID:      1632614243411000
Media File:     G:\xlights\Audio\john_carpenter_halloween_theme_-4329603140284364213.mp3
Created By:     xLights Windows 2021.29 64bit
```

rbearwy, 2021-09-30:
So how do I get those files.  Mine are all fseq, xlights program or xbck files.  Is there a way to convert the fseq to v2 files?

forkineye, 2021-09-30:
> So how do I get those files. Mine are all fseq, xlights program or xbck files. Is there a way to convert the fseq to v2 files?

xLights generates them for you.   They need to be uploaded via FPP Connect once you have your ESPixelStick configured as FPP Remote:
[image]

For standalone props that aren't controlled by your show, or off season effects, etc... You can setup a configuration for just that one prop / controller and change xLights via the preferences and use the fseq that xLights normally saves.  Note though, you don't want to do this for your current setup / show as it will have all the channel data for your show and the uncompressed file will be huge:
[image]

rbearwy, 2021-09-30:
That was what I was doing and did not see the files in the file manager. I reflashed with the tool and tried it again and no change.

forkineye, 2021-09-30:
> That was what I was doing and did not see the files in the file manager. I reflashed with the tool and tried it again and no change.

There is a bug I found today causes the files not to show if you refresh your browser while on the device setup tab. You have to navigate to another tab and back.  Would you be available on zoom later?

rbearwy, 2021-09-30:
yes I can zoom whenever you guys want

forkineye, 2021-09-30:
I'm on the xLights zoom in breakout room 1 right now.

rbearwy, 2021-09-30:
going there right now

MartinMueller2003, 2021-09-30:
FYI: Even if the files inside the zip were uncompressed sparse files, the zip file would still have been rejected. A zip file is NOT an fseq file. You can only upload a file with an FSEQ extension. Monster.fseq, foo.fseq, bar.fseq are all valid. MyFiles.zip will be rejected since the ESP cannot play a zip file.

MartinMueller2003, 2021-09-30:
Anyone have a link to the room? I get someone giving a tutorial on general usage.

patdelaney, 2021-09-30:
go to xlights.org  the link is on the right

forkineye, 2021-09-30:
Issue ended up being the SD card and was very odd. Card originally worked 3 months ago and he was able to upload files to it then. Those files still showed up. He was able to upload files via FPP Connect and via the web interface, but the ESP would not recognize them once written. He pulled the card and was able to see the files he had uploaded in Windows.  Re-inserted card into ESPixelStickV3 and they wouldn't show up.  Windows didn't automatically detect any errors with the card. Ran a force scan and Windows ejected the card saying it couldn't repair it.  Tried a new SD card and everything worked as expected.  Also flashed https://github.com/forkineye/ESPixelStick/actions/runs/1291838875.

closed 2021-09-30 by forkineye

rbearwy, 2021-09-30:
Thank you guys so much for the help.  Your company is truly customer service orientated. Something that is often hard to find in the Brick and mortar businesses

 
 

Closed #368.

MartinMueller2003, 2021-10-01:
Company? I am a Hobiest working on a fun project.