forkineye/ESPixelStick #371: [4.0-beta2] Refresh rate does not update when first enabling output

henrygab, 2021-09-30:
This one should be an easier fix.  😀 

--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0-beta2

**Hardware Version**
ESPixelStick v3

**Binary release or compiled yourself?**
Binary release

**Operating System (and version)**
Win10 64-bit (21H1 build 19043.1237)

**Web Browser (and version)**
94.0.992.31 (Official build) (64-bit)

**Access Point**
Ubiquiti Loco M2

**Describe the bug**

### Steps to Repro

1. Clean install 4.0-beta2 on ESPixelStick v3 hardware
2. Logon via WiFi
> <details><summary>Expand for screenshot</summary><P/>
>
> [image]
> </details>
3. Click on `Device Setup` tab
> <details><summary>Expand for screenshot</summary><P/>
>
> [image]
> </details>
4. Set primary input to DDP _[optional step]_
> <details><summary>Expand for screenshot</summary><P/>
>
> [image]
> </details>
5. Set output 1 to WS2812
> <details><summary>Expand for screenshot</summary><P/>
>
> [image]
> </details>

### Expected Results:

Refresh rate shown for default WS2812 configuration

### Actual Results:

Refresh rate does not initially appear (shows zero fps)

[image]

### Other details:

* If you do not save the config, this will reproduce easily by reloading the page.
* After enabling the output type, changing any setting related to `Output 1` will cause the refresh rate to update ... even for unrelated items such as `Gamma Value` or `Brightness`

forkineye, 2021-09-30:
That one was on mine todo list, was hoping to get to it before someone noticed :)

henrygab, 2021-09-30:
Amazing quantity of new features.  I'm quite impressed so far.

Low-pri request:  Consider providing ELF binaries / debug files with releases?  
Why: I've hit exceptions running Beta1, but didn't find the necessary files to decode it.
(yes, I will upload the exception next time, regardless)

forkineye, 2021-09-30:
> Amazing quantity of new features. I'm quite impressed so far.
> 
> Low-pri request: Consider providing ELF binaries / debug files with releases? Why: I've hit exceptions running Beta1, but didn't find the necessary files to decode it. (yes, I will upload the exception next time, regardless)

Thanks! The ELF files are in the Debug Files workflow artifact. For 4.0-beta2, it was this one - https://github.com/forkineye/ESPixelStick/actions/runs/1288169896

henrygab, 2021-09-30:
> We have been chasing a bug rooted in the asyncwebserver / websocket library that we're using.

Ah.  I was experiencing it when I was debugging the scripts, with breakpoints set in the validation routines. This cause fairly consistent loss of connectivity.   I offer this information as just one data point that may helps focus testing.

<details><summary>Steps (expandable)</summary><P/>

1. in edge, load `device setup` tab
2. enable `dev tools` by pressing F12
3. in dev tools, goto `sources` tab
4. click on `esps.js` file in left side panel
5. click `yes` to pretty-print the file
6. `CTRL-F` to find string `validity` to find function of interest
7. Right-click on line number for that function, choose `Add Breakpoint`
8. change device configuration to cause breakpoint
9. delay random time
10. resume script (I don't know keyboard shortcut... Click on play icon)
11. repeat steps 8-10

</details><HR/>

forkineye, 2021-10-01:
Fixed in https://github.com/forkineye/ESPixelStick/commit/e1774d04463ac1c35baedec807c2fb27cedca1f5 CI Build:  https://github.com/forkineye/ESPixelStick/actions/runs/1296427939

BTW, if you ever want to play around with editing the web side, you can change the scripts around in index.html and serve the pages with python, passing your device as target for the websocket stuff. Instructions here - https://github.com/forkineye/ESPixelStick/tree/main/html#development

henrygab, 2021-10-01:
> BTW, if you ever want to play around with editing the web side, you can change the scripts around in index.html and serve the pages with python, passing your device as target for the websocket stuff.

Well, now... that's a nifty feature!  Turns out Windows prevents the use of port 8000.  I've added PR https://github.com/forkineye/ESPixelStick/pull/374 to suggest use of port 8888 instead.

In addition, it appears `index.html` expects `esps.js` and `esps.css` to exist in the same directory.  files in the depot are in a subdirectory, while index.html expects those files to be in the root.  This adds steps to development / testing (and thus a potential place for errors to occur ... tested files not matching checked in files).

Q: Can you help me understand if this is intentional?  Do the development docs need an update?

forkineye, 2021-10-01:
That's odd that 8000 was blocked for you. I've worked on a few window machines and never experienced it. As for file structure, the js and css files are all processed via gulp to be minified and comments stripped. When setting up a dev environment, you're pointing to the sources.

henrygab, 2021-10-01:
Tested build 4.0-ci1296427939, and can happily confirm the issue is resolved.  Thanks!

_[[ As for dev:  OK.  I had just some hope that requiring edits, after testing from the dev environment, would be possible.  Sounds like it's not "low-hanging fruit" (easy to accomplish).  Thanks. ]]_

forkineye, 2021-10-01:
> Tested build 4.0-ci1296427939, and can happily confirm the issue is resolved. Thanks!
> 
> _[[ As for dev: OK. I had just some hope that requiring edits, after testing from the dev environment, would be possible. Sounds like it's not "low-hanging fruit" (easy to accomplish). Thanks. ]]_

Thanks! Yeah, I agree it's painful as we have to edit and restore index.html constantly. Without migrating to a more involved processing system for the web stuff we're stuck with it as is. I'm working on a reactjs replacement which is going to replace all of it anyways so we've just been dealing with for now.

closed 2021-10-01 by forkineye

henrygab, 2021-10-01:
So... for ESP32, with advanced mode enabled on admin tab...

There are no pins listed for output 7 (e.g., APA102) nor for output 8 (neither relay nor servo pca9685).

Is there any reason to not provide similar information on which pins are used for these?  I was going to file a bug, but if there's reason not to indicate which pins are being used for APA102's clock/data lines, or for I2C (for pca/relay), it'll save me filing a useless issue.

forkineye, 2021-10-02:
Relay has its own configuration page, can't recall the others though. Will have to check tomorrow. I'd open an issue on it or add to the one you have on grouping if somethings not right.

MartinMueller2003, 2021-10-02:
That is on purpose. Changing the spi pins is very complicated so it is not supported. You cant change the singular SPI pins without a recompile.

— the

forkineye, 2021-10-02:
Ah, ok. I couldn't recall if the pin mux on the esp32 supported that or not.

henrygab, 2021-10-02:
> That is on purpose. Changing the spi pins is very complicated so it is not supported. You cant change the singular SPI pins without a recompile.

Hi Martin,

Can you help me understand this more?

I understand that two sets are pre-defined as IOMUX ("direct") pins, theoretically supporting 80MHz.

At the same time, IIRC, any other pins could be used, although that means they'll go through the GPIO matrix, thus limiting SPI to theoretical 40MHz.  Since that's much faster than APA102 communicates at, I didn't think that would matter.

Seems I'm missing parts of this picture... Can you help me understand (or point me to stuff to read) to get me up to speed?  Thanks!