forkineye/ESPixelStick #372: [4.0-beta2] Device setup - Pin settings should be grouped

henrygab, 2021-09-30:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0-beta2

**Hardware Version**
ESPixelStick v3

**Binary release or compiled yourself?**
Binary release

**Operating System (and version)**
Win10 64-bit (21H1 build 19043.1237)

**Web Browser (and version)**
94.0.992.31 (Official build) (64-bit)

**Access Point**
Ubiquiti Loco M2

**Describe the bug**

### Steps to Repro

1. Clean install 4.0-beta2 on ESP32 board (4MB)
2. Logon via WiFi
3. On `Admin` tab, Enable "Advanced Mode" checkbox
4. Goto `Device Setup` tab

### Expected Results

The four SD-related pins all show in a group (similar to any other output)

### Actual Results

The four SD-related pins are grouped together, under a heading.
For an example, set `Primary Input` as `DDP`, and refer to the "DDP Configuration" section.

### Preferred Option

Put *ALL* pin selections into a single section.  When `Advanced Mode` is not enabled, only the `SD Chip Select Pin` is shown in this section.  When `Advanced Mode` is enabled, all pin settings should be shown in this single grouping.

MartinMueller2003, 2021-09-30:
The pin settings are not meant to be adjusted for the normal hardware. They should really be considered read only on the ESP8266. The ESP32 makes it possible to change many of the IO pin settings dynamically and that is when these screens are more useful.

henrygab, 2021-09-30:
@MartinMueller2003 ... isn't the SD's CS pin configurable, even on ESP8266?

Other than that minor item, I **_fully_** agree with you.  I'm getting excited to see the hardware that this ESP32 build is targeting, after looking at the outputs.  🙂

MartinMueller2003, 2021-09-30:
The ESP8266 is configurable but it is not reconfigurable. the config files tend to be read after an initial config has been set up. The means the ESP826 needs to have the pins defined at compile time. You can change them and recompile without issue.

henrygab, 2021-10-02:
[Per MartinMueller2003](https://github.com/forkineye/ESPixelStick/issues/371#issuecomment-932650321), the following pins are not dynamically configurable:

* (ESP8266 only) CS Pin for SD card
* (ESP32 only) SPI pins for APA102 (output 7)

I feel the pins should still be listed, even if not reconfigurable without recompilation.  I would recommend using the same layout, but setting input box to disabled.  Possibly with a tooltip (or the like) to indicate why they cannot be edited.

In addition, the pins used for output 8 appear to be I2C.  It's not clear which pins are used for output 8.  I feel these pins should also be listed.  If they are not reconfigurable without recompilation, see above (disabled but shown).

After all, the point is to show the advanced user information.  Presumably, others will want to add APA102, and need to know which pins get used.  Since on ESP32, any pins can be used(*) for SPI and I2C, this seems like stuff needed to expose when "advanced" mode is enabled.

(*) Ok, so the output pins can't use 34+.  And SPI is limited to 1/2 the ESP's max theoretical speed because it goes through the MUX.  But, both **could** be configured to use different pin (at least at next boot...).

forkineye, 2021-10-02:
Actually, CS is the only one configurable on ESP8266 since we use the hardware SPI for SD card access and UART for pixel / serial output. The default pins are listed here - https://github.com/forkineye/ESPixelStick/wiki/Supported-Inputs-and-Outputs#output-pin-mapping

closed 2022-05-02 by henrygab