forkineye/ESPixelStick #375: unable to assign GPIO0/D3 as a valid relay output pin

garymueller, 2021-10-02:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
head

**Hardware Version**
wemos mini D1

**Binary release or compiled yourself?**
compiled

**Operating System (and version)**
windows 10

**Web Browser (and version)**
google chrome Version 94.0.4606.61 (Official Build) (64-bit)

**Access Point**
??

**Describe the bug**
Unable to assign GPIO0 as a valid option for relay output. This is due to the Relay_DEFAULT_GPIO_ID being assigned to 0 which is also the value of GPIO0.  The validate function compares the gpioID to the Relay_DEFAULT_GPIO_ID and if they are the same disables the relay.  Solution is to use a value for default pin that does not overlap with actual pin values.  I tried -1 and it worked ... but the front end didnt appear to like it.

MartinMueller2003, 2021-10-02:
@Gary. Any relationship to Arnold and Meta Mueller?
Martin

garymueller, 2021-10-02:
no relation that I'm aware of. Mueller ancestry runs through Illinois/Michigan areas.

closed 2021-11-13 by forkineye

Panzercou, 2022-01-04:
Will the wemos d1 mini have more capacity than 1200 leds?
Can't you put 10,000 leds?

Panzercou, 2022-01-04:
How can I change the pin, or the number of leds?