forkineye/ESPixelStick #377: GRB order issue

Crabilouse, 2021-10-03:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
ESPixelStick 4.0-beta3

**Hardware Version**
ESP8266

**Binary release or compiled yourself?**
Binary release

**Operating System (and version)**
Windows 10

**Web Browser (and version)**
Chrome

**Access Point**
WIFI

**Describe the bug**
The "Color Order" option will always revert back to RGB, even after saving changes as GRB. This happens after a short period of time, but never immediately. All other options are saved correctly.

Crabilouse, 2021-10-03:
> Are you uploading outputs via xLights as well? If so, that will overwrite the color order as it'll use the controller color order connected to the model.

For some reason it seems that I still need to change the controller (ESP8266) via the web interface to GRB for xlights to input the correct colors, but yes I output from xLights. How would I fix this exactly? Thank you

forkineye, 2021-10-04:
If you're using xLights to manage everything, this is where you set the color order on the model for upload to the controller:
[image]