forkineye/ESPixelStick #410: Add Support for TTGO-T8 ESP32 board

thomastech, 2021-11-11:
Getting the TTGO-T8 ESP32 board to work with ESPixelStick requires very minor code changes. It would be great to add official support for it in a future release.

The TTGO-T8 has several desirable features for use with ESPixelStick firmware. It is inexpensive ($7-10 USD) so that checks-off one important box. But hardware wise, it has 4MB to 16MB Flash, Micro SD Card Socket, onboard antenna, U.FL socket for external antenna (optional feature, just move bridge resistor), and plenty of GPIO.

Typical vendor link:
https://www.aliexpress.com/item/32851423415.html

[image]
I'm new to the ESPixelStick project (just discovered it last week). So I don't know the expected procedures to follow in regards to Github pull requests. But since the code impact is minor, I will just share my changes in code braces, see below.

BTW, I tested with a V1.7 board and a modified 4.x-dev branch. The update should work with any TTG0-T8 hardware (flash size 4MB / 8MB / 16MB, board rev V1.1 / V1.3 / V1.7). I found that a level shifter IC (typically 74HCT125) is required on the data pins to avoid LED flicker, even on short cable lengths.

_Code changes begin here:_

**1. platformio.ini file**
At the bottom of platformio.ini add this new board env:
```
; ESP32 TTGO-T8 V1.x
[env:esp32_ttgo_t8]
extends = esp32git      ; use until 2.0.0 core makes it into PlatformIO
board = wemos_d1_mini32 ; use until platformio adds TTGO-T8
monitor_rts = 0
monitor_dtr = 0
build_flags =
    -DESP32_TTGO_T8
```
**2. FileMgr.cpp**
Find this code in FileMgr.cpp:
```
#ifdef ARDUINO_ARCH_ESP32
    SPI.begin (clk_pin, miso_pin, mosi_pin, cs_pin);
```

Replace it with this:
```
#ifdef ARDUINO_ARCH_ESP32
    SPI.begin (clk_pin, miso_pin, mosi_pin, cs_pin);
    pinMode (miso_pin, INPUT_PULLUP);  // MISO is missing required pull-up resistor, use internal pull-up.
```
**3. GPIO_Defs.hpp**
Find this code in GPIP_Defs.hpp:
```
#elif defined(ESP32_CAM)
//Output Manager
#define DEFAULT_UART_1_GPIO     gpio_num_t::GPIO_NUM_0
#define DEFAULT_UART_2_GPIO     gpio_num_t::GPIO_NUM_1
#define DEFAULT_RMT_0_GPIO      gpio_num_t::GPIO_NUM_3
#define DEFAULT_RMT_1_GPIO      gpio_num_t::GPIO_NUM_16
#define LED_GPIO                gpio_num_t::GPIO_NUM_4

// File Manager
#define SD_CARD_MISO_PIN        gpio_num_t::GPIO_NUM_2
#define SD_CARD_MOSI_PIN        gpio_num_t::GPIO_NUM_15
#define SD_CARD_CLK_PIN         gpio_num_t::GPIO_NUM_14
#define SD_CARD_CS_PIN          gpio_num_t::GPIO_NUM_13
```

Replace it with:
```
#elif defined(ESP32_CAM)
//Output Manager
#define DEFAULT_UART_1_GPIO     gpio_num_t::GPIO_NUM_0
#define DEFAULT_UART_2_GPIO     gpio_num_t::GPIO_NUM_1
#define DEFAULT_RMT_0_GPIO      gpio_num_t::GPIO_NUM_3
#define DEFAULT_RMT_1_GPIO      gpio_num_t::GPIO_NUM_16
#define LED_GPIO                gpio_num_t::GPIO_NUM_4

// File Manager
#define SD_CARD_MISO_PIN        gpio_num_t::GPIO_NUM_2
#define SD_CARD_MOSI_PIN        gpio_num_t::GPIO_NUM_15
#define SD_CARD_CLK_PIN         gpio_num_t::GPIO_NUM_14
#define SD_CARD_CS_PIN          gpio_num_t::GPIO_NUM_13

#elif defined(ESP32_TTGO_T8)
//Output Manager
#define DEFAULT_UART_1_GPIO     gpio_num_t::GPIO_NUM_0
#define DEFAULT_UART_2_GPIO     gpio_num_t::GPIO_NUM_4
#define DEFAULT_RMT_0_GPIO      gpio_num_t::GPIO_NUM_25
#define DEFAULT_RMT_1_GPIO      gpio_num_t::GPIO_NUM_26
#define DEFAULT_RMT_2_GPIO      gpio_num_t::GPIO_NUM_27
#define DEFAULT_RMT_3_GPIO      gpio_num_t::GPIO_NUM_14
#define LED_SDA                 gpio_num_t::GPIO_NUM_21  // Green LED and SDA. Will light-up if PCA9865 is used.

// File Manager
#define SD_CARD_MISO_PIN        gpio_num_t::GPIO_NUM_2
#define SD_CARD_MOSI_PIN        gpio_num_t::GPIO_NUM_15
#define SD_CARD_CLK_PIN         gpio_num_t::GPIO_NUM_14
#define SD_CARD_CS_PIN          gpio_num_t::GPIO_NUM_13
```
_End of code changes._

Build with Platformio using the ESP32_TTGO_T8 board env. Enjoy!

- Thomas

MartinMueller2003, 2021-11-15:
Changes have been pushed. My implementation differs from yours in some minor ways. I also restructured how GPIO_Defs is put together. If we continue to add platforms, that file would get very messy. In the new model, each platform has its own include file and the GPIO_Defs file is a selector.

thomastech, 2021-11-15:
Thanks for adding the TTGO T8. The restructured GPIO_DEFS will definitely keep things tidy as the supported platforms grows.

I don't believe the relocation of the MISO pullup is correct. After reviewing the SPI library it looks like SPI.begin() will overwrite the added MISO pin declaration. That is to say, the library will change the MISO pin back to a input without pullup. 

TLDR; The MISO pullup should go directly after SPI.begin(), before SD.begin() is called.

BTW, this pullup was the magic ingredient to getting the TTGO T8's SD card to work. It seems that TTGO decided not to include the customary 10K ohm MISO pullup on this board's design. The resistor is shown on the schematic, but is marked as "NC", so they deliberately do not stuff it. Bad decision.

It took an afternoon of hair pulling to find this mistake. Using the internal weak pullup is working fine, so no need to solder an external resistor.

- Thomas

forkineye, 2021-11-15:
> Changes have been pushed. My implementation differs from yours in some minor ways. I also restructured how GPIO_Defs is put together. If we continue to add platforms, that file would get very messy. In the new model, each platform has its own include file and the GPIO_Defs file is a selector.

I think we should draw the line at only adding ESP32 platforms that support SD cards as well. The generic definition should catch the rest.

MartinMueller2003, 2021-11-15:
In this case the card has a built in SD Card that might lead people to get frustrated. I have moved the pullup statement.

thomastech, 2021-11-15:
>  I have moved the pullup statement.

Looks good. Thanks.

- Thomas

closed 2021-11-17 by forkineye