forkineye/ESPixelStick #413: FSEQ Restarting

Almcknight, 2021-11-17:
V3 running 4.0-beta4

Not sure if it's a bug or design parameter/limitation: 

Symptom 1: FSEQ file appears to repeatedly restart when WebSocket reestablishes connection.

Symptom 2: Standalone prop (not intended for network connection after programming) with a 10 minute sequence doesn't make it past the first 2 minute effect within the sequence if the V3 is out of range of the programmed WiFi network.

MartinMueller2003, 2021-11-17:
Would you please attach the fseq file you are trying to play. I have some pretty long files but the file system does have a limit of about 2gb for a single file.

If the ESP stays within range of the WiFi, will it play beyond the 2 minute mark (as in plays the entire sequence)?

Almcknight, 2021-11-17:
If the ESP stays in range I believe there is still the occasional reset, albeit much more random. It seems especially prevalent when trying to access the web GUI, but it's not predictably repeatable. 

I'd imagine it's connected to a similar issue thread I read through last night. What may be different about my situation is that I'm specifically trying to use it out of WiFi range (program at home, take to office). To a layperson like me, it seems like the WiFi times out while trying to connect and a key process resets, restarting the FSEQ file.

[1-String-Christmas.zip](https://github.com/forkineye/ESPixelStick/files/7555964/1-String-Christmas.zip)

MartinMueller2003, 2021-11-17:
Symptom 1: FSEQ file appears to repeatedly restart when WebSocket reestablishes connection. - I cannot reproduce this.

Almcknight, 2021-11-17:
> Symptom 1: FSEQ file appears to repeatedly restart when WebSocket reestablishes connection. - I cannot reproduce this.

I opened up my browser (Chrome on Windows 11), connected to the device, and simply clicked through each GUI setting header and saw the WebSocket connection dialog and the device reset, starting the FSEQ over again. It doesn't always happen.

The power supply is 5v 2A, the string of lights is 50 pixels set to 30% brightness, so I'm assuming it's not a voltage drop issue.

Almcknight, 2021-11-17:
Is there a device configuration that would be most conducive for a standalone prop with no WiFi connection after being programmed?

forkineye, 2021-11-17:
> Is there a device configuration that would be most conducive for a standalone prop with no WiFi connection after being programmed?

Go to the WiFi Setup tab and un-check "AP Fallback" and "AP ReBoot".   It's been a while since I tested it, but I ran a dance prop with a 10 minute sequence with no issues.  Depending on the quality of the power supply, it could be attributing to your issues as well.  If you have a larger one to test with, I'd try it.  Also, I noticed you're using xLights 2020.51.  You'll want to upgrade to the latest if you want to upload via FPP Connect.

Almcknight, 2021-11-17:
Thanks to both of you. I'll do some more testing and troubleshooting as needed.

I hope this was helpful to development as opposed to being a time waster; I had noticed someone else had a somewhat similar open issue with reboots.

MartinMueller2003, 2021-11-18:
FSEQ does not use websockets. Those reset all the time. Reboot on no connect WILL cause you problems.

MartinMueller2003, 2021-11-18:
Changes to fix a bug for long running fseq files has been pushed and will be available shortly.

MartinMueller2003, 2021-11-19:
I am marking this as wont fix since the issue you were seeing is not caused by the issues I found while investigating.  We know we get random reboots when using the Web UI (there are tickets for that). It looks like your reboots while not in range are most likely caused by intentional reboots by the device to try to get a connection. Turning that feature off should leave you in a stable state.

forkineye, 2021-11-19:
@Almcknight  The web issue is being tracked [here](https://github.com/forkineye/ESPixelStick/issues/378). Let us know if the configuration change helped with your sequence playback.
thanks,
-shelby

Almcknight, 2021-11-19:
To help isolate the issue I changed my V3's WiFi network to a temporary router I use with FPP during the Christmas season along with turning off the AP Reboot function and that appears to have quelled the restart issue. 

With the latest version of xLights and a newly saved FSEQ I am getting an occasional albeit very brief "flash" from the lights that I hadn't seen before with the old FSEQ file. (Newest FSEQ attached)

[1-String-Christmas (2).zip](https://github.com/forkineye/ESPixelStick/files/7571622/1-String-Christmas.2.zip)

forkineye, 2021-11-19:
> To help isolate the issue I changed my V3's WiFi network to a temporary router I use with FPP during the Christmas season along with turning off the AP Reboot function and that appears to have [...]

Is the flash occurring at a particular point in the sequence? Can you post a video of it?

Almcknight, 2021-11-19:
It's almost like the sequence goes blank for a fraction of a second and keeps going. There is no apparent interval with which it repeats.

Almcknight, 2021-11-20:
Another variable I personally introduced into the mix was trying to get the Refresh Rate on the Device Setup tab closer to my actual sequence FPS. With 50 pixels and an Inter Frame Time at 300 microseconds the Refresh Rate shows as 2ms / 555fps, which is simply invalid. Not really understanding what Inter Frame Time was, I had adjusted the settings to 500 pixels at 10,000 microseconds to bring the calculated FPS down to 40. After resetting to 300 and disregarding the calculated FPS, the flicker seems to have disappeared. 

I believe this caused the flicker, while the AP Reboot was causing the reset.

closed 2021-12-28 by forkineye