forkineye/ESPixelStick #415: Display Format Issue When Using SD Card for Matrix Playback

thomastech, 2021-11-18:
Configuration: Self-compiled 4.x-dev on a EPS32 (TTGO T8). 

Channel 1 has a pair of WS2811 8x32 Matrix LEDs that displays scrolling text messages. The Sequencer plays correctly from xLights and xSchedule (proper character format and orientation). But when I play the FSEQ file from the ESPixel's SD Card the characters are upside down and mirrored.

See screenshot below for ESPixelStick's Device setup:

[image]
:

Here's a photo comparison:

[image]

- Thomas

Kamal-Sonani, 2021-11-18:
Verify your starting position

thomastech, 2021-11-18:
Starting position in xLights is set to bottom right. Chase pattern in ESPixelStick confirms this is the first pixel location.  Pixels move up, then zig zag to bottom of next row, working towards the left.

I suppose I could remount my LED matrix (rotate 180 degrees) and experiment with xLights so that ESPixelStick's SD Card playback agrees with it. But I'm hoping there's a magic feature in ESPixelStick to do this.

- Thomas

Kamal-Sonani, 2021-11-18:
If you can sacrifice first 8 led(will not show) 
Declare start null value = 8  in output configuration

May be their is some bug but you can  try this trick temporarily

thomastech, 2021-11-18:
Thanks for the advice. Yes, setting ESPixel's start NULL count to 8 corrects the orientation when the SD Card provides the sequence. But now the text from xLights (and xSchedule) is upside down. All these sequence sources need to play nice together.

I noticed what may be an important clue while troubleshooting the problem. With Start NULL count set to zero, during SD Card playback the matrix's first 5 columns (40 LEDs) are always blank. So that explains the weird orientation. 

BTW, ESPixel's Effects function works correctly. That is to say, no missing pixels.

Maybe this new bit of info is helpful in fixing the bug.

- Thomas

forkineye, 2021-11-18:
How are you getting the FSEQ onto the ESPixelStick?

Kamal-Sonani, 2021-11-18:
Send me image of node layout & channel info(starting)

thomastech, 2021-11-18:
I usually drop it into the ESPixel's File Management page. But I've also used Xlights FPP Connect->Upload, as well as sneaker net (copy to SD card using PC, walk it back to the ESPixel). 

I just tried several other xLights effects on the Matrix. All ignore the first five columns when SD Card is used.

And after the latest experiments another unusual symptom has appeared that wont go away. The SD Card's first five columns (40 pixels) are no longer blank. Instead, they have active data meant for another model on a different ESPixel device.

Those first 40 pixels are possessed by the devil.

- Thomas

thomastech, 2021-11-18:
@Kamal-Sonani

[image]

forkineye, 2021-11-18:
The problem is the fseq. When using multiple controllers they must be sparse files. What you're seeing is the 120 channels of data from your other controller. You have to use FPP Connect, it will upload the sparse files.

thomastech, 2021-11-18:
> The problem is the fseq. When using multiple controllers they must be sparse files.

@forkineye
Well that's embarrassing. I thought I tested it with FPP Connect during yesterday's hair pulling.  But that's all it was. 

I didn't realize that the other channels must not be present in the fseq file. And that FPP Connect is the only way to save the file on the SD card.

Would be convenient if there was some sort of "filter" in ESPixelStick firmware that stripped out the unwanted channel data. 

This issue is solved. Not a bug. Just a newb user on the keyboard.

- Thomas

forkineye, 2021-11-18:
With the limited resources available on the ESP's, we can't support the current fseq compression options and uncompressed files would be very large with full show data in it.  That's why the firmware only expects data for itself in the fseq. FPP Connect chunks up the fseq so that each controller only has data that's meant for it.  If you ever use Vixen 3, you can accomplish this by exporting fseq's per controller in their exporter.

closed 2021-11-18 by forkineye