forkineye/ESPixelStick #426: 4.0-beta3 DDP generates errors.

tmblue, 2021-11-27:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0-beta3

**Hardware Version**
2

**Binary release or compiled yourself?**
Binary

**Operating System (and version)**
MacOS Catalina

**Web Browser (and version)**
Chrome

**Access Point**

**Describe the bug**
DDP generates errors. Zero clean packets Xlights  2021.35

If I change to e1.31 I can get clean data, if I select DDP (no configuration available), each incremental inbound packet generates an error.

Not sure if this is related to the naming confusion that was pointed out earlier, but it would be odd for xlights to be using the hostname for anything.  The packets should be going to the IP via unicast, which it is, since I see the packets but the Stick, believes everything is coming in as an error.

forkineye, 2021-11-27:
Do you have ```Keep Channel Numbers``` checked in the xLights controller configuration?  It needs to be un-checked if so.

tmblue, 2021-11-27:
I don't actually. lights are outputting but no idea why the error count is
so high

DDP Status
Total Packets: 529
Total KBytes: 268.6328
Errors:                             529

Interesting tidbit. xlights test mode, have no channels selected, test is
off and I see packets/error packets incrementing as soon as i output to
lights.  Why is xlights and what is xlights sending to the ESP, when
nothing is selected?

tmblue, 2021-11-27:
Figured it out, I was being stupid and thus caused you some spleen damage.

xlights was configured as a full universe, but I set my ESP to be just the
number of lights that I had connected, so ya, it was receiving data it
didn't expect.  and thus the error count, sending 170 channels worth of
data to 16 channels and well..

bah , sorry man!

MartinMueller2003, 2021-11-30:
Good to hear. I was about to go and try to reproduce this. I suppose we can close the issue then.

closed 2021-12-28 by forkineye