forkineye/ESPixelStick #427: FPP Connect Multiple Files Corruption

marksat666, 2021-11-27:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0-ci1507513487

**Hardware Version**
Lolin ESP32 PRO

**Binary release or compiled yourself?**
Binary release

**Operating System (and version)**
Windows 10.0.19043

**Web Browser (and version)**
MS Edge Version 96.0.1054.34 (Official build) (64-bit)

**Access Point**
ASUS R7000 (router, dedicated to the props, no Internet)

**Describe the bug**
When I try to upload multiple sequences via FPP Connect, some of the sequences will not produce output; it looks like it is playing but there is no data going to the pixels. Another tell-tale sign is that the Time Remaining on the Local File Player Status screen is crazy large, like 71582788:02 (screenshot attached). Also, sometimes the file timestamps have a weird date, like 1980.

I can delete the non-working sequences from the ESPixelStick, reboot it, then upload the sequences individually and usually get things to work (sometimes I have to delete/re-upload a couple of times).

There is no consistency on which sequences do not work, and I am having the same issue on both of my ESP32 devices (oddly enough, this does not happen on my "genuine" ESPixelStick hardware ESP8266). Network strength is good (-47db/100%) - I have tested this with a fresh firmware flash and the ESP32 and my XLights laptop 10 feet from the router.

Additionally, I came home tonight and it was happening again with a sequence that were previously working today - it was showing as playing but had the crazy Time Remaining values. I rebooted both ESP32s and they started playing the sequence properly.

Attached is the screenshot of the abnormal Time Remaining value, also two sequencences that have had the issue
[image]
[Incandescent.zip](https://github.com/forkineye/ESPixelStick/files/7611935/Incandescent.zip)
[You Make It Feel Like Christmas.zip](https://github.com/forkineye/ESPixelStick/files/7611936/You.Make.It.Feel.Like.Christmas.zip)
,

forkineye, 2021-11-27:
When this happens again, can you pull the fseq file from the SD card and post it? If you can't pull the card and copy it off, you can download it by using ```/download/sequence_name.fseq``` as the url.

MartinMueller2003, 2021-11-27:
I assume Incandescent.zip and You make it feel like Christmas.zip are files that have these issues?

marksat666, 2021-11-28:
Here are a few sequences from the ESP32 SD card - "All I Want for Christmas Is You" was particulary troublesome (took four times to get it to load/play properly after the issue). Some of them look odly small in size - and I CAN eventually get them to copy to the SD card (requires deleting the bad on and a reboot of the ESP32
[Problem Sequences.zip](https://github.com/forkineye/ESPixelStick/files/7612679/Problem.Sequences.zip)
).

MartinMueller2003, 2021-11-28:
The file you sent me plays fine in local play mode. Have you tried playing the "bad" files locally (no FPP master)?

The parsed data below indicates the file is fine. I suspect you may have an unreliable SD card or your SD Card interface is not wired correctly. There are some platforms that are missing a pullup on some of the lines.

10:45:20.739 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:271: Start ------
10:45:20.832 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:289: FileHandleForFileBeingPlayed: 44481 ------
10:45:20.877 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:293:                     BytesRead: 32 ------
10:45:20.877 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:294:         sizeof (fsqRawHeader): 32 ------
10:45:20.924 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:303: Convert Raw Header into a processed header ------
10:45:20.971 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:321:                    dataOffset: 220 ------
10:45:20.971 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:322:                  minorVersion: 0 ------
10:45:20.971 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:323:                  majorVersion: 2 ------
10:45:21.017 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:324:             VariableHdrOffset: 32 ------
10:45:21.017 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:325:                  channelCount: 5632 ------
10:45:21.017 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:326: TotalNumberOfFramesInSequence: 3184 ------
10:45:21.064 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:327:                      stepTime: 50 ------
10:45:21.064 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:328:                         flags: 0 ------
10:45:21.064 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:329:               compressionType: 0x0 ------
10:45:21.064 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:330:           numCompressedBlocks: 0 ------
10:45:21.064 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:331:               numSparseRanges: 0 ------
10:45:21.111 -> ------ ParseFseqFile:InputFPPRemotePlayFile.cpp:332:                        flags2: 0 ------

MartinMueller2003, 2021-11-28:
And it plays just fine in remote mode as well.

marksat666, 2021-11-28:
I tested first with the local playback mode (if it plays OK that way, then it plays ok remote).  I originally thought it was an SD card issue, and I have tried three different brands with the same result (formatted with an SD Card Formatter app and also with the regular Windows format).  Again, they appear to be playing but the Time Remaining is whacky and there is no output (real or on the diagnostic screen).

I am using the Lolin ESP32 Pro boards with the SDC card already there - I am not sure how to check for a missing pullup.  Is there a proven reliable ESP32 board for this application?  The Lolin was attractive since it has the built-in SD card holder and also the micro USB port; I see some CAM boards that come with a shield that has the USB port - perhaps those are a better option?

Anyway, this isn't quite a deal-breaker as I can eventually get the sequences loaded (just have to do it one at a time) - I will have to keep an eye out to see if they geek out in the middle of the evening.

Thank you for your help.

MartinMueller2003, 2021-11-30:
I also use the LoLin ESP32 Pro without any issues.

marksat666, 2021-12-15:
As an update (after watching the show run for a few days) - the Lolin ESP32 controllers randomly stop working properly; they get WAY out of synch (sometimes running a totally different sequence) and also report that oddly large Time Remaining values.  In general they work properly, but this does happen.  Rebooting the malfunctioning controller fixes the problem, and it is not consistently the same sequence nor the same controller.  My instinct was to look at the network for some weirdness (the first Lolin is only 6 feet away from the router, and I have even pulled it out of the ammo box to eliminate electrical noise from the equation). 
Even though the RSSI numbers are good (-49dBm / 100%), ping latency is high, 400+ ms (sometimes over 700ms).  That pinging is with the FPP computer shut down (no sequences running), and nothing on the network besides three controllers and the pinging laptop (the show network is totally separate from my home network, no Internet connection). The ping latency to the ESP8266 controller is consistently 30-40 ms or lower (even though it is much farther away from the router), and the ping latency is single digits to the router itself.

So... my thought is to try some other ESP32 module (maybe I got a bad batch?) - any other ideas?

MartinMueller2003, 2021-12-19:
I did see this happen once and have not gotten it to reproduce. There is something going on, just need to keep poking at it.

MartinMueller2003, 2021-12-20:
I have found an issue that causes the ESP to output odd information when it cannot open/read the fseq file properly. There is an error message that is reported by the ESP in the status messages that is intentionally NOT displayed on the UI. You can see the message by going to your browser and displaying the JavaScript console messages. On chrome you press F12->console->verbose and the messages will start scrolling. Capture the output after the ESP has an issue (and before rebooting). Here is an example. In my example the ESP is unable to locate one of the files due to a bug in the FPP that mangles the name by loosing an apostrophe. 

[image]

MartinMueller2003, 2021-12-20:
Code for making sure stats are zero when there is a file error has been pushed to git. This only addresses the display oddity, not the reason for the problem.

MartinMueller2003, 2021-12-22:
I just pushed some code that will display the error message when the UI is in the advanced mode.

MartinMueller2003, 2021-12-22:
I suspect the issue is in your SD card hardware. The V3 daughter card has a solid implementation and is not giving you problems. It sounds like your other setups have a bit if DIY hardware that may not be working well at the speeds needed by the ESP. Check to make sure all of your pullups are in place and that the GPIO voltage level shifters are working properly. SD cards like to run at 5v. At this point the odd number issue is resolved. I think we can close this ticket.

marksat666, 2021-12-24:
So, after looking everything over (the Lolin ESP32 Pro boards have the SD card slot built-in and the solder connections look good) I saw weird dates (1980) on some of the sequence files on the controllers (again not consistent from controller to controller). I painstakingly re-uploaded all of the sequences ensuring that the dates on the files were correct on each controller (I still have to upload one sequence at a time, and sometimes re-upload on one or both of the ESP32s to get it uploaded properly).  I also reverted back to the router I used last year (yes, I know - changing too much at once).  The show has been stable/reliable since I did all of that.  I will flash the boards with the latest firmware after Christmas and see if the uploading is better (if not, I will plan on using something else next year).

MartinMueller2003, 2021-12-24:
Uploading to the ESP is slow and there is not much we can do about it. The WebServer code is part of the libraries we bring in and integrate. The other slowdown is the interface into the SD card. It is only using a single data pin. Super fast SD card implementations use Quad Data transfers but that uses a lot more IO pins.

forkineye, 2021-12-28:
Closed by mistake as there are a few things I want to look into first.   As for upload speeds, the main bottleneck is the size of the buffers that LWIP uses internally, more specifically, the MSS buffer. Even though we're using SPI to do the SD card transfers, that part is actually very fast. It's the MSS buffer size (which we can't make any larger) and internal wait states within LWIP that causes the transfers to take so long.  SD transfers are merely a blip on a logic analyzer compared to the idle time spent by the wifi stack trying to shuffle the data across.

DrAnonymous, 2021-12-28:
@marksat666 you said you were getting 400+ ms ping times. Was that right after a boot up before trying to play a show?

marksat666, 2021-12-30:
@DrAnonymous - the ping latency was strange, high then low, high then low.  It is better now (higher on the first, then settles down) with an <older> router.  This was with no sequences playing.

I have re-flashed one of the Lolin ESP32 Pros with FW Version4.0-ci1631009707.

To test I did a mulitple sequence load to just the one controller. The date and time stamps looked correct, however none of the newly loaded sequences would play - neither via FPP nor just making it play stand-alone; the Time Elapsed and Time Remaining both just stayed at 0:00 (if you recall, when a sequence was not working the Time Remaining was a ridiculously high number with the prior version).

I had to delete and re-upload them (via XLights FPP Connect) one at a time.  One of the sequences will not load/run at all now (was working before) - I tried to load it around 10 times (deleting the bad, rebooting the controller, rebooting the XLights computer, re-rendering, et al).  I took that one out of the playlist for now and I will try getting that one loaded tomorrow (perhaps try the drag and drop load instead of FPP Connect?).

DrAnonymous, 2021-12-30:
Can you send a link to the Lolin board you have and I'll order one and then go from there.

marksat666, 2021-12-30:
Here are the ESP boards that I bought:

https://www.amazon.com/Genuine-Original-LOLIN-D32-V2-0-0/dp/B07QDFP3WC/ref=sr_1_4?crid=1DF2FXEE443JX&keywords=lolin+esp32&qid=1640905632&sprefix=lolin+esp32%2Caps%2C70&sr=8-4

thomastech, 2021-12-31:
I'm uploading via XLights FPP Connect and have trouble saving the files too. I find that it takes several attempts to get a valid sequence file. There have been times I've had to repeat the upload a dozen times before success. Tried different cards, no joy.

If the uploaded file is corrupt then its size will be smaller than expected. My workaround is to first observe the file size on my desktop. Then after an upload I check to see if ESPixelStick reports nearly the same size (will be a few bytes different). The final check is to use the "FSEQ File To Play" drop-down and manually play from the ESPixelStick. 

BTW, never had an issue with FSEQ file playback, only file saving.

My ESPixelStick hardware uses TTGO-T8 ESP32, Three built, all experience the problem. I'm using self-compiled ESPixelStick firmware that was the latest 4.x beta version at the time I downloaded it late Nov-2021.

I don't know this information pertains to the OP's reported issue, but is provided in case it provides any clues to the problem. 

- Thomas

MartinMueller2003, 2022-01-01:
Interesting. I do not have any issues with a download not sending the expected data. The ESP uses the file transfer mechanisms built into the web services library. I will check to see if there are any bugs reported

DrAnonymous, 2022-01-05:
I ordered a couple ESP32 boards and will report back once they arrive and I've been able to do some testing.

DrAnonymous, 2022-01-06:
ESP32 boards came in. :) I only had time for limited testing tonight. With the Lolin I didn't see anything out of the ordinary, but I do need to get a flash card for more testing (thought I had some spares laying around).
Version:
[    ESP] ESPixelStick v4.0-ci1604277892 (Dec 20 2021 - 22:39:24)
[    ESP] v4.4-dev-2313-gc69f0ec32

Setup A: Connected to home wifi router / multiple devices connected to router via wifi and hardwired / TV streaming
- Average ping from computer to ESP was 227ms with a max of 516ms
- Accessing ESP webserver didn't seem to affect anything

Setup B: ESP setup as wifi access point / single laptop connection
- Average ping from laptop was 5ms with max of 55ms
- No change in times when clicking on webpages and adjusting settings
- Checked again after system was on 30 minutes and saw same behaviors

I'll get a flash card and switch to the same SW version and see what happens. I'll try the TTGO board as well.

thomastech, 2022-01-11:
> Setup A: Connected to home wifi router / multiple devices connected to router via wifi and hardwired / TV streaming
    Average ping from computer to ESP was 227ms with a max of 516ms

I found the reason for the horrible ping times, which also invite latency in other situations. Rather than muddy up this issue ticket I'll create another one that has a suggested fix. Spoiler alert: After the fix, average Ping times will be about 5mS.

- Thomas

thomastech, 2022-01-11:
> Interesting. I do not have any issues with a download not sending the expected data. The ESP uses the file transfer mechanisms built into the web services library. I will check to see if there are any bugs reported

I spent several hours investigating my ESP32 file upload problems. The test configuration is as follows:
```
ESP32 TTGO-T8
Primary Input=DDP
Output 1= WS2811
Secondary Input=FPP Remote
SEQ File To Play=PlayRemoteSequence
Static IP
fseq File size = 14551767 bytes
xLights V2021.40, uploads via its FPP connect tool
```
Under this configuration upload success is very poor. Perhaps 1 in 10 uploads are OK.

But I found the winning workaround, at least for me. If I disable the _Primary Input_  the uploads are 100% successful. Hopefully this will be an important clue to the devs.

- Thomas

marksat666, 2022-01-17:
Did another test, same results

XLights ver 2022.02 64bit
ESPixelStick FW Version4.0-ci1699645625

Primary Input was already disabled (as usual for my setup).  I have taken down my outdoor show, so I can do whatever testing is necessary.
I wonder if there is a way to copy the sequence file directly to the TF card inserted on the PC to bypass FPP Connect altogether (maybe give us a clue)?

Let me know if there is something else that I should try/test.

MartinMueller2003, 2022-01-18:
I have been downloading fseq updates to my ESPs all season. I use the file drag and drop via a browser. It seems to me that xLights may be doing something odd. Try to have xLights put the file onto your hard drive and then drag it to the ESP via the gui and see if there are still issues.

marksat666, 2022-01-19:
Did some testing with the drag and drop - pretty much the same results; some work well and others required multiple uploads or did not work altogether (one was resolved using the FPP Connect instead).  There was one sequence that I had to drop from the show mid-season because I could not get it to re-load (via FPP Connect) - that one too would not work with the drag & drop.  

What DID work with the "problem sequence" was coping the fseq file directly to the TF card using my laptop, then putting the card back into the ESP32.  Of course the file is larger on the TF card that way (and that deployment method would be a bit of a pain if the props/controllers are already deployed outside).  Maybe that provides some direction for troubleshooting...

MartinMueller2003, 2022-01-19:
Larger simply means the entire file made it. Yes bringing the SD card in so you can update it is not a good solution.

There may be something odd in the http file transfer libraries. very little of the code I have written is involved in the file transfer. It is all handled by the HTTP Async Server code.  FYI: I have never had drag and drop not work. What is the signal strength your device is getting?

marksat666, 2022-01-19:
I have the controller and laptop about 10 feet from the router, signal strength is as good as it gets:
[image].  I can try plugging the laptop directly into the router, but I am not convinced it is a bandwidth/latency issue.

My understanding was that the FPP Connect mechanism parses out the data so that only the data for the target controller is included in the file - is that correct (and is that how drag & drop works)?   At this point, I am just fishing for clues....

MartinMueller2003, 2022-01-20:
Dont confuse data content with the file delivery mechanism. The ESP expects the fseq files to be "Sparse". That means the file contains only the channels needed by the ESP itself. 

xLights and Vixen can produce those sparse files and place them on your hard drive. Drag&drop is one mechanism that can be used to transfer the sparse files to the ESP. 

xLights has added "FPP Connect" to automate those steps into a single operation.

marksat666, 2022-02-19:
OK, with the show taken down I have time to do more testing.  I erased and reformatted the SD card, made a new show directory and stripped everything down to just the one controller and corresponding props (icicles and bushes).  I re-rendered the sequence and uploaded it using FPP Connect.  Since I am testing with the controller connected to my laptop I have the monitor output (below).  There is an interesting error when it tried to play the sequence - perhaps that will shed some light on this problem? "All I Want for Chrismas Is You.fseq File does not contain enough data to meet the Stated Channel Count * Number of Frames value."  To double-check, I tried copying the .fseq file directly to the SD card (using a USB adapter) and then put the SD card back into the ESP32 and that WORKED (again, I have done that before but it is not very efficient when the show is up outside).

FW Version	4.x-dev
Build Date	Feb 17 2022 - 14:44:09

Initial Boot:

[    ESP] ESPixelStick v4.x-dev (Feb 17 2022 - 14:44:09)
[    ESP] v4.4-dev-2313-gc69f0ec32
[FileMgr] Flash file system initialized. Used = 208896 out of 327680
[FileMgr] SD Card Size: 15238MB
[FileMgr] Configuration File: '/config.json' loaded.
[FileMgr] Configuration File: '/output_config.json' loaded.
[FileMgr] Configuration File: '/input_config.json' loaded.
[FileMgr] ERROR: Cannot open 'wificonfig.json' for reading. File does not exist.
[FileMgr] SD file: 'wificonfig.json' not found.
[FileMgr] Configuration File: '/config.json' saved 321 bytes.
[WiFiDrv] WiFi Entering State: Connecting Using Config Credentials
[WiFiDrv] Using DHCP
[WiFiDrv] Connecting to 'T-FLO' as ESP32
[WiFiDrv] WiFi Entering State: Connected To AP
[WiFiDrv] Connected with IP: 192.168.88.182
[ WebMgr] Web server listening on port 80
[   FPPD] Listening on port 32320
[  5465][E][vfs_api.cpp:64] open(): /littlefs/www/index.html does not exist
[ WebMgr] WS client connect - 1

File Upload:
[FileMgr] Upload File: 'All I Want for Chrismas Is You.fseq' Started
[FileMgr] Upload File: 'All I Want for Chrismas Is You.fseq' Done (100s)

Fseq Play:

[FileMgr] Configuration File: '/config.json' saved 331 bytes.
[FileMgr] Configuration File: '/config.json' loaded.
[FileMgr] Configuration File: '/input_config.json' saved 1002 bytes.
[FileMgr] Configuration File: '/input_config.json' loaded.
[InputMgr] ParseFseqFile:: Could not start. All I Want for Chrismas Is You.fseq File does not contain enough data to meet the Stated Channel Count * Number of Frames value.
[FileMgr] Configuration File: '/output_config.json' saved 8138 bytes.
[FileMgr] Configuration File: '/output_config.json' loaded.

MartinMueller2003, 2022-02-19:
That means the file was too short. For some reason the file transfer skipped some parts of the file.

MartinMueller2003, 2022-05-23:
Is there any more data on this issue?

closed 2022-05-23 by MartinMueller2003

thomastech, 2022-05-23:
> Is there any more data on this issue?

Was a fix released? 

- Thomas

MartinMueller2003, 2022-05-23:
You would need to upload one of the recent dev builds. A new dev build was made today. An new beta release is probably a few months away.

thomastech, 2022-05-23:
> You would need to upload one of the recent dev builds. A new dev build was made today. 

Great. I will check out the latest dev in a few days and report back.

thomastech, 2022-05-24:
Couldn't wait to try it out. So i forked the latest ESPixelStick Dev 4 code so I could validate the file corruption fix.

TLDR; No good deed goes unpunished.

My intended one hour time allotment turned into an all-day event. Apparently Platformio released their new Core 6.0 a few days ago. And it causes a mess; After several PlatformIO re-installs, each with different build fail symptoms, I learned that the new PlatformIO update has caused a lot of problems for a lot of users. A temporary workaround (targeted to a Marlin user, but the fix applies to other builds) is explained here: https://github.com/platformio/platformio-core/issues/4270

So, several hours later I finally had the new ESPixelStick dev code on my ESP32 TTGO T8 board that controls my Tune-In scrolling marquee sign. 

Test Results: When I upload the FPP files from xLights V2022.10 the file corruption issue is much improved. But I still had a file upload fail (out of six tries). I ran out of time, so I'll try to do more testing at a later date.

BTW, I have a new issue that didn't occur in my previous dev release (a 4.x beta from Nov-2021). Every once in awhile the entire display twitches. That is to say, randomly the entire marquee goes black or is filled with some color. The twitch period only last a few milliSeconds, sort of like a lightning flash. I need to do more testing so I can open a coherent issue ticket on it.

- Thomas

MartinMueller2003, 2022-05-24:
Twitches. Interesting. Need more information about the setup and it would be good if you could attach the file that was playing and info about the master player.

MartinMueller2003, 2022-05-24:
As for issues with the core, I was using 6.0 and did not have a problem. We are not using source filters in the pio setup so builds should not have been impacted. At the moment I am on 6.0.1 and other than an occasional linking failure easily resolved with a "clean" operation, it is working fine.

thomastech, 2022-05-24:
FWIW, PlatformIO Devs report that build issues like I experienced will be solved with the upcoming 6.0.2 milestone release. 
See this: https://github.com/platformio/platformio-core/issues/4286

The latest PlatformIO Dev release (6.0.2a2) is suppose to have the fix. I've installed it and it works. But builds are still unreliable. Each time I launch VS Code the builds fail. But eventually success if I perform Clean operations and keep trying. Better than nothing, but a bit frustrating.

- Thomas

MartinMueller2003, 2022-05-24:
I just did an xLights upload of multiple sets of five files with no errors. Can you send me the file you are having issues with?

[FileMgr] Upload File: 'test.fseq' Started
[FileMgr] Upload File: 'test.fseq' Done (39s)
[FileMgr] Upload File: 'test01.fseq' Started
[FileMgr] Upload File: 'test01.fseq' Done (40s)
[FileMgr] Upload File: 'test02.fseq' Started
[FileMgr] Upload File: 'test02.fseq' Done (40s)
[FileMgr] Upload File: 'test03.fseq' Started
[FileMgr] Upload File: 'test03.fseq' Done (39s)

thomastech, 2022-05-24:
Your posted log seems to suggest you upload files via ESPixelStick's File Management upload tool or something similar to it. The issue affecting me is strictly related to using xLights Tools->FPP Connect tool. This method correctly packs the data (V2 Sparse / Uncompressed).

Just a hunch, but maybe the problem involves xLights communication?

- Thomas

MartinMueller2003, 2022-05-24:
Correct, that log is from the ESP console output which is where error information can be found on failed operations. I used xLights Tools->FPP Connect tool to upload a five file set with no gaps and each file took about 40s to transfer from xLigths to the ESP over my test WiFi network.

thomastech, 2022-05-25:
I wish I had your good fortune.

Here is my setup for the validation:
- In xLights->Tools-FPP Connect I select seven different files. Six are about 14.5MB and one is 8.5MB. Takes about 12 mins to upload.

- In ESPixelStick my Device Setup is as shown in the screenshot.
[image]

- At the end of the test I use ESPixelStick's File Management tab to check files sizes. Success will look like the screenshot below.
[image]

I have very good success (~100%) if I disable the _Primary Input_ on the _Device Setup_ tab. If I don't do this then randomly a file will be corrupted (wrong byte size, won't play). I can recognize a failure during upload because xLight's progress bar will speed up. For example, instead of taking the expected 100 secs to upload it will complete in 60 secs (time varies, just an example). This explains the smaller file size on the bad uploads.

Uploading one file at a time still results in random success.

If you are unable to duplicate this issue then no worries. My workaround (disable Primary Input) is not a big deal to do whenever I upload files.

- Thomas

MartinMueller2003, 2022-05-25:
is the primary input activly receiving data?

thomastech, 2022-05-25:
> is the primary input actively receiving data?

No, not sending pixel data to it.

BTW, I've also changed Primary Input to E1.31.  Same issue as when it is set to DDP. Success requires this input to be disabled. 

Perhaps one of these days I'll dig into the firmware and change it so that the primary input function is automatically disabled during an upload. But that's a project for when I have a lot of free time, an elusive situation for me.

- Thomas

MartinMueller2003, 2022-05-25:
Can you send me your file set? I have DDP input enabled and I do not have issues with my file set.

thomastech, 2022-05-25:
Attached are the FSEQ files.
[fseq_test.zip](https://github.com/forkineye/ESPixelStick/files/8773429/fseq_test.zip)

I suspect that you won't be able to duplicate the problem because it's likely a PC performance issue. My theory is based on your reported success using PlatformIO's version 6.0.x releases. Users like me are not happy campers with it and it has been suggested that PC performance is involved. And if true, perhaps the same reason this randomly occurring ESPixelStick issue does not happen to you.

- Thomas

thomastech, 2022-05-28:
Were you able to duplicate the issue with the file set?

Today I did more testing and I discovered that my workaround is not a reliable way to prevent file upload corruption. Even with the Primary Input disabled I had still had problems. I have no idea why the behavior has changed. 

- Thomas

MartinMueller2003, 2022-05-28:
I am positive the defect for this is on the xLights side. I download your files multiple times using the UI and there are no issues. I download using xLights and the files are truncated.
Setup

Main input E1.31
Secondary input FPP Remote
Output 0 set to WS2811 100 pixels
xLights is up and running for all tests
ESP Web UI is on the file Admin page for all tests
Drag fseq file from the xlights dir to the Web UI drop zone. File transfers properly.
Use Web UI to delete the file (Just to have a clean dir)
Use FPP-Connect to transfer the file and the file is truncated / Corrupted.

The test file is SugarPlumFairies.fseq. 
UI takes:
[FileMgr] Upload File: 'SugarPlumFairies.fseq' Started
[FileMgr] Upload File: 'SugarPlumFairies.fseq' Done (39s)

FPP-Connect:
[FileMgr] Upload File: 'SugarPlumFairies.fseq' Started
[FileMgr] Upload File: 'SugarPlumFairies.fseq' Done (3s)

I also noticed that FPP-Connect transferred a file that had not been selected just prior to sending the desired file.
Here is the complete log
[FileMgr] Upload File: 'SugarPlumFairies.fseq' Started
[FileMgr] Upload File: 'SugarPlumFairies.fseq' Done (39s)
----------- this is the end of the UI test. Everything after this is triggered by xLights ------------------------
[396935][E][vfs_api.cpp:104] open(): /littlefs/www/api/proxies.gz does not exist, no permits for creation
[396953][E][vfs_api.cpp:104] open(): /littlefs/www/api/proxies does not exist, no permits for creation
[396968][E][vfs_api.cpp:104] open(): /littlefs/www/api/proxies/index.html.gz does not exist, no permits for creation
[396983][E][vfs_api.cpp:104] open(): /littlefs/www/api/proxies/index.html does not exist, no permits for creation
[FileMgr] Upload File: 'DaylightSavingTime_Seq.fseq' Started
[FileMgr] Upload File: 'DaylightSavingTime_Seq.fseq' Done (5s)
[FileMgr] Upload File: 'SugarPlumFairies.fseq' Started
[FileMgr] Upload File: 'SugarPlumFairies.fseq' Done (3s)

And this is a snapshot of the xLights screen just prior to starting the upload:
[image]

MartinMueller2003, 2022-05-28:
There is an assumption I am making in the transfer of the files. I assume ALL of the data in the fseq file is going to be used by the target ESP. xLights will remove data not meant for the ESP before transferring the file. This means it is not unexpected to have a smaller file on the ESP than you might have in xLights.

thomastech, 2022-05-28:
> I am positive the defect for this is on the xLights side. I download your files multiple times using the UI and there are no issues. I download using xLights and the files are truncated.

Agreed, files will be truncated due to the omitted data (not used by the target). The screen print I provided earlier  lists the file sizes of successful uploads. If your file sizes match then that means you don't experience the problem that I have. In my use case the files are randomly corrupted (wrong size, don't play).

> I assume ALL of the data in the fseq file is going to be used by the target ESP. xLights will remove data not meant for the ESP before transferring the file. 

That is one of the reasons I use xLights' FPP Connect function. It provides "Sparse" data for my ESPixelStick. Nice feature when it works.

- Thomas

marksat666, 2022-10-11:
Still hit or miss with uploading fseq via FPP connect or drag & drop:

 

Laptop: Lenovo i7 (10th gen) Win10

EsPixelStick (installed via ESPSFlashTool) FW Version 4.0-ci2371263515, Build Date May 23 2022 - 12:31:21

 

Wifi RSSI -35dBm / 100%

Primary Input disabled, Secondary Input FPP Remote

 

Typical console message with files that won't play:

 

[FileMgr] Upload File: 'The Polar Express.fseq' Started
[FileMgr] Upload File: 'The Polar Express.fseq' Done (48s)
[FileMgr] Configuration File: '/config.json' saved 346 bytes.
[FileMgr] Configuration File: '/config.json' loaded.
[FileMgr] Configuration File: '/input_config.json' saved 989 bytes.
[FileMgr] Configuration File: '/input_config.json' loaded.
[InputMgr] ParseFseqFile:: Could not start. The Polar Express.fseq File does not contain enough data to meet the Stated Channel Count * Number of Frames value.
[FileMgr] Configuration File: '/output_config.json' saved 11815 bytes.
[FileMgr] Configuration File: '/output_config.json' loaded.

 

Also, after a few failed attempts the files (even successful ones) stop showing up in the File Management window.  Rebooting gets the files to show.  Deleting the bad file, rebooting, and re-uploading often works (I do have one that refuses to work properly).

 

Please let me know if there is something you would like me to try.

 

Thanks.

 

 

 

 

 

 

 

 
 

Sent: Wednesday, May 25, 2022 at 1:52 PM

MartinMueller2003, 2022-10-11:
I had a ton of issues integrating the latest platformIO and latest IDF into the PixelStick. That has been done and it is working and somethings are working better now.  I would suggest you try the latest build and see if things have improved. You can get pre-built versions at the below link.
dist.zip

| 
| 
|  | 
dist.zip

 |

 |

 |

Martin

marksat666, 2022-10-15:
Martin, there is no link for the dist.zip file (I will test it out once I get it)...

MartinMueller2003, 2022-10-15:
https://drive.google.com/file/d/1DG3wz8lbYNc371OIbeOqkzMaY5Zd2SKA/view?usp=sharing