forkineye/ESPixelStick #432: AnalogWrite Pin Specific to accommodate AC/DC dimming

9ghost9, 2021-11-30:
Would it be possible to have a variant of the awesome EspixelStick firmware ?
One that is designed to use the analogWrite feature ?
It would be excellent to have the option to select a pin and set as R, G, or B and also have the output map  user selectable 
for AC 0 -255 and for DC 255 - 0.

I have found that using a simple IGBT circuit and PWM via analogWrite is the answer to easily controlling/dimming both AC and DC from the Nodemcu/Wemos devices . 

Thank you for at least giving this a read .

MartinMueller2003, 2021-12-01:
Does the analog write used by the built in relay functionality meet your needs? I do know that the use of the I2C external servo driver works to drive LEDs as well.

9ghost9, 2021-12-01:
I was up all night trying to figure this out and  had the -> Aha we have light moment . One would really have to be in the designers head space to understand the workings of this . 

As far as meeting needs , this wasan exercise to get going and is not a clear path method for RGB single channel control .It would really require some work to really be useful in a show . the channel assignment is a guessing game and who knows how this would work via sequencing for other protocols .

However , that said it does dimm r,g, or b  single channel elements . I will test this to day for AC control and if it works will be 
the bees knees I'd say .

Will post results to this thread .
Thank you .

9ghost9, 2021-12-01:
I spoke to soon on testing this with AC . I have built working proto types but unfortunately they are for the 8266 wemos .

I will have to build another proto to support the Esp32 cam board . .

MartinMueller2003, 2021-12-02:
The ESP implementation uses a single byte per channel to determine intensity. Not sure this is enough of a reason to change it but it is possible to map a two byte range to a one byte value

forkineye, 2021-12-28:
> The only short fall is it does now dimm for the sliders full range .
> 
> example : the slider range is 0 - 1023 but the dimming is 0 -255 . Small potatoes when you can dimm AC with the same pwm for RGB .
> 

Can you clarify this a bit? What sliders are you talking about? We only process 8 bit dimming data. Are you able to dim from 0% to 100%?

9ghost9, 2021-12-29:
The range slider PWM needs to be 1k -to 5k for 0 - 255  to work right on the Esp32 . 
and inverting  the output needs tobe added .

MartinMueller2003, 2022-01-04:
Inverting is there already. We do PWM at a fixed frequency. It sounds like you want to change the output frequency instead of the pulse width. That is a whole different type of output. We could possibly do that using the RMT outputs.

closed 2022-05-03 by MartinMueller2003