forkineye/ESPixelStick #433: Network stability CI build DMX output

bloojoop, 2021-12-01:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0-ci1523389109 - downloaded and flashed with this after experiencing the problem on earlier build.
4.0-ci1448851863 - started with this

**Hardware Version**
Espixelstick v3

**Binary release or compiled yourself?**
Binary from release candidate builds

**Operating System (and version)**
Windows 10 / FPP 5.4.1

**Web Browser (and version)**
chrome 

**Access Point**
dedicated unifi AC lite for show network

**Describe the bug**
I configured this pixel stick with primary ddp input, secondary fpp remote.  I had used this device to drive 94 2811s with not much observed issue with both DDP as well as fpp remote.  I changed output to DMX and configured it for a DMX Dumb RGB controller for a couple flood lights.  I added it as a output from a different FPP player and was doing display testing.  If I test with just channel 1-3 (for first flood), testing went OK.  I observed a few dropped ping packets.  When I tested with 1-30 for channels, then it would test for just a short bit then the lights would quit responding.  In troubleshooting, I started a continuous ping.  Shortly after testing, the pixelstick would quick responding for 8 or so ping packets.  then after it recovers it may start outputing to the dmx controller after a short bit but then will halt again.  I tried two different CI builds listed above, and tried moving to e1.31 as well.  I then flashed espixelstick v3.2 serial to the same device and same setup and it is working like a charm with no issues on network timeouts.

MartinMueller2003, 2021-12-01:
What happens when you test the channels using the built in effects or a locally played fseq file? I need to try and figure out if this is an input vs output processing issue. I suspect this is a problem in the WiFi drivers modules but I cant be sure.

bloojoop, 2021-12-01:
Thanks Martin, I just reflashed back to the newest CI build from 11/30.  I have it set on rainbow effect locally and have not been able to get it to "crash".  I see an occasional ping packet dropped but that seems to happen when it is just idle too :).  I have 3 kulps on the same test bench on the same dedicated SSID and accesspoint and they never drop a ping.  It appears to be on the wifi input side to me.

forkineye, 2021-12-01:
When this happens, is the web UI open as well?  Could be related to https://github.com/forkineye/ESPixelStick/issues/378

MartinMueller2003, 2021-12-01:
So the issue is most likely not the output module. That moves it into the Input area or a conflict in the output interrupt handlers with the WiFi input interrupt handlers. You said the issue happens with both DDP and E131? Then that leads me even further into looking at the WiFi modules. Please rerun your tests with the UI turned off. 

I have not been able to get this to happen on my setup.

bloojoop, 2021-12-01:
Hey Shelby, This is Ron Armstrong from xlights zoom room and the Iowa Marshall show.  Anyway, yes the web interface was opened.  I tested by running the display test from fpp withOUT the web interface opened.. I had it run for 10-15 minutes with no issue.  I then opened the web page and it crashed within a handful of seconds.  It seems that this is likely related to 378.  Let me know if you would like any further testing.

bloojoop, 2021-12-01:
Sorry for delay, I wrote that previous response earlier today and just realized I never hit comment.

MartinMueller2003, 2021-12-02:
I have a slightly different test running:

- E1.31 continuously receiving 512 bytes of data
- DMX outputting 512 bytes of data
- UI on the Diagnostics page showing what is happening
- Continuous ping
- Both ESP32 and ESP8266

The ESP8266 crashes every hour or so. the ESP32 seems stable.

MartinMueller2003, 2021-12-15:
It turns out the crashes were self inflicted. I had the lwip setting wrong in my builds. When I gave lwip more bandwidth, the crashes went away.

MartinMueller2003, 2021-12-16:
on an overnight run with DDP + DMX, Chrome detected a period where no traffic was coming from the ESP and 292 messages were lost.. Not sure it is was due to an AP issue or something on the ESP:

`WS Paused - Holding msg: XJ
292 WS Paused - Discarding msg: XJ
WS SEND: XJ`

MartinMueller2003, 2021-12-16:
And this is what it looks like if the ESP stops responding. It is very different, leading me to believe the ESP is still there and the issue is in the network.

WS SEND: XJ
WS SEND: XJ
Discarding msg: XP
8 Discarding msg: XJ
15 WS is down - Discarding msg: XJ

MartinMueller2003, 2021-12-16:
It looks like WS Paused indicates Chrome has hidden the document. This is not an ESP issue. I have yet to see an issue with the DMX output stopping. I do not have pings running yet. Will add those to the mix shortly.

forkineye, 2021-12-16:
> It looks like WS Paused indicates Chrome has hidden the document. This is not an ESP issue. I have yet to see an issue with the DMX output stopping. I do not have pings running yet. Will add those to the mix shortly.

I added code a while back that detects when the page is hidden from view and pauses websocket traffic.

MartinMueller2003, 2021-12-18:
I am not seeing any stability issues in my testing. UI has the usual negative impact on the system.

forkineye, 2021-12-28:
Closing as duplicate of https://github.com/forkineye/ESPixelStick/issues/378

closed 2021-12-28 by forkineye