forkineye/ESPixelStick #434: Invert R,G,B output when selecting Relay as rgb control

9ghost9, 2021-12-02:
As the title requests ,In testing 0 = 255 brightness  and  128 = 0 
This does not work well .

Thank you

MartinMueller2003, 2021-12-03:
I have found a bug (aka missing statement) in the PWM processing path that ignored the invert option. This will be fixed shortly. I am also going to add a frequency selector to allow adjustments to the base PWM frequency on ESP32 devices (cant do it on 8266).

MartinMueller2003, 2021-12-03:
FYI: The ESP32 PWM frequency defaults to 12000 on this and the older ESP32 builds.

9ghost9, 2021-12-03:
You have been relentless in solving issues here Martin . Perfecting this firmware for the Esp32 is greatly appreciated .
It seems to me that 5Khz was a typical user setting for pwm channels and the Esp32 which is fine for rgb but compared to the 8266 @ 1k I believe the 5k still has issue with the optocoupler . I would have to retest a sketch to verify .

As far as pwm on the 8266 and your code , it needs no attention other than inverted output . 

Thanks again for all your great work with this  Martin.

closed 2021-12-07 by forkineye