forkineye/ESPixelStick #438: Unstable WS2811 output > 2 outputs

paultbarrett, 2021-12-07:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
Current Dev (139af2e4e9056ec436ed80319ce508cdaacc0593)
Also tried latest pull request MartinMueller2003
https://github.com/forkineye/ESPixelStick/pull/436
Same outcome with latest beta release

**Hardware Version**
Custom ESP32 ESP-WROOM-32
74HCT541 WS2811 Driver

**Binary release or compiled yourself?**
Compiled self

**Operating System (and version)**

**Web Browser (and version)**

**Access Point**

**Describe the bug**
Strings of WS2811 are connected to outputs 1 & 2 initially, running rainbow effects and DDP from Xlights without issues. Activate output 3, output 3 starts to flicker and not display the correct light sequence, simultaneously output 1 flickers at the same rate as 3 and also ouputs the incorrect light sequence.

Disable output 3 and the outputs 1 & 2 return to normal.

Next test, connect string to outputs 1 & 3, only enable these outputs. Lights flicker like crazy.
Leave output 3 physically connected but disable output 3, output 1 returns to normal.

The issue looks like anything output 3 and above causes all outputs to have errors.

MartinMueller2003, 2021-12-07:
What happens if you use output 3 only?

paultbarrett, 2021-12-08:
@MartinMueller2003, Also still a problem. 

I did further testing to rule out the circuit. If i use GPIO12 for Output 1 this works ok, if I use GPIO2 for Output 3 then this fails too.

The problem does not appear to be related to the GPIOs.

I next disabled all outputs, used GPIO 2 for each Output 1-6 and tested individually. Output 1-2 worked when they were individually set to GPIO 2, Outputs 3-6 all had flickering or no data at all (LEDs not working) when using GPIO2, basically same experince when using the default GPIO mappings for the outputs

paultbarrett, 2021-12-08:
@MartinMueller2003 , it looks like it might be crashing. I have already tested another ESP32 but maybe that is dodgy. Will try a third ESP32

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1100
ho 0 tail 12 room 4
load:0x40078000,len:12280
ho 0 tail 12 room 4
load:0x40080400,len:3076
entry 0x400805ec
E (386) psram: PSRAM ID read error: 0xffffffff

MartinMueller2003, 2021-12-08:
Does it work when you use just ports 3,4,5? Those all use RMT so share an interrupt. Ports 1&2 use the uarts and have a different interrupt structure. I am looking to see if there is an interaction between them.

paultbarrett, 2021-12-09:
If you mean Outputs when stating ports then yes used 3,4,5 there are issues using the default GPIOs.
The output either doesnt work or the light just randomly flickers
Tried a different ESP32 and tried a different PSU, same result

Below are screenshots of the configuration

[image]
[image]
[image]
[image]

MartinMueller2003, 2021-12-09:
The default GPIOS were for a LoLin D32 Pro board. You can assign the Outputs to any GPIO.

paultbarrett, 2021-12-09:
Yes I am aware of that. The ESP-WROOM-32 is just another ESP32 so I cannot see why the GPIOs used on the LoLin board dont work for me on my ESP32. I have also completed the PCB so also a bit painful to change.

Can you think of any reason I would be experiencing what I am.

MartinMueller2003, 2021-12-09:
No all the GPIOs below 30 are the same. There are a few analog output and analog input only GPIOs and they change based on the platform implementation.

paultbarrett, 2021-12-10:
Has anyone tested using WS2811's on all standard outputs. Just wanted to make sure that it is an issue I am only experiencing instead of a issue that has only just been found ?

paultbarrett, 2021-12-10:
Just did a basic setup with a single string of WS2811.
Grabbed a new board Wemos D1 R32, flashed the 4.0-beta4 and enabled effects (Rainbow)

Using GPIO 2 for all tests on the single string.

(Pass) - Enable only Output 1 in setup with GPIO 2 then the LEDs are showing the expected rainbow effect.
(Pass) - Enable only Output 2 in setup with GPIO 2 then the LEDs are showing the expected rainbow effect.
(Fail) - Enable only Output 3 in setup with GPIO 2 then the LEDs not working and a couple are flashing.
Rinse and repeat the same failed step as about with Outputs 4-6, all exhibit the same fault.

From this testing I must conclude that this feature doesnt work, I really cant see I am doing anything special.
I am getting brand new boards of various types and all get the same fault.

closed 2021-12-16 by forkineye