forkineye/ESPixelStick #441: Set WiFi Credentials using Optional Text File (on SD Card Builds only)

thomastech, 2021-12-13:
My Feature request is to allow credential installation via the SD Card (on the builds that support SD Cards).  SSID & PW Credentials would be ASCII strings stored in a text file.

Basic Framework:
- During boot, check the SD card for the presence of file named credentials.txt. 
- If the file exists, read the SSID and Password from it and update ESPixelStick flash with the new values.
- Delete the file (for security purposes).
- If the file is not provided then use the credentials from Flash.

Example raw code (for example only, untested / incomplete):
```
#define MAX_BUFF_SZ 30
SDcardCredentials()
{
  File myFile;
  char c;
  char ssidstr[MAX_BUFF_SZ];
  char pwstr[MAX_BUFF_SZ];
  int i = 0;

  myFile = SD.open("/credentials.txt", FILE_READ);
  if (myFile) { 
    while (myFile.available()) {
      c = (char)myFile.read();
      Serial.print(c);
      if (c == 0x0D) {
        ssidstr[i] = 0x00;
        break;
      }
      else {
        ssidstr[i] = c;
      }
      if (++i >= MAX_BUFF_SZ) {
        Serial.println("SSID Length too long.");
        myFile.close();
        return;
      }
    }
    
    if (myFile.available()) {
        c = (char)myFile.read();
        if (c == 0x0A || c == 0x0D) {
          i = 0;
        }
       else {
         i = 1;
       }
    }
    while (myFile.available()) {
       c = (char)myFile.read();
       Serial.print(c);
       if (c == 0x0D) {
         pwstr[i] = 0x00;
         break;
       }
       else { 
         pwstr[i] = c;
       }
      if (++i >= MAX_BUFF_SZ) {
        Serial.println("PASSWORD Length too long.");
        myFile.close()
        return;
      }
    }

    ssid[0] = ssidstr;
    password[0] = pwstr;
    myFile.close();
    // More code to do -- Store credentials to Flash & delete the credential file.
  }
}
```

- Thomas

MartinMueller2003, 2021-12-15:
Something similar has been discussed in the past. I even have a project that stores the entire config on the SD card as a backup. This is not hard to do. It IS however dangerous. Your WiFi credentials would be stored on a device that could be stuffed into a pocket and read offline. The current config cannot be easily extracted even if you have the ESP connected to your computer.

thomastech, 2021-12-16:
> Your WiFi credentials would be stored on a device that could be stuffed into a pocket and read offline.

I fully agree. That's why the third step in my framework summary explicitly states to delete the credentials.txt file immediately after the new SSID & PW values are stored to flash. Or do something more secure, such as overwrite the file then delete.

- Thomas

MartinMueller2003, 2021-12-16:
At the moment I am not deleting the file since I do not overwrite the flash config. The content of the file is a json structure like this:

{
	"ssid": "YourNetworkName",
	"passphrase": "YourNetworkPassphrase"
}

I have not added it to the readme file since I consider this a work in progress.

thomastech, 2021-12-16:
Thanks for adding your SD Card credential feature. 

I recommend that if the credentials are changed via SD card that they get updated in flash. This will allow removal of the card and still allow the ESPixelStick be able to make the WiFi connection. And if you do this you can erase the file too. :)

- Thomas

sig-tesla, 2021-12-17:
This is pretty dangerous.  It allows anyone to change the WiFi network the
devices connects so with or without your permission.  Imagine someone
taking over a matrix on your house and displaying profanity on it?   If
this is added please allow the feature to be disabled and default it in the
disables state.

(Spoken by someone who just spent  48 hours updating various
applications due to the log4j exploit, which started out as a
convenience feature just like this one. )

Keith

thomastech, 2021-12-17:
> This is pretty dangerous.  It allows anyone to change the WiFi network the devices connects so with or without your permission.  Imagine someone taking over a matrix on your house and displaying profanity on it? 

If someone had intimate knowledge of the ESP32 hardware we've installed, along with access to the SD Card slot, then there's nothing stopping them from unplugging the entire controller and installing their own hardware. No need to even open the enclosure, just swap it out and plug the pixels into the evil doppelganger box.

- Thomas

MartinMueller2003, 2021-12-17:
The design is that the SD card is a permanent part of the installation. Your FSEQ files are on that card. There is nothing stopping anyone from replacing the SD card with malicious files. My problem is that having the Network credentials on the SD card allows someone to access your show network. If you plan to do this, then I suggest a NAT firewall between the show and home networks. That means the person can only access the show network. FYI: FPP (and any PI outside) using the WiFI show network also has this exploit. Anyone can grab a WiFi connected PI and extract your network credentials.

MartinMueller2003, 2021-12-18:
Code has been modified to grab the SD card info when it is present, overwrite the configured settings, save the new values and then delete the file. NOTE: This is not a FIPS compliant delete. This is an FS level delete that does not comply with any FIPS specifications.

thomastech, 2021-12-18:
Fantastic. Thank you for the SD Card credential feature.

- Thomas

closed 2021-12-20 by forkineye