forkineye/ESPixelStick #443: Unable to flash V4 on an Esp-01s with 4MB of flash

thecrazy, 2021-12-13:
I found some Esp-01s' with 4mb of flash on Amazon and ordered a few. I confirmed in software and using the part# they really have a 4 mb flash, but the V4 flash tool is giving me errors and I suspect the higher baud rate of 460800 is the issue. How can we change lower it back to 115200?

thecrazy, 2021-12-13:
I found a data sheet for the flash chip: https://datasheet.lcsc.com/lcsc/1811072025_XMC-XM25QH32BHIG_C328460.pdf

thecrazy, 2021-12-14:
Ok I managed to flash V4 on the ESP-01s. To be able to do so I had to put the V4 firmware files in the V3 flash tool and added matching entries for it in the json files. I also had to replace mkspiffs with mklittlefs (just renamed it to mkspiffs cause it seemed harccoded?) thankfully they have the same parameters.

The only problem i'm having (and its the same problem on a nodemcu I tried V4 on before) the effects testing output is completely messed up. Most of the effects are messed up (except rainbow somehow) but the tests in xlights work. Also, I had to manually change the gamma setting which is not a big deal but its the first time I had to do it. It was at 1 instead of 2. Maybe I changed it by accident.

Now if someone could tell me how to keep gpio2 high at boot, that would be great!

forkineye, 2021-12-14:
You should be able to use the V4 flashtool. You can edit the baudrate in the firmware.json file.  I think you're talking about GPIO0? Pull it to 3.3v through a 10k resistor.

MartinMueller2003, 2021-12-15:
Not sure why you are having issues with the effects. They dont really know what hardware you are on. All they are told is how many channels are being used.

thecrazy, 2021-12-16:
It's not the baud rate, It works fine with V4's default speed using the V3 flash tool. I'll run more tests tomorrow, but V4 definitely has communication issues with the esp-01. I don't think the problem is with my computer, I flashed it on many NodeMCU and never had an error.

While googling around, I read something about one of the tools in the bin folder having communication difficulties with the esp at some point. Didn't check the versions, so I'm not 100% sure.

Did any of the toll/library change going from V3 to V4?

thecrazy, 2021-12-21:
@MartinMueller2003 @forkineye Whats the difference between selecting Espixelstick V3 and Wemos mini D1 in the flash tool? Which one should we be using for a stand alone esp8266?

forkineye, 2021-12-21:
The Generic Wemos version adds a secondary output for relays and pwm.  The V3 binary doesn't include it since it's a DIY thing and not physically supported on the V3 hardware. If rolling your own, I'd use the generic wemos version.

closed 2021-12-28 by forkineye