forkineye/ESPixelStick #446: Unable to update firmware from web interface on 4.0-Beta3

jeffu231, 2021-12-19:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0-Beta3

**Hardware Version**
V3 Wemos

**Binary release or compiled yourself?**
Beta release from Github

**Operating System (and version)**
Mac OS

**Web Browser (and version)**
Safari 15.1

**Access Point**
Microtik

**Describe the bug**
I downloaded the 4.0-Beta 4 and used the tool to generate the EFU files to upload. I uploaded to 4.0-Beta3 and it uploads and appears to flash. Upon reboot version is unchanged. Device was deployed and I did not want to bring it in to physically flash it.

DrAnonymous, 2021-12-21:
I tried the same thing using my D1 mini and it didn't work. Looking at the serial output, the ESP threw an exception shortly after it started the update and rebooted. I was using OSX Catalina and Chrome.

forkineye, 2021-12-21:
The 4.0 betas don't support EFU updates yet.

DrAnonymous, 2021-12-23:
Should this be left open until that feature is implemented? Is there a feature list for v4 beyond this?

forkineye, 2021-12-28:
There was a note about it in the original 4.0 beta. I'll leave this open until it's implemented.

MartinMueller2003, 2022-07-29:
Changes are in main

closed 2022-07-29 by MartinMueller2003