forkineye/ESPixelStick #451: Add Platformio Script to Support littlefs Data Uploads

thomastech, 2021-12-28:
Using the Arduino IDE to upload ESPixelStick's littlefs data files is the recommended flash method, but it's a bit inconvenient. I'm using a different workaround that has restored Platformio's ability to flash littlefs onto my ESP32 builds. Everything is explained here:
https://github.com/lorol/LITTLEFS/tree/master/examples/LITTLEFS_PlatformIO

I think it would help other project builders if the littlefs script was included in ESPixelStick's source distribution. Here is a summary of what needs to be done.

```
1. Include the two files in the project's root directory, as follows:
    mklittlefs.exe
    LittleFSBuilder.py
2. Add this line of text under platformio.ino's extra_scripts section:
    LittleFSBuilder.py
``` 
After these additions the _Upload Filesystem Image_ task will correctly install littlefs data files. BTW, I've tested it on my ESP32 builds, but as yet I have not tried it on a ESP8266.

I will admit that I am surprised that any of this monkey business is needed. Platformio claims to support littlefs when the "board_build.filesystem = littlefs" is used in platformio.ino. But it doesn't work on my ESP32 builds unless the project has the files described above. Perhaps I am doing it wrong and my proposed solution is not needed, so please fill me in if that is the case.

- Thomas

MartinMueller2003, 2022-02-15:
I have added the functionality needed to upload LittleFs on ALL supported platforms. It is a manual process and is invoked from this menu:

[image]

brmeadows2841, 2022-02-15:
I did a build filesystem image then upload FS image to ESP32cam.
The ESP32 displayed page not found error.
What additional steps do I need to do ?

thomastech, 2022-02-16:
> What additional steps do I need to do ?

The latest code download does not appear to have the required changes (last PR was 10 days ago). So maybe all that is needed is to wait for the PR that adds the new functionality.

- Thomas

MartinMueller2003, 2022-02-16:
The latest code is on my branch. The PR has been opened but not yet reviewed by Shelby.

MartinMueller2003, 2022-02-16:
> I did a build filesystem image then upload FS image to ESP32cam. The ESP32 displayed page not found error. What additional steps do I need to do ?

That should be all that is needed. I will dig up my CAM board and see what is happening

MartinMueller2003, 2022-02-16:
Just verified with my latest build. Loads fine and browser works fine. I dod notice that the LED did not turn off as expected. I will make a fix for that as well.

MartinMueller2003, 2022-02-16:
Code with LED adjusted to GPIO 33 and disabled the brownout detector has been uploaded and added to the PR.

brmeadows2841, 2022-02-17:
> Just verified with my latest build. Loads fine and browser works fine. I did notice that the LED did not turn off as expected. I will make a fix for that as well.

I got the ESP32 cam running with your last PR. I gulped then uploaded  filesystem image then the build image.
Good job! The Flash LED is off and the ESP32 is running a local sequence file.

MartinMueller2003, 2022-02-17:
I think it is a browser issue because I am not seeing these issues in my setup.

MartinMueller2003, 2022-02-17:
My latest PR has been integrated into Shelby's main branch.

MartinMueller2003, 2022-03-21:
A user has sent in a log clearly indicating that the pio script is working as intended.

closed 2022-03-21 by MartinMueller2003