forkineye/ESPixelStick #453: E1.31 - DMX Mode Setup / Feature Request

mxklb, 2022-01-03:
Remote control ESPixelStick effects using DMX --> E1.31

I think of something like https://github.com/Aircoookie/WLED/wiki/E1.31-DMX#3-effect

Actually it's possible to send RGB values for each pixel via E1.31. To address LED pixel colors one can send RGB values to the DMX addresses [1,2,3],[4,5,6], ..., [510,511,512]. This is a really cool feature (=[Multiple RGB](https://github.com/Aircoookie/WLED/wiki/E1.31-DMX#4-multiple-rgb)). To further enhance this - or to avoid sending every pixel value individually (like for solid effects) - it would be nice to be able to configure different DMX modes within the firmware.

An advantage of this is to be able to control multiple ESPixelSticks within a single DMX universe using less DMX addresses.

Just something similar to [the WLED settings](https://github.com/Aircoookie/WLED/wiki/E1.31-DMX#settings)

MartinMueller2003, 2022-01-04:
Actually that feature has almost nothing to do with DMX and simply provides an external mechanism to perform bulk operations and effects on the outputs by interpreting control information in the E1.31 stream. 

This is implementable but with limited remaining code space it will be Shelby's call whether we want to add this.

forkineye, 2022-01-04:
I see where such a feature is useful for WLED as they already have a pretty large set of effects in place, whereas we do not.  I could see this being useful in a live setting where you are actively manipulating built-in effects over multiple devices similar to how many commercial DMX fixtures work.  For it to be of any use though, we would need to implement the underlying foundation first (a broad effects set). For now, we won't be implementing this, but I'd like to leave it open for ideas.

MartinMueller2003, 2022-01-04:
I went through the WLED effects set a few months ago and added some effects to our effect engine. We have most of the core effects. Many of the others are derivative effects and did not look like they made sense for this project.

mxklb, 2022-01-04:
I do some DMX stage lightning and really enjoy this WLED feature to simply trigger LED stripes like other DMX fixtures ..

It's up to you, just an idea to get some more/other E1.31 controls out of ESPixelSticks.

Bear1007, 2022-10-12:
Will WLED work with our DMX512-P4 RGBW pixels? If so what controller do you recommend we try?

MartinMueller2003, 2022-10-12:
I am not sure you are asking in the right place. You might want to look at the WLED discussions on Discord. To try to answer your question: ESPixelStick IS a controller software package that runs on many ESP32 / ESP8266 platforms. WLED IS controller software that runs on many platforms (they are companion / competing products). You will need to go to the WLED site to get a list of the platforms they support.