forkineye/ESPixelStick #457: Max Pixel limit on ESP8266 increase

TownshendLights, 2022-01-08:
**Is your feature request related to a problem? Please describe.**
My matrix is made up of 34 rows of 36 pixels. making a total of 1224 pixels. I had to run 1188 to be under the limit. Emitting one line of pixels.
The  matrix which runs a ESP8266 with SD card.

**Describe the solution you'd like**
 The Matrix has 1224 pixels. Can the limit be increased to 1224 so I can run the last line of pixels please.

**Additional context**
Add any other context or screenshots about the feature request here.

MartinMueller2003, 2022-01-08:
There is not enough ram available to increase the number of supported channels.

TownshendLights, 2022-01-08:
OK. Thanks. I'll wait until I can get the ESP32 working with the SD card.

closed 2022-01-08 by TownshendLights