forkineye/ESPixelStick #458: ESPS32 CAM - Led

brmeadows2841, 2022-01-09:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
ESPixleStick 4 beta 4

**Hardware Version**

ESP32 CAM
https://www.amazon.com/dp/B08G4LB3W6?psc=1&ref=ppx_yo2_dt_b_product_details

**Binary release or compiled yourself?**
self complied

**Operating System (and version)**
Windows 11

**Web Browser (and version)**
Version 97.0.4692.71 (Official Build) (64-bit)
Chrome 
**Access Point**

**Describe the bug**
I have ESPixelStick firmware installed on an Esp32 cam board, version below.
LANDZO ESP32 CAM Module Development Board ESP32 with Camera Module
The on board LED come on startup after re-booting. What options do I have for turning off the on board LED?

MartinMueller2003, 2022-01-13:
There is a setting in the ESP-Cam GPIO Defs file. You define the pin being used for your LED on your platform and the startup code will turn it off.

MartinMueller2003, 2022-01-13:
Is that the correct GPIO for your device?

brmeadows2841, 2022-01-14:
Yes GPIO4  control the flash, but apparently if I use the SD card the flash is also turned on.

The ESP32-CAM has a very bright built-in LED that can work as a flash when taking
photos. That LED is internally connected to GPIO 4.

That GPIO is also connected to the microSD card slot, so you may have troubles when trying to use both at the same time – the flashlight will light up when using the microSD card.

Note: one of our readers shared that if you initialize the microSD card as follows, you won’t have this problem because the microSD card won’t use that data line.*

SD_MMC.begin("/sdcard", true)
* we found that this works and that the LED will not make that flash effect. However, the LED remains on with low brightness – we’re not sure if we are missing something.

brmeadows2841, 2022-01-14:
The make is wireless-tag
http://www.wireless-tag.com/portfolio/wt-esp32-cam/

MartinMueller2003, 2022-01-14:
I assume you built the image for the "ESP32_CAM" target? If so then the SD card will not work anyway since the SPI pins are different between your board and the ESP32-CAM board.

Our SD Lib only uses two data lines and Data2 in your chart is unused. If you built for the ESP32-CAM board type then GPIO 4 should be held high after boot. It is possible that your card needs the line held low. I can add a selector for defining the proper state for LED off.

closed 2022-01-14 by forkineye