forkineye/ESPixelStick #459: Suggestion (with solution): Reduce WiFi latency on ESP32 builds

thomastech, 2022-01-11:
ESPixelStick ESP32 builds can experience random WiFi latency during STA connections. For example, here are sample ping times from a Beta 4.0 build that demonstrates the issue (avg 257mS Ping Times):
```
Reply from 192.168.1.195: bytes=32 time=28ms TTL=255
Reply from 192.168.1.195: bytes=32 time=45ms TTL=255
Reply from 192.168.1.195: bytes=32 time=117ms TTL=255
Reply from 192.168.1.195: bytes=32 time=346ms TTL=255
Reply from 192.168.1.195: bytes=32 time=288ms TTL=255
Reply from 192.168.1.195: bytes=32 time=318ms TTL=255
Reply from 192.168.1.195: bytes=32 time=315ms TTL=255
Reply from 192.168.1.195: bytes=32 time=329ms TTL=255
Reply from 192.168.1.195: bytes=32 time=130ms TTL=255
Reply from 192.168.1.195: bytes=32 time=555ms TTL=255
Reply from 192.168.1.195: bytes=32 time=223ms TTL=255
Reply from 192.168.1.195: bytes=32 time=577ms TTL=255
Reply from 192.168.1.195: bytes=32 time=212ms TTL=255
Ping statistics for 192.168.1.195:
    Packets: Sent = 39, Received = 39, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 28ms, Maximum = 577ms, Average = 257ms
```

The latency is caused by a power saving feature during WiFi STA connections (AP mode is not affected). To reduce power the ESP32 disables the WiFi module between two DTIM Beacon intervals and wakes up automatically before the next Beacon arrival. The sleep time is decided by the DTIM Beacon interval time of the router (typically 100ms to 1000ms).

But there is an easy fix: We can turn off the WiFi sleep feature.

The solution is to add a sleep disable function just before wifi.begin(). Here is a simplified snippet of the fix:

```
    WiFi.setSleep(false);  // Add this, improve ping times.
    WiFi.begin();
```
The ESP8266 is not involved so the existing build defines can be used to add the disable function only to the ESP32 builds.

Here are sample Ping times on my TTGO-T8 ESP32 board after I added the fix (avg 5mS Ping Times):
```
Reply from 192.168.1.195: bytes=32 time=3ms TTL=255
Reply from 192.168.1.195: bytes=32 time=3ms TTL=255
Reply from 192.168.1.195: bytes=32 time=3ms TTL=255
Reply from 192.168.1.195: bytes=32 time=13ms TTL=255
Reply from 192.168.1.195: bytes=32 time=3ms TTL=255
Reply from 192.168.1.195: bytes=32 time=6ms TTL=255
Reply from 192.168.1.195: bytes=32 time=9ms TTL=255
Reply from 192.168.1.195: bytes=32 time=4ms TTL=255
Reply from 192.168.1.195: bytes=32 time=7ms TTL=255
Reply from 192.168.1.195: bytes=32 time=3ms TTL=255
Reply from 192.168.1.195: bytes=32 time=7ms TTL=255
Reply from 192.168.1.195: bytes=32 time=3ms TTL=255
Reply from 192.168.1.195: bytes=32 time=3ms TTL=255
Reply from 192.168.1.195: bytes=32 time=6ms TTL=255
Ping statistics for 192.168.1.195:
    Packets: Sent = 46, Received = 46, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 3ms, Maximum = 13ms, Average = 5ms
```

- Thomas

MartinMueller2003, 2022-01-13:
It is interesting that this line:
     esp_wifi_set_ps (WIFI_PS_NONE);
which was intended to do this did not have the expected effect.

thomastech, 2022-01-13:
> It is interesting that this line:
esp_wifi_set_ps (WIFI_PS_NONE);
which was intended to do this did not have the expected effect.

Yes, that should have worked. It's called early, so maybe something along the way is resetting it.

- Thomas

closed 2022-01-14 by forkineye