forkineye/ESPixelStick #461: Minor error on SD card file detail

brmeadows2841, 2022-01-15:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
version 4 beta 4

**Hardware Version**
ESP32

**Binary release or compiled yourself?**

compiled
**Operating System (and version)**
Win 11

**Web Browser (and version)**

Chrome
**Access Point**

**Describe the bug**
When SEQ files are uploaded to the SD card file details are incorrectly displayed.
Example:
Notice the T at end of date and Z at end of time.

Wizards in Winter -test 4.fseq | 2022-01-15T08:28:08.000Z

MartinMueller2003, 2022-01-16:
That is a specific time format that conforms to the ISO time display specification. The value between the T and the Z tell you the string includes a time and that the Time Zone offset that was applied is zero.

brmeadows2841, 2022-01-16:
I understand.

MartinMueller2003, 2022-01-17:
In reality the time info is almost useless since the ESP has no access to a time server (no NTP and no RTC) so the time is "taken" from the browser at the moment you transfer the files. That is at best a close approximation and at worst misleading. I suspect we may want to remove the time portion of the string.

brmeadows2841, 2022-01-17:
It would be beneficial to have a time stamp on the seq upload.

MartinMueller2003, 2022-02-06:
I updated the time stamps to a more common format.

closed 2022-02-06 by forkineye