forkineye/ESPixelStick #475: ESP32 CAM random reboot beta3 and beta4

pakeonoahu, 2022-02-18:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
V4 beta3 and beta4

**Hardware Version**
ESP32-CAM

**Binary release or compiled yourself?**
Binary release

**Operating System (and version)**
Windows 10

**Web Browser (and version)**
Chrome

**Access Point**
TP-Link

**Describe the bug**
Device configured for testing, play fseq standalone from TF card, tried different fseq files, with or without browser connected, using serial monitor, device rebooted every few minutes.
[espixelstick4error.txt](https://github.com/forkineye/ESPixelStick/files/8095015/espixelstick4error.txt)

MartinMueller2003, 2022-02-18:
Please retry using the latest baseline code.

pakeonoahu, 2022-02-19:
Thanks, I flashed it with the latest CI built and it's been stable, but the built date show in Nov 13 2021 - 14:54:11, did more testing last night found the Beta4 that was having issue were with 25ms fseq, 50ms didn't reboot.  The CI build has be running well with 25ms fseq so far, will see how long before any reboots.

MartinMueller2003, 2022-02-20:
I have been looking at the OTA and found a few issues with it. They have been fixed in the latest PR that is already on Shelby's main branch.

pakeonoahu, 2022-02-22:
Hmm.. Tried to build the main branch and got error on secrets.h missing:

*****************************************************************
* Looking for secrets.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:secrets.h"
* Web  > https://registry.platformio.org/search?q=header:secrets.h
*
*****************************************************************

 #include "secrets.h"
          ^~~~~~~~~~~
compilation terminated.
*** [.pio\build\d1_mini_mhetesp32minikit\src\src\network\WiFiDriver.cpp.o] Error 1
===================================================================== [FAILED] Took 6.89 seconds =====================================================================

Environment               Status    Duration
------------------------  --------  ------------
espsv3                    FAILED    00:00:06.457
d1_mini                   FAILED    00:00:04.511
d32_pro                   FAILED    00:00:07.035
d32_pro_eth               FAILED    00:00:06.880
esp32_cam                 FAILED    00:00:06.737
esp32_ttgo_t8             FAILED    00:00:06.890
d1_mini32                 FAILED    00:00:06.868
d1_mini32_eth             FAILED    00:00:06.960
d1_mini_mhetesp32minikit  FAILED    00:00:06.895
================================================================ 9 failed, 0 succeeded in 00:00:59.232 ================================================================

MartinMueller2003, 2022-02-22:
You need to create a secrets.h file or comment out the line in WiFiMgr that brings in secrets.h. The main build scripts comment out the line but those are not running when you do your own build.

pakeonoahu, 2022-02-22:
Thank you, I see it in the build.yaml on creating it, I am able complied it now on VSC or Arduino.

MartinMueller2003, 2022-02-22:
Arduino will NOT work. We are using a non Arduino partition scheme. You must use platformio (aka VSC) to build and upload.

pakeonoahu, 2022-02-23:
I found that out, the webpage never work with the Arduino, there's no option for me to split the flash in half either for ESP32-CAM.  Platformio works and it's been stable so far, will let it run for a day or two, I have noticed Output 5 has been removed and the LED(camera light) is disabled also, less work to invert the GPIO 4 to turn it off.  Thank you for the hard work on supporting this hobby.

MartinMueller2003, 2022-05-10:
No activity and nothing much to do.

closed 2022-05-10 by MartinMueller2003