forkineye/ESPixelStick #481: ESPixelStick_Firmware-4.0-ci1859266670

brmeadows2841, 2022-03-03:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**

ESPixelStick_Firmware-4.0-ci1859266670

**Hardware Version**

**Binary release or compiled yourself?**
release
**Operating System (and version)**
Win11

**Web Browser (and version)**
brave

**Access Point**
home

**Describe the bug**
 There is some strangeness going on with this build of ESPixelStick ver4.

I have tried to use this Alpha/Beta version and both have similar issues. When setting the outputs 1-6 to ws2811 I save the configuration. If I leave the setting page then come back the outputs are set to disable. I have had a similar occurrence with this version and the previous PR. Sometimes the outputs will stick for output 1 and 2, then if setting more outputs then coming back to check all output are all disabled. I have seen all six outputs to ws2811 be set then after a short time without leaving the page will update then all will be disabled again.

This occurs on ESP32 Cam, ESP32 Nodemcu and Wemos D1 ESP32.

jensrene, 2022-03-03:
I had that Issue and noticed an issue that space Was full.
Deleting some on my build unused Web files, then full rebuild including FS, esp32 full wipe and Flash and the Problem was fixed.

MartinMueller2003, 2022-03-04:
The latest dev builds using platformio fix this issue. However, the fix makes it impossible to build the current version using Arduino IDE. Ther Arduino IDE does not allocate enough space to the littlefs partition.

MartinMueller2003, 2022-03-04:
As for the issue being presented, please capture the console output at the time you try to save the config. If there was a problem saving the settings, that is where it will get reported.

brmeadows2841, 2022-03-05:
I will do that, I'm guessing from the serial output or the status screen. I don't want to taint this issue but appears I may have a grounding issue with my FT232RL device.

brmeadows2841, 2022-03-08:
Long story shortened, apparently I have ghosts in the machine.

brmeadows2841, 2022-03-08:
Here is the log file from a ESP32 experiencing the settings not staying.

ESP] ESPixelStick v4.x-dev (Feb 17 2022 - 14:44:09)
[    ESP] v4.4-dev-2313-gc69f0ec32
[FileMgr] Flash file system initialized. Used = 200704 out of 327680
[   485][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[   485][E][sd_diskio.cpp:790] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[   788][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[FileMgr] No SD card installed
[FileMgr] Configuration File: '/config.json' loaded.
[NetworkMgr] Using old style WiFi Settings
[  1016][E][vfs_api.cpp:64] open(): /littlefs/output_config.json does not exist
E (1438) rmt: rmt_tx_stop(273): RMT CHANNEL ERR
E (1439) rmt: rmt_tx_stop(273): RMT CHANNEL ERR
E (1440) rmt: rmt_tx_stop(273): RMT CHANNEL ERR
E (1441) rmt: rmt_tx_stop(273): RMT CHANNEL ERR
E (1566) rmt: rmt_tx_stop(273): RMT CHANNEL ERR
E (1568) rmt: rmt_tx_stop(273): RMT CHANNEL ERR
E (1569) rmt: rmt_tx_stop(273): RMT CHANNEL ERR
E (1571) rmt: rmt_tx_stop(273): RMT CHANNEL ERR
E (1595) rmt: rmt_tx_stop(273): RMT CHANNEL ERR
E (1596) rmt: rmt_tx_stop(273): RMT CHANNEL ERR
E (1596) rmt: rmt_tx_stop(273): RMT CHANNEL ERR
E (1597) rmt: rmt_tx_stop(273): RMT CHANNEL ERR
E (1636) rmt: rmt_tx_stop(273): RMT CHANNEL ERR
E (1637) rmt: rmt_tx_stop(273): RMT CHANNEL ERR
E (1637) rmt: rmt_tx_stop(273): RMT CHANNEL ERR
E (1638) rmt: rmt_tx_stop(273): RMT CHANNEL ERR
[FileMgr] Configuration File: '/output_config.json' saved 8068 bytes.
[  1692][E][vfs_api.cpp:64] open(): /littlefs/input_config.json does not exist
[FileMgr] Configuration File: '/input_config.json' saved 1000 bytes.
[FileMgr] Configuration File: '/input_config.json' loaded.
[FileMgr] Configuration File: '/output_config.json' loaded.
[FileMgr] Configuration File: '/config.json' saved 333 bytes.
[WiFiDrv] WiFi Entering State: Connecting Using Config Credentials
[WiFiDrv] Using DHCP
[WiFiDrv] Connecting to 'MeadTec2' as ESPShop
[WiFiDrv] WiFi Entering State: Connected To AP
[WiFiDrv] Connected with IP: 10.0.0.113
[ WebMgr] Web server listening on port 80
[   FPPD] Listening on port 32320
[ WebMgr] WS client connect - 1
[ WebMgr] WS client connect - 2
[ WebMgr] WS client disconnect - 1
[ WebMgr] WS client connect - 3
[ WebMgr] WS client disconnect - 3
[ WebMgr] WS client connect - 4
[ WebMgr] WS client disconnect - 4
[ WebMgr] WS client connect - 5
[ 24984][E][vfs_api.cpp:64] open(): /littlefs/www/index.html does not exist
[ WebMgr] WS client disconnect - 2
[ WebMgr] WS client disconnect - 5
[ WebMgr] WS client connect - 6

brmeadows2841, 2022-03-08:
ESPixelStick V4beta4 does not have this issue. I just checked on the same hardware device as above.

I the problem on ESPixelStick_Firmware-4.0-ci1945838590 and 4.0-ci1859266670

MartinMueller2003, 2022-03-11:
This error: "[ WebMgr] *** onWsEvent() error: Incoming message is too long." indicates the configuration messages has unexpected data in it. I would like to see your config files. 

1. go to the device setup page
2. press F12
3. refresh the page
4. Make your changes
5. Save your changes
6. Select "User Messages" in the debug pane.
7. Scroll back in the message list to find this line: "WS SEND: {"cmd":{"set":{"output":{"output_config"........" See below for the full text for my system.
8. Post that entire message as a comment on this ticket..

WS SEND: {"cmd":{"set":{"output":{"output_config":{"cfgver":1,"MaxChannels":9000,"channels":{"0":{"0":{"type":"WS2811","color_order":"rgb","pixel_count":"100","group_size":"1","zig_size":"0","gamma":"1","brightness":"100","interframetime":"300","prependnullcount":"0","appendnullcount":"0","data_pin":"2","showgamma":false},"1":{"type":"GECE","pixel_count":63,"brightness":80,"data_pin":2},"2":{"type":"DMX","num_chan":64,"baudrate":57600,"data_pin":2},"3":{"type":"Renard","num_chan":64,"baudrate":57600,"data_pin":2},"4":{"type":"Serial","num_chan":"64","baudrate":"57600","gen_ser_hdr":"1","gen_ser_ftr":"1","data_pin":"2"},"7":{"type":"UCS1903","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":25,"prependnullcount":0,"appendnullcount":0,"data_pin":2},"8":{"type":"TM1814","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":500,"prependnullcount":0,"appendnullcount":0,"data_pin":2,"currentlimit":50},"11":{"type":"GS8208","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":300,"prependnullcount":0,"appendnullcount":0,"data_pin":2},"12":{"type":"Disabled"},"type":0},"1":{"0":{"type":"WS2811","color_order":"rgb","pixel_count":"100","group_size":"1","zig_size":"0","gamma":"1","brightness":"100","interframetime":"300","prependnullcount":"0","appendnullcount":"0","data_pin":"13","showgamma":false},"1":{"type":"GECE","pixel_count":63,"brightness":80,"data_pin":13},"2":{"type":"DMX","num_chan":64,"baudrate":57600,"data_pin":13},"3":{"type":"Renard","num_chan":64,"baudrate":57600,"data_pin":13},"4":{"type":"Serial","num_chan":64,"baudrate":57600,"gen_ser_hdr":"","gen_ser_ftr":"","data_pin":13},"7":{"type":"UCS1903","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":25,"prependnullcount":0,"appendnullcount":0,"data_pin":13},"8":{"type":"TM1814","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":500,"prependnullcount":0,"appendnullcount":0,"data_pin":13,"currentlimit":50},"11":{"type":"GS8208","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":300,"prependnullcount":0,"appendnullcount":0,"data_pin":13},"12":{"type":"Disabled"},"type":0},"2":{"0":{"type":"WS2811","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":300,"prependnullcount":0,"appendnullcount":0,"data_pin":12},"7":{"type":"UCS1903","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":25,"prependnullcount":0,"appendnullcount":0,"data_pin":12},"8":{"type":"TM1814","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":500,"prependnullcount":0,"appendnullcount":0,"data_pin":12,"currentlimit":50},"11":{"type":"GS8208","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":300,"prependnullcount":0,"appendnullcount":0,"data_pin":12},"12":{"type":"Disabled"},"type":12},"3":{"0":{"type":"WS2811","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":300,"prependnullcount":0,"appendnullcount":0,"data_pin":14},"7":{"type":"UCS1903","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":25,"prependnullcount":0,"appendnullcount":0,"data_pin":14},"8":{"type":"TM1814","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":500,"prependnullcount":0,"appendnullcount":0,"data_pin":14,"currentlimit":50},"11":{"type":"GS8208","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":300,"prependnullcount":0,"appendnullcount":0,"data_pin":14},"12":{"type":"Disabled"},"type":12},"4":{"0":{"type":"WS2811","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":300,"prependnullcount":0,"appendnullcount":0,"data_pin":32},"7":{"type":"UCS1903","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":25,"prependnullcount":0,"appendnullcount":0,"data_pin":32},"8":{"type":"TM1814","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":500,"prependnullcount":0,"appendnullcount":0,"data_pin":32,"currentlimit":50},"11":{"type":"GS8208","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":300,"prependnullcount":0,"appendnullcount":0,"data_pin":32},"12":{"type":"Disabled"},"type":12},"5":{"0":{"type":"WS2811","color_order":"rgb","pixel_count":100,"group [... cut ...] :10,"en":false,"Min":650,"Max":2350,"rev":false,"b16":false,"sca":true},{"id":11,"en":false,"Min":650,"Max":2350,"rev":false,"b16":false,"sca":true},{"id":12,"en":false,"Min":650,"Max":2350,"rev":false,"b16":false,"sca":true},{"id":13,"en":false,"Min":650,"Max":2350,"rev":false,"b16":false,"sca":true},{"id":14,"en":false,"Min":650,"Max":2350,"rev":false,"b16":false,"sca":true},{"id":15,"en":false,"Min":650,"Max":2350,"rev":false,"b16":false,"sca":true}]},"12":{"type":"Disabled"},"type":12}}}}}}}

brmeadows2841, 2022-03-11:
WS SEND: {"cmd":{"set":{"output":{"output_config":{"cfgver":1,"MaxChannels":9000,"channels":{"0":{"0":{"type":"WS2811","color_order":"rgb","pixel_count":"100","group_size":"1","zig_size":"0","gamma":"1","brightness":"100","interframetime":"300","prependnullcount":"0","appendnullcount":"0","data_pin":"2","showgamma":false},"1":{"type":"GECE","pixel_count":63,"brightness":80,"data_pin":2},"2":{"type":"DMX","num_chan":64,"baudrate":57600,"data_pin":2},"3":{"type":"Renard","num_chan":64,"baudrate":57600,"data_pin":2},"4":{"type":"Serial","num_chan":64,"baudrate":57600,"gen_ser_hdr":"","gen_ser_ftr":"","data_pin":2},"7":{"type":"UCS1903","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":25,"prependnullcount":0,"appendnullcount":0,"data_pin":2},"8":{"type":"TM1814","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":500,"prependnullcount":0,"appendnullcount":0,"data_pin":2,"currentlimit":50},"11":{"type":"GS8208","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":300,"prependnullcount":0,"appendnullcount":0,"data_pin":2},"12":{"type":"Disabled"},"type":0},"1":{"0":{"type":"WS2811","color_order":"rgb","pixel_count":"100","group_size":"1","zig_size":"0","gamma":"1","brightness":"100","interframetime":"300","prependnullcount":"0","appendnullcount":"0","data_pin":"13","showgamma":false},"1":{"type":"GECE","pixel_count":63,"brightness":80,"data_pin":13},"2":{"type":"DMX","num_chan":64,"baudrate":57600,"data_pin":13},"3":{"type":"Renard","num_chan":64,"baudrate":57600,"data_pin":13},"4":{"type":"Serial","num_chan":64,"baudrate":57600,"gen_ser_hdr":"","gen_ser_ftr":"","data_pin":13},"7":{"type":"UCS1903","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":25,"prependnullcount":0,"appendnullcount":0,"data_pin":13},"8":{"type":"TM1814","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":500,"prependnullcount":0,"appendnullcount":0,"data_pin":13,"currentlimit":50},"11":{"type":"GS8208","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":300,"prependnullcount":0,"appendnullcount":0,"data_pin":13},"12":{"type":"Disabled"},"type":0},"2":{"0":{"type":"WS2811","color_order":"rgb","pixel_count":"100","group_size":"1","zig_size":"0","gamma":"1","brightness":"100","interframetime":"300","prependnullcount":"0","appendnullcount":"0","data_pin":"12","showgamma":false},"7":{"type":"UCS1903","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":25,"prependnullcount":0,"appendnullcount":0,"data_pin":12},"8":{"type":"TM1814","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":500,"prependnullcount":0,"appendnullcount":0,"data_pin":12,"currentlimit":50},"11":{"type":"GS8208","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":300,"prependnullcount":0,"appendnullcount":0,"data_pin":12},"12":{"type":"Disabled"},"type":0},"3":{"0":{"type":"WS2811","color_order":"rgb","pixel_count":"100","group_size":"1","zig_size":"0","gamma":"1","brightness":"100","interframetime":"300","prependnullcount":"0","appendnullcount":"0","data_pin":"14","showgamma":false},"7":{"type":"UCS1903","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":25,"prependnullcount":0,"appendnullcount":0,"data_pin":14},"8":{"type":"TM1814","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":500,"prependnullcount":0,"appendnullcount":0,"data_pin":14,"currentlimit":50},"11":{"type":"GS8208","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":300,"prependnullcount":0,"appendnullcount":0,"data_pin":14},"12":{"type":"Disabled"},"type":0},"4":{"0":{"type":"WS2811","color_order":"rgb","pixel_count":"100","group_size":"1","zig_size":"0","gamma":"1","brightness":"100","interframetime":"300","prependnullcount":"0","appendnullcount":"0","data_pin":"32","showgamma":false},"7":{"type":"UCS1903","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":25,"prependnullcount":0,"appendnullcount":0,"data_pin":32},"8":{"type":"TM1814","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":500,"prependnullcount":0,"appendnullcount":0,"data_pin":32,"currentlimit":50},"11":{"type":"GS8208","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":300,"prependnullcount":0,"appendnullcount":0,"data_pin":32},"12":{"type":"Disabled"},"type":0},"5":{"0":{"type":"WS2811","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":300,"prependnullcount":0,"appendnullcount":0,"data_pin":33},"7":{"type":"UCS1903","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":25,"prependnullcount":0,"appendnullcount":0,"data_pin":33},"8":{"type":"TM1814","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":0,"gamma":1,"brightness":100,"interframetime":500, [... cut ...] 0,"Max":2350,"rev":false,"b16":false,"sca":true},{"id":11,"en":false,"Min":650,"Max":2350,"rev":false,"b16":false,"sca":true},{"id":12,"en":false,"Min":650,"Max":2350,"rev":false,"b16":false,"sca":true},{"id":13,"en":false,"Min":650,"Max":2350,"rev":false,"b16":false,"sca":true},{"id":14,"en":false,"Min":650,"Max":2350,"rev":false,"b16":false,"sca":true},{"id":15,"en":false,"Min":650,"Max":2350,"rev":false,"b16":false,"sca":true}]},"12":{"type":"Disabled"},"type":12}}}}}}}
esps.js:33 WS SEND: XP

MartinMueller2003, 2022-03-12:
That output looks fine and it is less than 9KB long so it will fit in the WS buffer. Not sure why you are getting that message. I just turned on all 7 of my outputs and it is working just fine. The buffer in the latest version was increased from 9KB to 10KB for platforms with no PSRAM and to 20KB for platforms that do have PSRAM. The beta versions all have 9KB buffers which is big enough but unforgiving if the Generic Serial headers are really big.

closed 2022-05-03 by brmeadows2841